Koenkk/zigbee2mqtt #4630: LQI slowly goes up to 255, then drops to 0 and climbs again...bug?

s3frank, 2020-10-10:
Hi there,
I am running Conbee II on RPI4. I have Debian 10 installed, Docker and run Homeassistant (0.116.2) in containers on it. I have installed the official hassio addon for zigbee2mqtt and i am absolutely loving it. You have done amazing work and packaged it so well, thank you for that. My network is stable and automations in NodeRed work super smooth and fast.

But my LQI data seems to not reflect that reality at all. In fact I think something is going wrong in the code around endpoint availability checking. I have configured my zigbee2mqtt addon version: 1.15.0 with:
```
advanced:
  availability_timeout: 120
```

My LQI across all devices appears to be looping. It starts with a number say 35 across most devices no matter how far or close they are. And then slowly they all climb up to 255 and then drop to 0. And then it starts again slowly to 255.
I think it has something to do with the availability check that is turned on by the 2 minute timeout routine I have activated.

Here are some screenshots for you to further show what is happening.
[image]

[image]

Koenkk, 2020-10-10:
@chrishae could this be a bug in the LQI calculation in the Conbee firmware?

s3frank, 2020-10-14:
Let me know if you want me to provide more details and or extract some debugging information. I've got a day off tomorrow so can spend some time on this if I know what you need.

Koenkk, 2020-10-14:
I noticed a similar use was reported here: https://github.com/Koenkk/zigbee2mqtt/issues/4039 , The Conbee developer mentioned it's a bug in the firmware. Let's close this and continue there.

closed 2020-10-14 by Koenkk

ChrisHae, 2020-10-14:
Yes the lqi values in the data indications have the wrong value. This is a FW issue and will be fixed sometime?
You can use LQI requests (network scan) in z2mqtt. This will report the right lqi values.