Koenkk/zigbee2mqtt #4631: Devices randomly stop responding and can't repair

LouisMT, 2020-10-10:
## What happened
Devices randomly stop responding on my network. For example, pressing a button on a device doesn't show any logging at all. Replacing the battery of the device sometimes helps, but it seems that another device gets kicked off the network when I do this.

I suspect there are too many devices connected directly to the coordinator. I bought a TRADFRI signal repeater a long time ago, thinking devices would pair through that when my network would get too big for a single coordinator. However, when I checked the network map, only 4 devices seem to be connected through it (I have around 20 total). In an effort to pair one of my unresponsive devices through the signal repeater, it has left the network but I cant repair it (no logging at all), even when it is next to the signal repeater.

Then I found #2888, which should allow me to only allow pairing on the signal repeater. Unfortunately this issue has been closed because there has been no activity. However, it sounds like the perfect solution for me (and I guess many others). I would like to know if there is any status update for this issue, or if there is any other way for me to allow me to use all of my devices.

## What did you expect to happen
All devices to stay responsive.

## How to reproduce it (minimal and precise)
* Use a coordinator and a TRADFRI signal repeater as router
* Pair more devices than the coordinator can be connected to directly

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2530
Adapter firmware version:

LouisMT, 2020-10-10:
There are actually less than 16 devices connected to the coordinator according to the network map and the device list... however there were some more (unused) devices in the configuration.yaml and database.db. I've removed those too.

But still I can't repair my device that's left the network earlier. I also tried to reset another device that wasn't responding anymore, it's flashing (pairing mode) but nothing shows up in the logs, not even in debug mode...

LouisMT, 2020-10-11:
Yesterday I've also plugged in a new coordinator. Everything was working fine, but today a device had already stopped responding again. The signal repeater doesn't seem to be used at all now. Maybe I should get a coordinator that can have more direct children, and hope that'll work.

danieledwardgeorgehitchcock, 2020-10-15:
Hi,

Could you please provide a bit more information?

It would be good to know the following to help assess your issue:

What devices (brand / type) are dropping out from your network?

How far away is your Tradfri router from your coordinator, and what is it’s average LQI?

What firmware and version are you running on your CC2530?

Could you also post a full version of your configuration.yaml?

Are all of the devices in your network (excluding the router) endpoints (i.e. battery) or do you have any further mains powered devices?

LouisMT, 2020-11-07:
I bought a zig-a-zig-ah Zigbee stick, which can handle 100+ devices (according to the zigbee2mqtt website). So far it's working really great, and as a bonus everything is responding instantly too (with the CC2531 everything had a little delay).

closed 2020-11-07 by LouisMT