Koenkk/zigbee2mqtt #4632: Hue Motion (9290012607) does not get configure on device_announcement

sjorge, 2020-10-10:
## What happened
Every so often my Hue Motion sensor will stop working, removing the battery and plugging it back in, or hitting the setup button makes it send a device announce again.

However this is not enough, I need to manually trigger a configure action for it before it starts working again.

## What did you expect to happen
Since it looks like it sometimes forgets it configuration, I expected it to be (re)configure on a device_announcement when we already know the device and interviewing is complete.

## How to reproduce it (minimal and precise)
Bit hard, I only had one of mine drop off the network twice so far. I saw a thread that they tend to do that.
It's not the end of the world for me, but it would be nice that it would work again after it announce itself again.

## Debug info
Zigbee2MQTT version: `1.15.0-dev`
Adapter hardware: 
- [X] CC2531
- [] CC2530
- [] CC26X2R1
- [] CC1352P-2
- [] Conbee II
Adapter firmware version:
```json
{
  "meta": {
    "maintrel": 3,
    "majorrel": 2,
    "minorrel": 6,
    "product": 0,
    "revision": 20200728,
    "transportrev": 2
  },
  "type": "zStack12"
}
```

Koenkk, 2020-10-10:
@sjorge you are running a source routing firmware right? The fix applied for CC2652/CC1352 is not in the CC2531 firmware yet.

sjorge, 2020-10-10:
Correct, I am running the custom source routing firmware with the pairing fix.

ChessSpider, 2020-10-10:
subscribing as i have the same issue, i use CC2530_CC2591

Koenkk, 2020-10-11:
@sjorge please try this fw: 
[CC2531ZNP-Prod_20201011_source_routing.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/5360998/CC2531ZNP-Prod_20201011_source_routing.hex.zip) . If this works I think it's time to release a new fw.

@ChessSpider the zstack 3 cc2530 firmwares don't have source routing enabled, I doubt if your issue is exactly the same.

ChessSpider, 2020-10-11:
Ok, that's too bad. I'm experiencing the same type of problem and workaround. Next time it happens I'll make a new topic.. should be in about a week or 2-3 :)

sjorge, 2020-10-11:
> @sjorge please try this fw:
> [CC2531ZNP-Prod_20201011_source_routing.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/5360998/CC2531ZNP-Prod_20201011_source_routing.hex.zip) . If this works I think it's time to release a new fw.

Won't have time to do so until tuesday, I assume this firmware has both the hue motion and the pairing fix?

Koenkk, 2020-10-11:
yes both

sjorge, 2020-10-12:
firmware update, I guess we wait now.

sjorge, 2020-10-25:
So far, seems stable with the new firmware.
Had a few red led moment but on the next motion event it worked again.

~ sjorge

sjorge, 2020-11-18:
Haven't had the issue again after the newer firmware, every few days one of them doesn't respond, but then 30 seconds later on it works again. So it seems to send an update, doesn't get a reply and rediscovered the coordinator and all is good.

closed 2020-11-18 by sjorge