Koenkk/zigbee2mqtt #4634: ZZH Coordinator - Failed to connect to the adapter

Anjerlaan, 2020-10-10:
## What happened

I had succesfully installed the ZZH adapter.
And I could repair all devices wihout a problem.
Also the groups were joining all relevant devices

However this was all done without restarting the Raspberry Pi
After the re-start or shutdown/start Zigbee2MQTT runs with an error 
zigbee2mqtt has difficulties connecting to the ZZH adapter but not always, on average 1 out of 5:

The message shown is: (which repeats, because Z2M is trying to connect )

```
okt 10 18:59:32 Domotica-Pi systemd[1]: Started zigbee2mqtt.
okt 10 18:59:33 Domotica-Pi npm[990]: > zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
okt 10 18:59:33 Domotica-Pi npm[990]: > node index.js
okt 10 18:59:34 Domotica-Pi npm[990]: Zigbee2MQTT:info  2020-10-10 18:59:34: Logging to console and directory: 'data/log/2020-10-10.18-59-34' filename: log.txt
okt 10 18:59:34 Domotica-Pi npm[990]: Zigbee2MQTT:debug 2020-10-10 18:59:34: Removing old log directory 'data/log/2020-10-10.18-15-37'
okt 10 18:59:35 Domotica-Pi npm[990]: Zigbee2MQTT:debug 2020-10-10 18:59:35: Loaded state from file /opt/zigbee2mqtt/data/state.json
okt 10 18:59:35 Domotica-Pi npm[990]: Zigbee2MQTT:info  2020-10-10 18:59:35: Starting Zigbee2MQTT version 1.15.0 (commit #ed8b4e5)
okt 10 18:59:35 Domotica-Pi npm[990]: Zigbee2MQTT:info  2020-10-10 18:59:35: Starting zigbee-herdsman...
okt 10 18:59:35 Domotica-Pi npm[990]: Zigbee2MQTT:debug 2020-10-10 18:59:35: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyUSB1"}}'
okt 10 18:59:55 Domotica-Pi npm[990]: Zigbee2MQTT:error 2020-10-10 18:59:55: Error while starting zigbee-herdsman
okt 10 18:59:55 Domotica-Pi npm[990]: Zigbee2MQTT:error 2020-10-10 18:59:55: Failed to start zigbee
okt 10 18:59:55 Domotica-Pi npm[990]: Zigbee2MQTT:error 2020-10-10 18:59:55: Exiting...
okt 10 18:59:55 Domotica-Pi npm[990]: Zigbee2MQTT:error 2020-10-10 18:59:55: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
okt 10 18:59:55 Domotica-Pi npm[990]:     at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:92:31)
okt 10 18:59:55 Domotica-Pi npm[990]:     at Generator.throw (<anonymous>)
okt 10 18:59:55 Domotica-Pi npm[990]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
okt 10 18:59:55 Domotica-Pi npm[990]: npm ERR! code ELIFECYCLE
okt 10 18:59:55 Domotica-Pi npm[990]: npm ERR! errno 1
okt 10 18:59:55 Domotica-Pi npm[990]: npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`
okt 10 18:59:55 Domotica-Pi npm[990]: npm ERR! Exit status 1
okt 10 18:59:55 Domotica-Pi npm[990]: npm ERR!
okt 10 18:59:55 Domotica-Pi npm[990]: npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
okt 10 18:59:55 Domotica-Pi npm[990]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
okt 10 18:59:56 Domotica-Pi npm[990]: npm ERR! A complete log of this run can be found in:
okt 10 18:59:56 Domotica-Pi npm[990]: npm ERR!     /home/pi/.npm/_logs/2020-10-10T16_59_55_951Z-debug.log
okt 10 18:59:56 Domotica-Pi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
okt 10 18:59:56 Domotica-Pi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
okt 10 18:59:57 Domotica-Pi systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
okt 10 18:59:57 Domotica-Pi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 2.
okt 10 18:59:57 Domotica-Pi systemd[1]: Stopped zigbee2mqtt.
```

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200925

Anjerlaan, 2020-10-10:
It looks that the USB PORTS sometimes switche after a reboot/restart:

During ZZH installation this command returned /ttyUSB1 as port for ZZH
But I also have a RFXCOM_RFXtrx433XL USB adapter

pi@Domotica-Pi:~ $ ls -l /dev/serial/by-id
totaal 0
lrwxrwxrwx 1 root root 13 okt 10 19:09 usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root 13 okt 10 19:09 usb-RFXCOM_RFXtrx433XL_DO2VYPCR-if00-port0 -> ../../ttyUSB1

I have changed my configuration now to /ttyUSB0
Now it is working correctly..

How can I prevent that the USB ports switch?

Koenkk, 2020-10-11:
You can fix this by creating an udev rule: https://www.zigbee2mqtt.io/information/FAQ.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port

Anjerlaan, 2020-10-11:
I already did figuren it out this morning...
Only i followed this guide
https://www.freva.com/nl/2019/06/20/geef-vaste-usb-poortnamen-aan-je-raspberry-pi/

Koenkk, 2020-10-11:
Cool! I assume this can be closed :smile:

closed 2020-10-11 by Koenkk