Koenkk/zigbee2mqtt #4635: Ikea e1743 don’t show payload on or off

dustardly, 2020-10-10:


dustardly, 2020-10-11:
Only xiaomi button give payload hold release while tradfri payload is empty

https://hastebin.com/yutewajage.rust

Koenkk, 2020-10-12:
- Try updating to the latest firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default
- If that doesn't help try to force configure device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure make sure to wakeup the device right before doing so (by clicking a button on it) and check the logs if executed successfully.

dustardly, 2020-10-22:
after upgrade firmware on cc2531 all working again! Thanks!

neerdoc, 2020-11-08:
Mine works too now. I think I accidentally had an IKEA lamp too close to the switch while pairing (I was pairing lots of stuff one after the other). It all worked out once I removed the battery from the switch -> re-paired the lamp -> moved the lamp to another room -> chose "Forget device" for the switch -> rebooted Raspberry Pi -> put the battery back in the switch -> factory reset the switch (4 clicks within 5 seconds) -> re-paired the switch.

closed 2020-12-17 by github-actions[bot]