Koenkk/zigbee2mqtt #4637: YSRSAI ZB-CT01

gerripeach, 2020-10-11:


gerripeach, 2020-10-18:
devices.js
```
    {
        zigbeeModel: ['ZB-CT01'],
        model: 'ZB-CT01',
        vendor: 'YSRSAI',
        description: 'Zigbee LED controller (WW/CW)',
        fromZigbee: [fz.on_off, fz.brightness, fz.ignore_basic_report],
        toZigbee: [
            tz.ysrsai_zb_ct01,
        ],
    },
```

toZigbee.js
```
    // YSRSAI
    ysrsai_zb_ct01: {
        key: ['state', 'brightness', 'brightness_percent'],
        convertSet: async (entity, key, value, meta) => {
            let {message} = meta;
            let state = message.hasOwnProperty('state') ? message.state.toLowerCase() : undefined;
            let brightness = undefined;
            if (message.hasOwnProperty('brightness')) brightness = Number(message.brightness);
            else if (message.hasOwnProperty('brightness_percent')) brightness = Math.round(Number(message.brightness_percent) * 2.55);

            if (state == undefined && brightness !== undefined && brightness === 0) {
                meta.message.state = 'off';
            }

            return await converters.light_onoff_brightness.convertSet(entity, key, value, meta);
        },
        convertGet: async (entity, key, meta) => {
            return await converters.light_onoff_brightness.convertGet(entity, key, meta);
        },
    },
```

**OR add to general light_onoff_brightness:**
```
            if (state == undefined && brightness !== undefined && brightness === 0) {
                meta.message.state = 'off';
            }
```

What is the "correct" way to do it? General approach or individual controller?

Koenkk, 2020-10-18:
- What does happen when you set brightness: 0?
- Can you provide the herdsman debug log of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

closed 2020-11-26 by github-actions[bot]