Koenkk/zigbee2mqtt #4638: BTicino living now - Device doesn't expose anything

theoden9017, 2020-10-11:
Hi,
Last week I bought the BTicino smart switch K4003C without its Gateway.
using **zigbee2mqtt-edge 1.15.0-dev (commit #a132094)** add-on with my conbee2 I have discovered the device, but unfortunatly no action was expose

```
0x0004740000b18711
Device doesn't expose anything
```

so i set zigbee_shepherd_devices to true
and added to my data_path the folder for supporting new devices (**v12.0.208**):
- https://github.com/danielwelch/hassio-zigbee2mqtt#adding-support-for-new-devices
- https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v12.0.208

Once i restarted i get this error:

```
Zigbee2MQTT:info  2020-10-11 10:11:45: Logging to console and directory: '/share/zigbee2mqtt/log/2020-10-11.10-11-45' filename: log.txt
/app/node_modules/zigbee-herdsman-converters/devices.js:1641
        device.exposes = device.exposes.concat([exposes.numeric('linkquality').withUnit('lqi')]);
                                                        ^
TypeError: exposes.numeric is not a function
    at /app/node_modules/zigbee-herdsman-converters/devices.js:1641:57
    at Array.map (<anonymous>)
    at Object.<anonymous> (/app/node_modules/zigbee-herdsman-converters/devices.js:1630:26)
    at Module._compile (internal/modules/cjs/loader.js:1137:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)
    at Module.load (internal/modules/cjs/loader.js:985:32)
    at Function.Module._load (internal/modules/cjs/loader.js:878:14)
    at Module.require (internal/modules/cjs/loader.js:1025:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/app/node_modules/zigbee-herdsman-converters/index.js:3:17)
npm
 ERR! code ELIFECYCLE
npm
 ERR! errno 1
npm ERR! zigbee2mqtt@1.15.0-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-10-11T08_11_46_517Z-debug.log
```

I tryed the device list using just Philips and BTicino, and commenting this line
`// device.exposes = device.exposes.concat([exposes.numeric('linkquality').withUnit('lqi')]);`

once i commented it start, but I still have no action exposed on Bticino switch discovered by zigbee2mqtt.

lurking for the same problem i found something similar here:
Koenkk/zigbee2mqtt#2626
but I'm not sure he is working with the zigbee2mqtt-edge add-on

Maybe the problem its related with the firmware version of BTicino switch:
[image]

did you have any clue of what could cause this issue?

Koenkk, 2020-10-11:
Not all devices have been migrated to `exposes` yet, please wait until this has been done.

closed 2020-10-11 by Koenkk

theoden9017, 2020-10-11:
well ok, but according with guys in this post (https://github.com/Koenkk/zigbee2mqtt/issues/2626) 
K4003C device should be able to be toggle on and off.
At least this is what i undestood from the post of:  @keyita and @marcgarciamarti

Koenkk, 2020-10-11:
@theoden9017 yes that is true, but for that `exposes` does not have to be added. I think you saw `0x0004740000b18711
Device doesn't expose anything` in the frontend?

theoden9017, 2020-10-11:
Yes exactly.
so if that message appear also to other guys, and is not related with the problem on my K4003C I have not the faintest idea of what should i do, I am stuck 😢

marcgarciamarti, 2020-10-11:
> well ok, but according with guys in this post (https://github.com/Koenkk/zigbee2mqtt/issues/2626) 
> 
> K4003C device should be able to be toggle on and off.
> 
> At least this is what i undestood from the post of:  @keyita and @marcgarciamarti
> 
> 
I confirm that I can happily switch off and on the light circuit the 4003 is part of. I could be happier. @Koenkk you are a semi-God!!

Koenkk, 2020-10-11:
@theoden9017 you can control it via MQTT (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset). Once exposes is add for this device you can also control it via the exposes section in the frontend.

theoden9017, 2020-10-11:
> 
> 
> > well ok, but according with guys in this post (#2626)
> > K4003C device should be able to be toggle on and off.
> > At least this is what i undestood from the post of:  @keyita and @marcgarciamarti
> 
> I confirm that I can happily switch off and on the light circuit the 4003 is part of. I have no idea what "exposes" is, and what it may be used for. I could be happier. @Koenkk you are a semi-God!!

did use my same configuration? 
what is your firmware version of K4003C?

could you send me your device.js, maybe i could find what's is wrong in my configuration.

@Koenkk, i already controll my philips hue light with MQTT so, i saw that appear an entity in Mosquitto to turn on and off the device. btw with the K4003C if I turn on the switch it just turn back after a 1-2 second.

another important thing is the color of the led inside the K4003C, is still Red. ⚱️

theoden9017, 2020-10-11:
Maybe also this could be usefull: 
```

Zigbee2MQTT:error 2020-10-11 16:12:26: Publish 'set' 'state' to '0x0004740000b18711' failed: 'Error: Command 0x0004740000b18711/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
Zigbee2MQTT:info  2020-10-11 16:12:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x0004740000b18711' failed: 'Error: Command 0x0004740000b18711/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (no response received)'","meta":{"friendly_name":"0x0004740000b18711"},"type":"zigbee_publish_error"}'
```

Koenkk, 2020-10-11:
`Zigbee2MQTT:error 2020-10-11 16:12:26: Publish 'set' 'state' to '0x0004740000b18711' failed: 'Error: Command 0x0004740000b18711/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
`  is the issue.

- What adapter are you using?
- Did you try to repair the device?
- Can you move the device closer to the adapter?

theoden9017, 2020-10-11:
1. Here the adapter (conbee2):

      -   adapter: deconz
      -   port: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2150543-if00

2. I tryed to repair without any result.

3. Moving the device closer to the adapter is possibile but i should completly dismount and remount the switch

Koenkk, 2020-10-12:
Can you provide the herdsman debug logging so @chrishae can look into this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

theoden9017, 2020-10-12:
Ok, This is what I get after trying to change state of K4003C

```
Zigbee2MQTT:error 2020-10-12 09:38:59: Publish 'set' 'state' to '0x0004740000b18711' failed: 'Error: Command 0x0004740000b18711/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (no response received)'
Zigbee2MQTT:info  2020-10-12 09:38:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x0004740000b18711' failed: 'Error: Command 0x0004740000b18711/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (no response received)'","meta":{"friendly_name":"0x0004740000b18711"},"type":"zigbee_publish_error"}'
2020-10-12T07:39:06.226Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 52
2020-10-12T07:39:06.231Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-10-12T07:39:06.231Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-12T07:39:07.084Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2020-10-12T07:39:07.087Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-12T07:39:07.130Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 53
2020-10-12T07:39:07.135Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xcdbf request id: 5 confirm status: 0
2020-10-12T07:39:07.136Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-12T07:39:07.137Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 5
2020-10-12T07:39:16.205Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 54
2020-10-12T07:39:16.210Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-10-12T07:39:16.210Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-10-12T07:39:26.224Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 55
2020-10-12T07:39:26.233Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-10-12T07:39:26.234Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

@Koenkk It's ok or should I pass you a file?

@marcgarciamarti 
did use my same configuration?
what is your firmware version of K4003C?

marcgarciamarti, 2020-10-12:
@theoden9017 happy to contribute but please let me know where I can find the firmware version please.
Also, my configuration is pretty simple. I ditched manual configuration and enabled MQTT discovery prior to pairing the 4003. It all happens automagically. The only thing I do is defining a friendly name in the configuration file, and HA handles the rest on its own.

theoden9017, 2020-10-12:
> @theoden9017 happy to contribute but please let me know where I can find the firmware version please.
> Also, my configuration is pretty simple. I ditched manual configuration and enabled MQTT [...]

**automagically** made my day 🤣 

Btw in order to see the firmware just go in the front-end page of zigbee2mqtt in your home assistant portal and click on the device. _(if it's not visible, just go to supervisor >> zigbee2mqtt and enable  "Show in sidebar")_
once you do that should appear the same page i have posted in the OP

1 more question, are you using zigbee2mqtt-edge (development version)? or the stable one?

marcgarciamarti, 2020-10-12:
> **automagically** made my day 🤣
>
> Btw in order to see the firmware just go in the front-end page of zigbee2mqtt in your home assistant portal and click on the device. _(if it's not visible, just go [...]

Hi, I believe the show in sidebar feature is not available in the version I'm runnig. 

[image]

any other place I could find that information? Give me a while. I'm going to try it with z2massistant and get back

theoden9017, 2020-10-12:
mmm, maybe is something introduced with 1.15.0 version

marcgarciamarti, 2020-10-12:
[image]
Apparently, zigbee2mqttassistant struggles to display the firmware version. If you figure out how to get the firmware version from CLI, just let me know.
thanks

theoden9017, 2020-10-12:
ok, for what i understand the firmware version should be 00b as well as mine.
so... i'll try to use the not edge version, maybe it's enought

theoden9017, 2020-10-12:
meh, even with main version i get the same error, 
wich adapter and wich device are you using @marcgarciamarti ? (i have set deconz in the config file, and the device is a conbee2)

marcgarciamarti, 2020-10-12:
> meh, even with main version i get the same error, 
> 
> wich adapter and wich device are you using @marcgarciamarti ? (i have set deconz in the config file, and the device is a conbee2)

CC2351 followed the instructions on z2m site.

theoden9017, 2020-10-12:
mmm could you explain better this? 

> I ditched manual configuration and enabled MQTT discovery prior to pairing the 4003

how you do that? could you share me your configuration **[without your network_key of corse]**

Thanks for your help guys 🤟

marcgarciamarti, 2020-10-12:
> mmm could you explain better this? 
> 
> 
> 
> > I ditched manual configuration and enabled MQTT discovery prior to pairing the 4003
> 
> 
> 
> how you do that? could you share me your configuration **[without your network_key of corse]**
> 
> 
> 
> Thanks for your help guys 🤟 

Add-on or file?

theoden9017, 2020-10-12:
add-on is ok

theoden9017, 2020-10-12:
for mqtt broker are you using mosquitto?

jd766, 2020-10-16:
I have multiple Bticino K4003C switches and K4531C smart outlets (identified as Legrand 067775) in my house, I recently disconnected them from the Bticino Gateway and connected to z2m using Slaesh's CC2652R stick and Mosquitto, but I'm having many issues. 

First of all, after joining the network for the first time, they leave it after the successful interview, only to join again after a reset (short press) or power cycle; if I'm lucky then they start working, otherwise some of them don't report their state. 
They accept /set commands but after the first one they start lagging or behaving weirdly.

Other switches work perfectly. Some of them start working after a few re-pair attempts, some still refuse to work. After adding those devices I feel like my zigbee network slowed down (the smart outlets report power once a second, so there's much more traffic), so I'm still trying to understand if there's a network issue causing/compounding this problem, but after all I have 42 devices, there's people with many more who allegedly have no issues.

theoden9017, 2020-10-16:
maybe, could it be the gateway doing some interference?

I just ordered CC2531, with Conbee2 it doesn't want to work.

jd766, 2020-10-16:
The gateway coexisted with both USB sticks before my "migration", and I had
1 bticino light switch and two smart outlets already connected to z2m for a
couple of months as a test, with no issue

theoden9017, 2020-10-20:
trying with CC2531 i didn't see any device, I'll try with a fresh install, one of these days.

thanks for your support.

jd766, 2020-10-20:
These Bticino switches are kinda peculiar, if they don't see any open coordinator they apparently give up searching, you have to reset them with the small button behind the cover on the front, or power cycle them, in order to connect them again

theoden9017, 2020-10-20:
yes, I already did it, but although with conbee the device was at least detected with the cc2531, I can't find any.
i'm pretty sure it's a matter of wrong configuration of the cc2531 or the broker.

majorhector, 2021-04-24:
Someone have solved? I still have this problem. I see the device in z2m but doesn't expose anything. I can't see the state. I have resetted them 100 times but nothing change