Koenkk/zigbee2mqtt #4640: (IKEA) remote issues, both pairing and double messages

mvjt, 2020-10-11:
#4132  What happened
I´ve been having some issues with IKEA remotes falling out and having a really hard time re-pairing them. The only solution has been to force_remove in Z2M and then repair them and they work for a while. Now I also realized that they send double messages.

## What did you expect to happen
Easy pairing and no double messages

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: latest and latest-dev
Adapter hardware: ZZH
Adapter firmware version: 2020-09-25

Debug log: https://hastebin.com/yubepediqo.apache

Koenkk, 2020-10-11:
@mvjt looking at your log it looks that the switch produces 2 actions (and Zigbee2MQTT converters it two 2 MQTT messages as expected). Would you be able to sniff the traffic of this: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

mvjt, 2020-10-11:
@Koenkk  thanks for quick feedback.

Attached is a dump and in the end I toggle the button on the 1744/Symfonisk twice (generating four messages). 
0x14b457fffe6b2899 is the MAC
Network key(s) are default.

[testZigbeeDump.zip](https://github.com/Koenkk/zigbee2mqtt/files/5361720/testZigbeeDump.zip)

Koenkk, 2020-10-12:
Your log contains 2x toggle (at no. 41 and no. 97), did you toggle twice?

mvjt, 2020-10-12:
Log attached. I added the following to the docker compose file. First I tried to pair the problematic device and then I toggled the button twice on the second Symfonisk...
```
    environment:
      - TZ=Europe/Stockholm
      - DEBUG=zigbee-herdsman*
```

[debugLog.txt](https://github.com/Koenkk/zigbee2mqtt/files/5367075/debugLog.txt)

Koenkk, 2020-10-13:
The herdsman debug logging is only logged to stdout, (use `docker logs -f CONTAINER`) to retrieve logs.

ASNNetworks, 2020-10-25:
I'm not sure if this is the code requested, this is all I see (with `zigbee_herdsman_debug: true`). I'm running the zigbee2mqtt addon through hassio btw (RPI4) and Conbee II.

`Zigbee2MQTT:debug 2020-10-25 23:54:37: Received Zigbee message from 'Symfonisk Woonkamer', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2020-10-25 23:54:37: MQTT publish: topic 'zigbee2mqtt/Symfonisk Woonkamer', payload '{"action":"toggle","battery":47,"brightness":201,"last_seen":"2020-10-25T23:54:37+01:00","linkquality":215,"rate":195,"update":{"state":"updating"},"update_available":false}'
Zigbee2MQTT:info  2020-10-25 23:54:37: MQTT publish: topic 'zigbee2mqtt/Symfonisk Woonkamer', payload '{"action":"","battery":47,"brightness":201,"last_seen":"2020-10-25T23:54:37+01:00","linkquality":215,"rate":195,"update":{"state":"updating"},"update_available":false}'
Zigbee2MQTT:info  2020-10-25 23:54:37: MQTT publish: topic 'zigbee2mqtt/Symfonisk Woonkamer/action', payload 'toggle'
Zigbee2MQTT:debug 2020-10-25 23:54:37: Received Zigbee message from 'Symfonisk Woonkamer', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2020-10-25 23:54:37: MQTT publish: topic 'zigbee2mqtt/Symfonisk Woonkamer', payload '{"action":"toggle","battery":47,"brightness":201,"last_seen":"2020-10-25T23:54:37+01:00","linkquality":215,"rate":195,"update":{"state":"updating"},"update_available":false}'
Zigbee2MQTT:info  2020-10-25 23:54:37: MQTT publish: topic 'zigbee2mqtt/Symfonisk Woonkamer', payload '{"action":"","battery":47,"brightness":201,"last_seen":"2020-10-25T23:54:37+01:00","linkquality":215,"rate":195,"update":{"state":"updating"},"update_available":false}'
Zigbee2MQTT:info  2020-10-25 23:54:37: MQTT publish: topic 'zigbee2mqtt/Symfonisk Woonkamer/action', payload 'toggle'`

Koenkk, 2020-10-26:
@ASNNetworks You need to be on the edge version I believe (otherwise this is not logged). Please don't post the log here directly but on pastebin.com and link it here.

ASNNetworks, 2020-10-26:
Can I stop the stable addon, install the edge version and start that (without pairing and setting up configuration)? Or will mess that up my current installation?

mvjt, 2020-10-26:
https://0bin.net/paste/LQrFdQvc#3QFGXRqqUqf6gzsb7AMQalwRzgk2z-smxSk1g6cwXUP

Device: kok_volume_control

I pressed the button (toggle) once.

Koenkk, 2020-10-28:
@mvjt could you also post the lines logged before the first line? This line should contain `zigbee-herdsman:controller:log Received 'zcl' data`

ASNNetworks, 2020-11-06:
> https://0bin.net/paste/LQrFdQvc#3QFGXRqqUqf6gzsb7AMQalwRzgk2z-smxSk1g6cwXUP
> 
> Device: kok_volume_control
> 
> I pressed the button (toggle) once.

Hi, do you have the logs for Koen? I would debug myself, but I'm not at since 2 weeks and the coming week.

mvjt, 2020-11-06:
Sorry for the delay. I hope this is sufficient.
https://hastebin.com/uxesuwipal.yaml

mvjt, 2020-11-06:
I hope this is enough. I just pressed one time.
https://hastebin.com/soyivelewo.yaml

Koenkk, 2020-11-07:
Thanks, I see some possibilities to fix this. But first: does this only happen for the toggle command?

ASNNetworks, 2020-11-07:
Yes, I see it twice, also when double/triple tapping + using the dial (scrubbing). 

https://hastebin.com/ololiponac.apache

mvjt, 2020-11-08:
> Yes, I see it twice, also when double/triple tapping + using the dial (scrubbing).
> 
> https://hastebin.com/jibibaviwi.apache

I can confirm, it is double messages all the time...

Koenkk, 2020-11-08:
Should be fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

ASNNetworks, 2020-11-08:
@Koenkk is it possible to switch to dev branch and stable back and forth, without having to pair everything? Using the HA add-one (Supervisor). Or will the dev branch overwrite stable when I install and start that?

mvjt, 2020-11-08:
@Koenkk  works great thanks a lot!!

Koenkk, 2020-11-09:
@ASNNetworks yes that is possible. Good that it is fixed!

closed 2020-11-09 by Koenkk