Koenkk/zigbee2mqtt #4641: Bitron Video thermostat (AV2010/32) missing convertors

Underknowledge, 2020-10-11:
## What happened
I just tried to integrate the Bitron Video thermostat (AV2010/32) with hasska. but apparently `/set/system_mode`cant get rendered as well as the action_topic (system_mode/running_state?) so the thermostat always got a unknown state and cant be set via the external api call. ( answer, does not react)
[Devices page](https://www.zigbee2mqtt.io/devices/AV2010_32.html)

```bash
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 13:17:09: Received MQTT message on 'lights/thermostat/set/occupied_heating_setpoint' with data '18.0'
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 13:17:09: Publishing 'set' 'occupied_heating_setpoint' to 'thermostat'
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 13:17:23: Received MQTT message on 'lights/thermostat/set/occupied_heating_setpoint' with data '18.5'
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 13:17:23: Publishing 'set' 'occupied_heating_setpoint' to 'thermostat'
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 13:17:24: Received MQTT message on 'lights/thermostat/set/occupied_heating_setpoint' with data '22.5'
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 13:17:24: Publishing 'set' 'occupied_heating_setpoint' to 'thermostat'
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 13:17:25: Received MQTT message on 'lights/thermostat/set/system_mode' with data 'heat'
zigbee2mqtt          | Zigbee2MQTT:error 2020-10-11 13:17:25: No converter available for 'system_mode' (heat)
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 13:55:55: Received MQTT message on 'lights/thermostat/set/system_mode' with data 'off'
zigbee2mqtt          | Zigbee2MQTT:error 2020-10-11 13:55:55: No converter available for 'system_mode' (off)
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 13:55:57: Received MQTT message on 'lights/thermostat/set/system_mode' with data 'auto'
zigbee2mqtt          | Zigbee2MQTT:error 2020-10-11 13:55:57: No converter available for 'system_mode' (auto)
zigbee2mqtt          | Zigbee2MQTT:debug 2020-10-11 14:03:30: Received MQTT message on 'lights/thermostat/set/running_state' with data ''
zigbee2mqtt          | Zigbee2MQTT:error 2020-10-11 14:03:30: No converter available for 'set' 'running_state' ()
#no awnser on base_topic/device/running_state
```
## What did you expect to happen
system_mode/running_state should be fed back to MQTT
## How to reproduce it (minimal and precise)
Pair a AV2010/32
## Debug info
Zigbee2MQTT version:
Adapter hardware: CC1352P-2
Adapter firmware version: revision:20191106 type:zStack3x0

Koenkk, 2020-10-11:
Try adding `tz.thermostat_system_mode` to the `toZigbee` list in `devices.js` and see if it works.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

Underknowledge, 2020-10-11:
Works! damn awesome!

closed 2020-10-12 by Koenkk