Koenkk/zigbee2mqtt #4643: Handling re-transmitted Xiaomi message causes an error

PanJ, 2020-10-11:
## What happened

I have a group of devices (bulbs) located further away from the coordinator, having light bulbs as routers. After updating to 1.15.0, I have encountered the problem that the bulbs in the group are unresponsive to command. With further investigation, I found the following log that might be related to the issue.

```
Zigbee2MQTT:debug 2020-10-12 00:12:17: Received Zigbee message from 'lights.bed.6', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 31998
Zigbee2MQTT:debug 2020-10-12 00:12:17: Handling re-transmitted Xiaomi message
Zigbee2MQTT:error 2020-10-12 00:12:17: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'constructor' of undefined
    at Zigbee.resolveEntity (/zigbee2mqtt-1.15.0/lib/zigbee.js:165:17)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.15.0/lib/extension/receive.js:112:42)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.15.0/lib/controller.js:371:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.15.0/lib/controller.js:260:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.15.0/lib/zigbee.js:68:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/controller/controller.js:544:22)
    at Generator.next (<anonymous>)
    at /zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/controller/controller.js:27:71)
```

I also found that controlling the device individually works as expected. Only group control is unresponsive.

## What did you expect to happen

The group of bulbs should respond to the toggle command.

## How to reproduce it (minimal and precise)

My Setting:
- Xiaomi Bulbs
- A group of bulbs further from the coordinator (having router in between)

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: latest source routing firmware

PanJ, 2020-10-11:
It is also weird that a light bulb (ZNLDP12LM) emits a message about occupancy (I have a motion sensor RTCGQ11LM in the network)

PanJ, 2020-10-12:
I try to debug it today and found that the `Handling re-transmitted Xiaomi message` message is gone. So the message may not be relevant to the issue. I enabled the herdsman-zigbee logs and this is what it said when I tried to send a command to the light group

```
Zigbee2MQTT:debug 2020-10-12 22:55:59: Received MQTT message on 'zigbee2mqtt/lights.bed/set' with data '{"state": "ON", "brightness": 227}'
Zigbee2MQTT:debug 2020-10-12 22:55:59: Publishing 'set' 'brightness' to 'lights.bed'
2020-10-12T15:55:59.793Z zigbee-herdsman:controller:group Command 2 genLevelCtrl.moveToLevelWithOnOff({"level":227,"transtime":0})
2020-10-12T15:55:59.795Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000002","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":8,"transid":3,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[17,4,4,227,0,0]}}
2020-10-12T15:55:59.795Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,1,2,0,0,0,0,0,0,0,255,0,0,1,8,0,3,0,30,6,0,17,4,4,227,0,0,32]
2020-10-12T15:55:59.807Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2020-10-12T15:55:59.807Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2020-10-12T15:55:59.807Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2020-10-12T15:55:59.807Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2020-10-12T15:55:59.807Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-12T15:55:59.813Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,3,197]
2020-10-12T15:55:59.813Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,3,197]
2020-10-12T15:55:59.813Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,3] - 197
2020-10-12T15:55:59.813Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":3}
2020-10-12T15:55:59.813Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Koenkk, 2020-10-12:
Does the group only contain Xiaomi bulbs, if so can you add a non-Xiaomi bulb and see if it works?

PanJ, 2020-10-12:
@Koenkk Thanks for the reply! 

The group contains only Xiaomi bulbs. Unfortunately, I do not have any non-Xiaomi bulb to try. I just power-cycled the intermediate router devices and it is now working as expected. Not sure what was the problem... Maybe the routing table state is corrupted perhaps (Just my guess, I do not have any Zigbee basics). Closing the issue for now.

closed 2020-10-12 by PanJ