Koenkk/zigbee2mqtt #4648: Lights group toggle doesn't work properly

rsimiciuc, 2020-10-12:
## What happened
I have created a group of 3 lights. When I send the toggle command, I want all lights in the group to be in sync (ON or OFF)

## What did you expect to happen
If one light is individually turned OFF (the rest are ON), then when I toggle the group, the rest go OFF and the one that was OFF goes ON. Zigbee2mqtt reports wrong status as well for the one that is async
I've tried with both `optimistic: true/false`, but it doesn't seem to make any difference
## How to reproduce it (minimal and precise)
```
Lights are OFF -> turning them on via group state ON
Oct 12 19:32:59 zigbee2mqtt/all_ikea_lights/set {"state": "toggle"}
Oct 12 19:33:00 zigbee2mqtt/kitchen_light {"brightness":25,"color_temp":273,"linkquality":42,"state":"ON","update_available":false}
Oct 12 19:33:00 zigbee2mqtt/all_ikea_lights {"brightness":25,"color_temp":273,"state":"ON"}
Oct 12 19:33:00 zigbee2mqtt/living_room_light {"brightness":25,"color_temp":273,"linkquality":63,"state":"ON","update_available":false}
Oct 12 19:33:00 zigbee2mqtt/ambient_light {"brightness":25,"color_temp":273,"linkquality":39,"state":"ON","update_available":false}
All lights are ON -> turning kitchen light OFF individually
Oct 12 19:33:30 zigbee2mqtt/kitchen_light/set {"state": "toggle"}
Oct 12 19:33:30 zigbee2mqtt/kitchen_light {"brightness":25,"color_temp":273,"linkquality":42,"state":"OFF","update_available":false}
Toggle all lights       
Oct 12 19:34:07 zigbee2mqtt/all_ikea_lights/set {"state": "toggle"}
Oct 12 19:34:07 zigbee2mqtt/kitchen_light {"brightness":25,"color_temp":273,"linkquality":42,"state":"OFF","update_available":false}
Oct 12 19:34:07 zigbee2mqtt/all_ikea_lights {"brightness":25,"color_temp":273,"state":"OFF"}
Oct 12 19:34:07 zigbee2mqtt/living_room_light {"brightness":25,"color_temp":273,"linkquality":63,"state":"OFF","update_available":false}
Oct 12 19:34:07 zigbee2mqtt/ambient_light {"brightness":25,"color_temp":273,"linkquality":39,"state":"OFF","update_available":false}
State says all OFF, but Kitchen light acctualy toggled and it's ON, it can be verified with querying the state
Oct 12 19:34:45 zigbee2mqtt/kitchen_light/get {"state":""}
Oct 12 19:34:45 zigbee2mqtt/kitchen_light {"brightness":25,"color_temp":273,"linkquality":44,"state":"ON","update_available":false}
```

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: latest

sjorge, 2020-10-12:
on/off/toggle are different zigbee commands, toggle is literally switch to whatever you are not.

So that behavior is expected. Using a remote, say the ikea puck vs the on/off remote has the same behavior when using the puck.

For state I think if 1/n have a state of on the group will have the state of on, so I think that is also expected behavior.

MattWestb, 2020-10-12:
If you have the IKEA "puck" long press the toggle is sending one reset command to the binded group and all devices is in sync (if not messing the default behaviour with hacks).
If using HA (or node red) you can making one software toggle that is sending on / off to the groupe then clicking and all is in sync.

rsimiciuc, 2020-10-12:
Ok, maybe the behaviour is expected (it says the opposite on the documentation for `optimistic: true`), but the reported state back of the light is wrong then.  
You can see on this line that the reported state is OFF when the light in fact is OFF
````
Oct 12 19:34:07 zigbee2mqtt/kitchen_light {"brightness":25,"color_temp":273,"linkquality":42,"state":"OFF","update_available":false}
````

rsimiciuc, 2020-11-12:
This seems like a bug to me. The state is reported back as OFF, when in fact the state of the light is ON

closed 2020-12-21 by github-actions[bot]