Koenkk/zigbee2mqtt #4649: Home Assistant discovery

cody82, 2020-10-12:
I followed the guide at https://www.zigbee2mqtt.io/integration/home_assistant.html to integrate with Home Assistant. I set broker to localhost. I installed the MQTT integration in Home Assistant.
My Zigbee devices already work great with zigbee2mqtt and Node Red.
But no device shows up in Home Assisstant.

What could be wrong? Seems the guide is not idiot proof.

maxiwolleb, 2020-10-13:
could you provide your config from the mqtt broker in HA and your config from the zigbee2mqtt addon ?

cody82, 2020-10-13:
Wait. So I dont only need the MQTT integration but also the HA zigbee2mqtt addon?
The linked documentation says
> NOTE 2: If you are using the Zigbee2MQTT Hass.io add-on use their documentation

So I thought this is not needed or optional?

maxiwolleb, 2020-10-13:
You need to install the Zigbee2mqtt addon and connect it to the HA mqtt broker

maxiwolleb, 2020-10-13:
No Problem, if you have any further question, feel free to ask

cody82, 2020-10-13:
Ok, I was not able to install hassio-zigbee2mqtt. Although this is not a zigbee2mqtt issue, maybe you can still help.
My problem: I learned that home assistant is not hass.io. I installed home assistant on a linux machine just like zigbee2mqtt. But the addon store from which I should install hassio-zigbee2mqtt is not available in home assistant. It is hass.io exclusive.
I dont even know if hassio-zigbee2mqtt is just an adapter to connect to zigbee2mqtt via mqtt or if it embeds zigbee2mqtt.

So my real question is:
Is it possible at all what I try to do? Currently I run zigbee2mqtt, mosquitto and node red on a linux box.
Now I want to add home assistant and use the devices that are configured in zigbee2mqtt.
I dont want to use hass.io and an embedded zigbee2mqtt or mqtt broker.

It looks like using home assistant instead of hass.io is really bad documented or not supported.

insipiens, 2020-10-14:
@cody82 You mean home assistant core, since the renaming? They are the same thing just the direction of travel for HA to progress out of beta is to make it more user friendly, hence the docker version etc.

You’re not wrong...the core version is for bad ass, hard core programming nerds who like to get down and dirty with the OS etc. ;)

cody82, 2020-10-14:
@insipiens Yes I installed home assistant core, although the command was 
`pip3 install homeassistant`
So it was not clear to me.

I already installed MQTT integration but devices did not show up(should they?).

My zigbee2mqtt configuration.yaml:
```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyUSB1
devices:
...
```

Home Assistant configuration.yaml:
```
default_config:

# Text to speech
tts:
  - platform: google_translate

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

mqtt:
  discovery: true
  broker: localhost  # Remove if you want to use builtin-in MQTT broker
  # birth_message and will_message is not required anymore for Home Assistant 0.113 >=
  #birth_message:
  #  topic: 'hass/status'
  #  payload: 'online'
  #will_message:
  #  topic: 'hass/status'
  #  payload: 'offline'```

maxiwolleb, 2020-10-14:
for me localhost as server doesent work, i had to set the ip of mqtt server to a static ip

insipiens, 2020-10-14:
Easiest to set it up through the ui - integration/mqtt/configure/reconfigure

You should see this?

[image]

cody82, 2020-10-14:
I reconfigured the broker to 127.0.0.1 via gui and restarted HA.
Now it works, great! I think it really does not allow a hostname which is strange.
Thank you all!

closed 2020-10-14 by cody82