Koenkk/zigbee2mqtt #4650: Zigbee mesh collapses - SREQ '--> AF - dataRequestExt - 0x10: MEM_ERROR

Vertikar, 2020-10-12:
## What happened
Zigbee mesh network failed after installing Telstra Zigbee smart plugs (|Sercomm Telstra smart plug (SZ-ESW01-AU))
Error in logs:
First occurence:
```
Oct 12 00:39:50 mele docker[1930]: (node:17) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":200,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,35,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')

```
Most recent occurrence:
```
Oct 12 16:19:51 mele docker[1930]: (node:17) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":92,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,97,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
Oct 12 16:19:51 mele docker[1930]:     at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
Oct 12 16:19:51 mele docker[1930]:     at Generator.next (<anonymous>)
Oct 12 16:19:51 mele docker[1930]:     at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
Oct 12 16:19:51 mele docker[1930]: (node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 95)
```

## How to reproduce it (minimal and precise)
Pair or turn on Sercomm Telstra smart plug (SZ-ESW01-AU)
Unsure on how long it takes after turning device on before crash
## Debug info

Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: 20190619
{"commit":"654817a","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.14.4"}

Koenkk, 2020-10-13:
Try with the latest Zigbee2MQTT version (1.15.0) first, contains some fixes related to crashes for CC2531.

Vertikar, 2020-10-13:
Will do, it normally crashes within 12 hours or so.

Vertikar, 2020-10-19:
Latest release seems to have fixed it, thanks for that.

closed 2020-10-19 by Vertikar