Koenkk/zigbee2mqtt #4651: TS0203 Door Sensor Missing Converter

sloopee, 2020-10-13:
## What happened
After adding the TS0203 Tuya door sensor, zigbee2mqtt recognises the sensor and manages to obtain link quality and battery information only but not the critical "contact" state. The debug messages however show that zigbee2mqtt is aware of the state changes but is unable to convert the " zone status" message into an mqtt message. It says converter not available. I subsequently tried adding the manual configuration lines into my home assistant's configuration.yaml which did not help since the mqtt message still did not show the contact status (only {"battery":80,"linkquality":65,"voltage":3000}). I've attached the debug log below in the hastebin link.
## What did you expect to happen
The Tuya door sensor should have been able to publish its contact status onto the mqtt topic.
## How to reproduce it (minimal and precise)
Install home assistant, zigbee2mqtt, zigbee2mqtt assistant and pair the TS0203 sensor. Added the manual configuration lines as mentioned here with the correct friendly names and the problem still occurs. (https://www.zigbee2mqtt.io/devices/TS0203.html)
## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: CC2531_SOURCE_ROUTING_20190619
[https://hastebin.com/eqavugewok.apache](url)

Koenkk, 2020-10-13:
Should be fixed now.

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sloopee, 2020-10-14:
Thanks, i'll definitely try it out when it gets to the non-edge version since I don't know how to go back and forth between versions without reconfiguring everything!

sloopee, 2020-11-03:
It works! Thanks Koenkk

closed 2020-11-03 by Koenkk

freitas666, 2021-05-20:
@Koenkk could you have the converter for that?
 
I saw on cr_smart_home.js.... bu actually looks like not work good here.

Some ideia about

freitas666, 2021-05-20:
@sloopee could you share your converter?