Koenkk/zigbee2mqtt #4653: Sonoff SNZB-02 wrong identified as SNZB-04

vzagaevschi, 2020-10-13:
I tried to connect Sonoff SNZB-02, but it is parring as a Sonoff SNZB-04. Before that I parred Sonoff SNZB-04 and it was identified correctly.
I used the latest version of Zigbee2MQTT in the docker. In dev version received the same result.

Database extract:
{"id":16,"type":"EndDevice","ieeeAddr":"0x00124b002228bf20","nwkAddr":40355,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"DS01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"DS01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20200718"}}},"binds":[],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20200718","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1602585337903}

Zigbee2MQTT version: 1.15.0-dev
Adapter hardware: Conbee II
Adapter firmware version: 0x26580700

Koenkk, 2020-10-13:
Fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

vzagaevschi, 2020-10-14:
Now it works. Thanks.

closed 2020-10-14 by vzagaevschi