Koenkk/zigbee2mqtt #4655: TS0216. Audible Alarm. Sound and Flash Siren

kirovilya, 2020-10-13:
Siren for tuya https://aliexpress.ru/item/4001309486955.html

Database:
```{"id":6,"type":"Router","ieeeAddr":"0xbc33acfffe36d751","nwkAddr":32814,"manufId":4098,"manufName":"_TYZB01_a5tf0bpl","powerSource":"Mains (single phase)","modelId":"TS0216","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,10,1282,1280,61185],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"3���6�Q[�\u0018'\u0013��\u0018'\u0013","65506":19,"modelId":"TS0216","manufacturerName":"_TYZB01_a5tf0bpl","powerSource":4,"zclVersion":3,"appVersion":86,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0014b554eb","zoneState":0}},"ssIasWd":{"attributes":{"2":0,"61440":0,"maxDuration":60}},"genPowerCfg":{"attributes":{"batteryVoltage":50,"batteryPercentageRemaining":200}}},"binds":[],"meta":{}}},"appVersion":86,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1602614926897}```

[image]

But the interview doesn't complete:
```
2020-10-13 21:35:02.102  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:controller:endpoint Read 0xbc33acfffe36d751/1 ssIasZone(["iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-10-13 21:35:02.103  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal %s:%i/%i (%i,%i,%i)
2020-10-13 21:35:02.103  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":32814,"destendpoint":1,"srcendpoint":1,"clusterid":1280,"transid":42,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[16,35,0,16,0,0,0]}}
2020-10-13 21:35:02.103  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,46,128,1,1,0,5,42,0,30,7,16,35,0,16,0,0,0,143]
2020-10-13 21:35:02.105  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-10-13 21:35:02.105  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-10-13 21:35:02.105  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-10-13 21:35:02.105  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-10-13 21:35:02.105  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-13 21:35:02.109  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,42,236]
2020-10-13 21:35:02.109  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,42,236]
2020-10-13 21:35:02.109  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,42] - 236
2020-10-13 21:35:02.109  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":42}
2020-10-13 21:35:02.109  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-13 21:35:02.113  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,46,128,0,44]
2020-10-13 21:35:02.113  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,46,128,0,44]
2020-10-13 21:35:02.113  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [46,128,0] - 44
2020-10-13 21:35:02.113  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":32814,"relaycount":0,"relaylist":[]}
2020-10-13 21:35:02.113  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-13 21:35:02.162  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,40,68,129,0,0,0,5,46,128,1,1,0,160,0,99,15,32,0,0,20,24,35,1,16,0,0,240,235,84,181,20,0,75,18,0,0,0,0,48,0,46,128,29,160]
2020-10-13 21:35:02.163  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,40,68,129,0,0,0,5,46,128,1,1,0,160,0,99,15,32,0,0,20,24,35,1,16,0,0,240,235,84,181,20,0,75,18,0,0,0,0,48,0,46,128,29,160]
2020-10-13 21:35:02.163  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 40 - 2 - 4 - 129 - [0,0,0,5,46,128,1,1,0,160,0,99,15,32,0,0,20,24,35,1,16,0,0,240,235,84,181,20,0,75,18,0,0,0,0,48,0,46,128,29] - 160
2020-10-13 21:35:02.163  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":32814,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":160,"securityuse":0,"timestamp":2101091,"transseqnumber":0,"len":20,"data":{"type":"Buffer","data":[24,35,1,16,0,0,240,235,84,181,20,0,75,18,0,0,0,0,48,0]}}
2020-10-13 21:35:02.163  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":35,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":16,"status":0,"dataType":240,"attrData":"0x00124b0014b554eb"},{"attrId":0,"status":0,"dataType":48,"attrData":0}]},"address":32814,"endpoint":1,"linkquality":160,"groupID":0}'
2020-10-13 21:35:02.163  - [34mdebug[39m: zigbee.0 (10180) handleMessage. {"type":"readResponse","device":{"ID":6,"_type":"Router","_ieeeAddr":"0xbc33acfffe36d751","_networkAddress":32814,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":1027,"inputClusters":[0,10,1282,1280,61185],"outputClusters":[25],"deviceNetworkAddress":32814,"deviceIeeeAddress":"0xbc33acfffe36d751","clusters":{"genBasic":{"attributes":{"modelId":"TS0216","manufacturerName":"_TYZB01_a5tf0bpl","powerSource":4,"zclVersion":3,"appVersion":86,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0014b554eb","zoneState":0}},"ssIasWd":{"attributes":{"2":0,"61440":2,"maxDuration":60}},"genPowerCfg":{"attributes":{"batteryVoltage":50,"batteryPercentageRemaining":200}}},"_binds":[],"meta":{}}],"_manufacturerName":"_TYZB01_a5tf0bpl","_powerSource":"Mains (single phase)","_modelID":"TS0216","_applicationVersion":86,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":false,"_interviewing":true,"meta":{},"_lastSeen":1602614102163},"endpoint":"[Circular]","data":{"iasCieAddr":"0x00124b0014b554eb","zoneState":0},"linkquality":160,"groupID":0,"cluster":"ssIasZone","meta":{"zclTransactionSequenceNumber":35,"manufacturerCode":null,"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0}}}
2020-10-13 21:35:02.163  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-13 21:35:02.163  - [34mdebug[39m: zigbee.0 (10180) Received Zigbee message from '0xbc33acfffe36d751', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b0014b554eb","zoneState":0}' from endpoint 1 with groupID 0
2020-10-13 21:35:02.163  - [34mdebug[39m: zigbee.0 (10180) Type readResponse device {"type":"device","device":{"ID":6,"_type":"Router","_ieeeAddr":"0xbc33acfffe36d751","_networkAddress":32814,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":1027,"inputClusters":[0,10,1282,1280,61185],"outputClusters":[25],"deviceNetworkAddress":32814,"deviceIeeeAddress":"0xbc33acfffe36d751","clusters":{"genBasic":{"attributes":{"modelId":"TS0216","manufacturerName":"_TYZB01_a5tf0bpl","powerSource":4,"zclVersion":3,"appVersion":86,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0014b554eb","zoneState":0}},"ssIasWd":{"attributes":{"2":0,"61440":2,"maxDuration":60}},"genPowerCfg":{"attributes":{"batteryVoltage":50,"batteryPercentageRemaining":200}}},"_binds":[],"meta":{}}],"_manufacturerName":"_TYZB01_a5tf0bpl","_powerSource":"Mains (single phase)","_model [... cut ...]
    at fulfilled (D:\iobroker\home\node_modules\zigbee-herdsman\dist\controller\model\device.js:24:58)'
2020-10-13 21:35:02.189  - [34mdebug[39m: zigbee.0 (10180) zigbee-herdsman:controller:error Interview failed for '0xbc33acfffe36d751 with error 'Error: Interview failed because of failed IAS enroll (zoneState didn't change ('0xbc33acfffe36d751')'
```
I will write converters, but I don’t know what to do with the interview yet.

Sniffing log with original gateway:
[tuya_siren.zip](https://github.com/Koenkk/zigbee2mqtt/files/5373501/tuya_siren.zip)

Koenkk, 2020-10-14:
Did you also try after this change: https://github.com/Koenkk/zigbee-herdsman/commit/b217b91bb723b31a73ebecfa0e80690bccf44f8b  (adds auto enrollment support)

kirovilya, 2020-10-14:
yes i tried with the latest herdsman

Koenkk, 2020-10-14:
I've check the TuYa sniff and the behaviour is the same. At No. 422 the sensor sends a enroll response (indicating is enrolled), on 424 gateway reads the zoneState, in 426 the sensor report it's not enrolled. So I think we can ignore this error if the device works properly.

kirovilya, 2020-10-14:
Yes, the device is working fine. So even zonestate == 0 is normal for some devices. 
perhaps this applies to the siren, because she is not the source of the security message... although, I don't understand much about this topic

kirovilya, 2020-10-16:
Can we disable raising exception? Only write error to log https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L518

Koenkk, 2020-10-16:
No, but you can add the modelID here, this will prevent the interview from failing: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L364 add:

```
'TS0216': {},
```

kirovilya, 2020-10-16:
hmm... 
maybe it's better to add some option to the device description that will ignore enroll ... although, herdsman does not know anything about converters

Koenkk, 2020-10-16:
have you test with this? (curious if it works)

kirovilya, 2020-10-17:
Yes. Is it work with `'TS0216': {}` 👍 
```2020-10-17 11:52:11.392  - [34mdebug[39m: zigbee.0 (11316) Received Zigbee message from '0xbc33acfffe36d751', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b001936776b","zoneState":0}' from endpoint 1 with groupID 0
2020-10-17 11:52:11.392  - [34mdebug[39m: zigbee.0 (11316) Type readResponse device {"type":"device","device":{"ID":2,"_type":"Router","_ieeeAddr":"0xbc33acfffe36d751","_networkAddress":39722,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":1027,"inputClusters":[0,10,1282,1280,61185],"outputClusters":[25],"deviceNetworkAddress":39722,"deviceIeeeAddress":"0xbc33acfffe36d751","clusters":{"genBasic":{"attributes":{"65503":"3���6�Q��\u001d'\u0013","65506":19,"modelId":"TS0216","manufacturerName":"_TYZB01_a5tf0bpl","powerSource":4,"zclVersion":3,"appVersion":86,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001936776b","zoneState":0}},"ssIasWd":{"attributes":{"2":30,"61440":0,"maxDuration":60}},"genPowerCfg":{"attributes":{"batteryVoltage":50,"batteryPercentageRemaining":200}}},"_binds":[],"meta":{}}],"_manufacturerName":"_TYZB01_a5tf0bpl","_powerSource":"Mains (single phase)","_modelID":"TS0216","_applicationVersion":86,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":false,"_interviewing":true,"meta":{},"_lastSeen":1602924731392},"mapped":{"zigbeeModel":["TS0216"],"model":"TS0216","vendor":"Tuya","description":"Sound and flash siren","supports":"alarm","fromZigbee":[{"cluster":"ssIasWd","type":["attributeReport","readResponse"]},{"cluster":"genPowerCfg","type":["attributeReport","readResponse"]}],"toZigbee":[{"key":["alarm"]},{"key":["duration"]},{"key":["volume"]},{"key":["scene_store"]},{"key":["scene_recall"]},{"key":["scene_add"]}],"meta":{"configureKey":1}},"name":"0xbc33acfffe36d751"} incoming event: {"type":"readResponse","device":{"ID":2,"_type":"Router","_ieeeAddr":"0xbc33acfffe36d751","_networkAddress":39722,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":1027,"inputClusters":[0,10,1282,1280,61185],"outputClusters":[25],"deviceNetworkAddress":39722,"deviceIeeeAddress":"0xbc33acfffe36d751","clusters":{"genBasic":{"attributes":{"65503":"3���6�Q��\u001d'\u0013","65506":19,"modelId":"TS0216","manufacturerName":"_TYZB01_a5tf0bpl","powerSource":4,"zclVersion":3,"appVersion":86,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001936776b","zoneState":0}},"ssIasWd":{"attributes":{"2":30,"61440":0,"maxDuration":60}},"genPowerCfg":{"attributes":{"batteryVoltage":50,"batteryPercentageRemaining":200}}},"_binds":[],"meta":{}}],"_manufacturerName":"_TYZB01_a5tf0bpl","_powerSource":"Mains (single phase)","_modelID":"TS0216","_applicationVersion":86,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":false,"_interviewing":true,"meta":{},"_lastSeen":1602924731392},"endpoint":"[Circular]","data":{"iasCieAddr":"0x00124b001936776b","zoneState":0},"linkquality":126,"groupID":0,"cluster":"ssIasZone","meta":{"zclTransactionSequenceNumber":40,"manufacturerCode":null,"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0}}}
2020-10-17 11:52:11.393  - [34mdebug[39m: zigbee.0 (11316) zigbee-herdsman:controller:device:log Interview - IAS - after enrolling state (19): '{"iasCieAddr":"0x00124b001936776b","zoneState":0}'
2020-10-17 11:52:11.393  - [34mdebug[39m: zigbee.0 (11316) zigbee-herdsman:controller:device:log Interview procedure failed but got modelID matching 'TS0216', assuming interview succeeded
2020-10-17 11:52:11.405  - [34mdebug[39m: zigbee.0 (11316) zigbee-herdsman:controller:device:log Interview - completed for device '0xbc33acfffe36d751' because of quirks ('Error: Interview failed because of failed IAS enroll (zoneState didn't change ('0xbc33acfffe36d751')')
2020-10-17 11:52:11.412  - [34mdebug[39m: zigbee.0 (11316) zigbee-herdsman:controller:log Succesfully interviewed '0xbc33acfffe36d751'
2020-10-17 11:52:11.412  - [34mdebug[39m: zigbee.0 (11316) handleDeviceInterview. {"status":"successful","device":{"ID":2,"_type":"Router","_ieeeAddr":"0xbc33acfffe36d751","_networkAddress":39722,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":1027,"inputClusters":[0,10,1282,1280,61185],"outputClusters":[25],"deviceNetworkAddress":39722,"deviceIeeeAddress":"0xbc33acfffe36d751","clusters":{"genBasic":{"attributes":{"65503":"3���6�Q��\u001d'\u0013","65506":19,"modelId":"TS0216","manufacturerName":"_TYZB01_a5tf0bpl","powerSource":4,"zclVersion":3,"appVersion":86,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001936776b","zoneState":0}},"ssIasWd":{"attributes":{"2":30,"61440":0,"maxDuration":60}},"genPowerCfg":{"attributes":{"batteryVoltage":50,"batteryPercentageRemaining":200}}},"_binds":[],"meta":{}}],"_manufacturerName":"_TYZB01_a5tf0bpl","_powerSource":"Mains (single phase)","_modelID":"TS0216","_applicationVersion":86,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"meta":{},"_lastSeen":1602924731392}}
2020-10-17 11:52:11.413  - [32minfo[39m: zigbee.0 (11316) Successfully interviewed '0xbc33acfffe36d751', device has succesfully been paired
2020-10-17 11:52:11.413  - [32minfo[39m: zigbee.0 (11316) Device '0xbc33acfffe36d751' is supported, identified as: Tuya Sound and flash siren (TS0216)```

closed 2020-10-17 by Koenkk

Caizzii, 2020-11-10:
Hi all,

Is it possible to trigger the alarm? Because I don't see any switch in Manual Home Assistant configuration (https://www.zigbee2mqtt.io/devices/TS0216.html). 

Thanks for your help.

ConsoleCriminal, 2020-12-03:
@Caizzii : Judging from commit https://github.com/Koenkk/zigbee-herdsman-converters/pull/1674/commits/c145065eb47c6133680e10755d16aff63a3196e3 it looks like it, but is underdocumented. (Volume, Duration....)

Polimac, 2020-12-07:
> 
> 
> Hi all,
> 
> Is it possible to trigger the alarm? Because I don't see any switch in Manual Home Assistant configuration (https://www.zigbee2mqtt.io/devices/TS0216.html).
> 
> Thanks for your help.

also interested in that, at least how to trigger default sound over mqtt

Caizzii, 2020-12-08:
Hi Polimac,

This is possible with topic : zigbee2mqtt/FRIENDLY_NAME/set and message : {"alarm": true}
You can also adjust the volume with {"volume": INTERGER}, like {"volume": 10}.

I plan to update the documentation soon.

I hope this is helpful.

Polimac, 2020-12-08:
Wow , thanks a lot 👍