Koenkk/zigbee2mqtt #4656: Can't reconnect bulb after removing it

trashmaster9000, 2020-10-13:
## What happened
tested renaming a ikea lightbulb (LED1649C5) in the UI, broke something in Homeassistant (Device not avaliable), Removed the lightbulb via UI, cant reconnect the bulb.

tried everything on the FAQ
Bulb flashes and pulsates normal after reset, but i dont get anything in the log.

## Debug info
Zigbee2MQTT version: 1.15.0, Home Assistant
Adapter hardware: CC2531, currently 7 Devices in Network

Koenkk, 2020-10-14:
- Is the device still in the `data/database.db`?
- Have you tried reflashing your CC2531?

trashmaster9000, 2020-10-14:
how do i check that? it doesnt appear in the device list in the UI

i dont have the tools to reflash the stick :/

Koenkk, 2020-10-14:
As a trick you can try to switch channel (https://www.zigbee2mqtt.io/information/configuration.html), but that requires to repair your devices.

trashmaster9000, 2020-10-15:
hmm, it seems like i cant pair any new devices...
got the philips hue play (Philips 915005733701) today, reset them via touchlink but they wont join...
the reset always fails on the first try, second try works.
the light flashes while resetting so thats working as it should?

```
Zigbee2MQTT:info  2020-10-15 20:25:16: Start Touchlink scan
Zigbee2MQTT:info  2020-10-15 20:25:28: Finished Touchlink scan
Zigbee2MQTT:info  2020-10-15 20:25:28: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{"found":[{"channel":11,"ieee_address":"0x0017880106fe210a"}]},"status":"ok"}'
Zigbee2MQTT:info  2020-10-15 20:25:30: Start Touchlink factory reset of '0x0017880106fe210a' on channel 11
Zigbee2MQTT:error 2020-10-15 20:25:34: Request 'zigbee2mqtt/bridge/request/touchlink/factory_reset' failed with error: 'Timeout - null - 254 - null - 4096 - 1 after 500ms'
Zigbee2MQTT:info  2020-10-15 20:25:34: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/factory_reset', payload '{"data":{},"error":"Timeout - null - 254 - null - 4096 - 1 after 500ms","status":"error"}'
Zigbee2MQTT:info  2020-10-15 20:25:36: Start Touchlink factory reset of '0x0017880106fe210a' on channel 11
Zigbee2MQTT:info  2020-10-15 20:25:41: Successfully factory reset device through Touchlink
Zigbee2MQTT:info  2020-10-15 20:25:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/factory_reset', payload '{"data":{"channel":11,"ieee_address":"0x0017880106fe210a"},"status":"ok"}'
```

closed 2020-11-23 by github-actions[bot]