Koenkk/zigbee2mqtt #4657: mqtt topic different from friendly name. possible?

bartplessers, 2020-10-13:
## Is your feature request related to a problem? Please describe
Till now, I always used the unique IEEE-address as 'friendly name'.
I switched now to more meaningfull names
However, I noticed that this changes also the zigbee topic of the device

## Describe the solution you'd like
I would like to have a friendly name/alias/displayname/... that I can change and that will appear in home-assistant, without modifying the underlying datastructure (zigbeetopic). 
So If I move a bulb from my kitchen to my living room and rename it accordingly (i.e. in the homeassistant GUI), I want to have my workflows/logs/automations/node-red/etc... intact. 
Database wise: a separation between technical name and display name

## Additional context
I started renaming my devices as consequence of the great feature:
https://github.com/Koenkk/zigbee2mqtt/issues/4266
Having my devices displayed with all their properties and with a meaningfull name, is very handy.
However, messing up my current setup wasn't... :-)

kind regards,
Bart

dirstel, 2020-10-22:
As we speak about mqtt some thoughts about naming topics should be done at start.

Using the IEEE Adress as Topic in some sort of automation-software (node-red in my case) seems stable, but does not work well when replacing a device.

I started using a naming scheme which allows localization and replacing. so I set the friendly names acording to this scheme:

<floor>/<room>/<type>/<location>[/<additional location>]
f.e.: 
1_ground/living/light/celing  or in german 1_eg/wohnen/licht/decke

this allows quite good sorting, replacing and interpreting the name. Maybe this will help you instead of your desired feature?!

bartplessers, 2020-10-22:
@dirstel: yes, good idea.
However, I work with zigbee2mqtt.
The message structure there is https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html

`zigbee2mqtt/[FRIENDLY_NAME]`

So, I have no influence on this.
For me the (meaninless) IEEE-adress is more stable. If

sjorge, 2020-10-24:
You can use `/` in the friendlyName, I do with a similar schema as mentioned above.

dirstel, 2020-10-25:
> @dirstel: yes, good idea.
> However, I work with zigbee2mqtt.
> The message structure there is https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html
> 
> `zigbee2mqtt/[FRIENDLY_NAME]`

Gues what: me too! I just skipped the base-topic in my examples above. Indeed I use a custom base topic (home instead of zigbee2mqtt). and the structure mentioned above as friendly name. This gives you some more mqtt-native benefits as

every device on ground level
`base/1_ground/#`

every light in my house
`base/+/+/light/#`

etc. As you are using homeassistant, that my not be that useful, but using different software like node-red for example this gives you much flexibility.

Anyway: You like to have zigbee2mqtt to use IEEE instead of friendly name even when friendly name is given, correct? That should be somewhat easier than adding a seperate topic to each and any device.

bartplessers, 2020-10-25:
@dirstel , @sjorge 

OMG this is really soooooo cool!!! How could I look over this feature.
Didn't know this was possible in zigbee2mqtt. Thanx for pointing this out.

@dirstel 
_Anyway: You like to have zigbee2mqtt to use IEEE instead of friendly name even when friendly name is given, correct? That should be somewhat easier than adding a seperate topic to each and any device._

Indeed.
If -for whatever reason- a device is renamed (i.e. in your schedule: a light is moved from one room to another), I want all my automations still like to work.
But apparently, in homeassistant, I can address all the devices with IEEE

```
- id: '0x00158d0004245877_single_left'
  alias: '(S08) single_left'
  initial_state: true
  trigger:
  - entity_id: sensor.0x00158d0004245877_action
    platform: state
    to: 'single_left'
  condition: []
  action:
  - data:
      entity_id: light.0x14b457fffe51a9f2_light
    service: light.toggle
```

although my "friendly names" are now (I added an extra level "my" to you suggestion):

```
'0x14b457fffe51a9f2':
  friendly_name: my/n00/living/light/l08
'0x00158d0004245877':
  friendly_name: my/n00/living/switch/s08
```

So with my convention above
`my/level/room/devicetype/display name`
I can do exactly what I want:
- show all messages of "my" devices, without interference with zigbee2mqtt logging: `zigbee2mqtt/my/#`
- have my automations independent from naming
- do advanced filtering

The only (small) drawback is that HomeAssistant shows initially the device as:
[image]

but this is just a minor issue and can easily be renamed in hass.

thanx thanx thanx again.

kind regards,
Bart

bartplessers, 2020-11-02:
mmm. My enthousiasm is a lot less now.
Situation:
  - I renamed all my devices:
[image]

  - Home Assistant picks up the new name (1)
[image]

  - but also changes the Entity ID (2)

This means I fully have to rewrite my automations, i.e. I have to replace all the IEEE-names by my custom ID's
[image]

So original question / feature request remains definitly!
- a "Friendly Name" (eventually with slashes to create hierarchical MQTT topic)
- an (**fixed**) ID that is used by Home Assistant

kind regards,
Bart

dirstel, 2020-11-03:
> This means I fully have to rewrite my automations, i.e. I have to replace all the IEEE-names by my custom ID's

Bart, I fully understand your trouble, but please keep in mind: This renaming is a one time effort. 
When done, you may replace a device, rename the friendly name to the one of the device you replaced, and you are done. No changes to hass or any other software referencing to the device.

Anyway: I would prefere a config option to use the IEEE instead of the friendly name as MQTT topic. This wo'nt break any existing config when default behaviour is "use friendly name as mqtt topic" and would allaow you to use friendly name as display name and IEEE as MQTT topic when configured to "use IEEE as mqtt topic".

To be honest: you posted above:
> For me the (meaninless) IEEE-adress is more stable. If I need to replace a bulb, I can live with a procedure of "search and replace" the old IEEE with the new one.

So a one time "search and replace" would do.

closed 2020-12-12 by github-actions[bot]

bartplessers, 2020-12-26:
Just FYI: 
I think I have a much more flexible solution:
1. I renamed my zigbee devices according to something like my/n00/living/light/l08
2. I created a **lightgroup** for all my devices. Even if the group only contains one light.
3. in my automations, I adress the ID of the lightgroup

By doing so, I have full flexibility:
- If I have to renew a light, I just add it to the network, eventually rename it, and finally add (or replace) it to the lightgroup
- no need to change automations, because they address the new light only indirectly

kind regards
Bart

moshed, 2022-12-13:
just putting my .02 that a feature like this would be great. Tasmota allows you to set MQTT topic independently of the friendly name which is useful for homebridge plugins that auto grab the friendly name. ie now my Ames inside of HomeKit are home/upstairs/master bedroom/overhead whereas if I could set an independent friendly name it would show up in HomeKit as Master Bedroom Overhead (and HomeKit smartly removes the room name if it is in the beginning of the name so when looking inside a my master bedroom room it would just show as overhead)
[image]
[image]
[image]

HerrSpace, 2024-06-08:
I'd also love to be able to be able to send messages to the stable ieee address and have the friendly name in the json body instead.