Koenkk/zigbee2mqtt #4660: Please add Somgoms ZSQB-SMB-ZB switch

SPEC1AL1ST, 2020-10-14:
## Information about the device + link
Please add Somgoms ZSQB-SMB-ZB switch to zigbee2mqtt
https://aliexpress.ru/item/4001133814342.html

## data/database.db entry of the device
..
Stick CC2538 not see it :(

SPEC1AL1ST, 2020-10-15:
> i have the same and it work perfect after updating zigbee2mqtt ;)
> zigbee smart switch
> 2 gang one way
> 
> https://de.aliexpress.com/item/4000298926256.html
> https://de.aliexpress.com/item/4000807168877.html
> 
> i think its all the same ;)

Nope its different switches. You show lonsonho and its pairing. Somgoms nope

SPEC1AL1ST, 2020-10-15:
> i have the same and it work perfect after updating zigbee2mqtt ;)
> zigbee smart switch
> 2 gang one way
> 
> https://de.aliexpress.com/item/4000298926256.html
> https://de.aliexpress.com/item/4000807168877.html
> 
> i think its all the same ;)

And you can tell me how you pairing that? Step by step

cyber-funk, 2020-10-30:
@SPEC1AL1ST If you do 4 presses, and then long press on the last/4th press. Do you hear beeping from this switch?

SPEC1AL1ST, 2020-10-30:
Yes

SPEC1AL1ST, 2020-10-30:
> @SPEC1AL1ST If you do 4 presses, and then long press on the last/4th press. Do you hear beeping from this switch?

Yes

closed 2020-12-07 by github-actions[bot]