Koenkk/zigbee2mqtt #4661: Orvino / A-OK AM25 blinds motor - how to set limit positions?

sanderjonkers, 2020-10-14:
## What happened
I have an A-OK branded blinds motor, type AM25, which is correctly paired to zigbee2mqtt as an Orvibo AM25 motor. Up, down and stop commands are working perfectly. However, the position command does not work, probably because the position limits need to be configured first. 
The instructions that came with the motor only have instructions to set the limit positions with an RF remote, which I don't have.

## What did you expect to happen
Is it possible to set the limit positions through zigbee2mqtt? In the home assistant addon I don't see this option, but maybe it's possible by sending some MQTT messages?

The seller of the motor told me that setting the limit positions is possible with their zigbee gateway and app, which I obviously don't have.

## How to reproduce it (minimal and precise)
Not relevant.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: 20190608

aokhrimenko, 2020-11-19:
@sanderjonkers have you found a solution on how to configure it without using the RF?

sanderjonkers, 2020-11-19:
@aokhrimenko unfortunately not. I ended up buying an RF remote to set the limit positions. A remote wasn't very expensive, I just had to wait a few weeks for shipping.

aokhrimenko, 2020-11-19:
@sanderjonkers maybe you can record the rf packets?
Don't want to to waist the time waiting for rf remote ;)

sanderjonkers, 2020-11-19:
I do have an RFXcom somewhere, so I could do that yes. Do you have a device to send RF signals with?

aokhrimenko, 2020-11-20:
@sanderjonkers thanks, but I don't have it. I've decided to buy the remote too.

closed 2020-12-28 by github-actions[bot]