Koenkk/zigbee2mqtt #4662: After reconnection: Error: SREQ '--> SYS - osalNvWrite

golddragon007, 2020-10-14:
## What happened
I've debugged if I need an extension cable or not. Therefore I've connected and removed the USB Zigbee without stoping it.
```
> zigbee2mqtt@1.15.0 start /zigbee2mqtt-1.15.0
> node index.js
Zigbee2MQTT:info  2020-10-14 15:44:44: Logging to console and directory: 'data/log/2020-10-14.15-44-44' filename: log.txt
Zigbee2MQTT:debug 2020-10-14 15:44:44: Removing old log directory 'data/log/2020-10-14.15-43-39'
Zigbee2MQTT:debug 2020-10-14 15:44:45: Loaded state from file /share/zigbee2mqtt/state.json
Zigbee2MQTT:info  2020-10-14 15:44:45: Starting Zigbee2MQTT version 1.15.0 (commit #1.15.0)
Zigbee2MQTT:info  2020-10-14 15:44:45: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-14 15:44:45: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0"}}'
2020-10-14T13:44:45.890Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2020-10-14T13:44:45.894Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-10-14T13:44:45.901Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-10-14T13:44:45.929Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-10-14T13:44:45.951Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0]
2020-10-14T13:44:45.953Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0]
2020-10-14T13:44:46.035Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,65,128,0,2,1,2,7,1,192]
2020-10-14T13:44:46.036Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,254,6,65,128,0,2,1,2,7,1,192]
2020-10-14T13:44:46.038Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 2 - 1 - 128 - [0,2,1,2,7,1] - 192
2020-10-14T13:44:46.043Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- SYS - resetInd - {"reason":0,"transportrev":2,"productid":1,"majorrel":2,"minorrel":7,"hwrev":1}
2020-10-14T13:44:46.047Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-14T13:44:46.259Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [0]
2020-10-14T13:44:46.260Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0]
2020-10-14T13:44:46.347Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,65,128,0,2,1,2,7,1,192]
2020-10-14T13:44:46.347Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,254,6,65,128,0,2,1,2,7,1,192]
2020-10-14T13:44:46.348Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 2 - 1 - 128 - [0,2,1,2,7,1] - 192
2020-10-14T13:44:46.348Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- SYS - resetInd - {"reason":0,"transportrev":2,"productid":1,"majorrel":2,"minorrel":7,"hwrev":1}
2020-10-14T13:44:46.348Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-14T13:44:46.403Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-10-14T13:44:46.403Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-10-14T13:44:47.408Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-10-14T13:44:47.410Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-10-14T13:44:47.415Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61]
2020-10-14T13:44:47.415Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61]
2020-10-14T13:44:47.415Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61
2020-10-14T13:44:47.416Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625}
2020-10-14T13:44:47.417Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-14T13:44:47.418Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2020-10-14T13:44:47.418Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2020-10-14T13:44:47.423Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,1,2,7,1,221,61,52,1,0,255,255,255,255,191]
2020-10-14T13:44:47.423Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,1,2,7,1,221,61,52,1,0,255,255,255,255,191]
2020-10-14T13:44:47.423Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,1,2,7,1,221,61,52,1,0,255,255,255,255] - 191
2020-10-14T13:44:47.424Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200925}
2020-10-14T13:44:47.424Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-14T13:44:47.425Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16
2020-10-14T13:44:47.425Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200925})
2020-10-14T13:44:47.428Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2020-10-14T13:44:47.430Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74]
2020-10-14T13:44:47.434Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,8,2,0,105]
2020-10-14T13:44:47.434Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,8,2,0,105]
2020-10-14T13:44:47.435Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 8 - [2,0] - 105
2020-10-14T13:44:47.436Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":2,"len":0,"value":{"type":"Buffer","data":[]}}
2020-10-14T13:44:47.436Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-14T13:44:47.437Z zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is invalid, got '{"type":"Buffer","data":[]}', expected '{"type":"Buffer","data":[85]}'
2020-10-14T13:44:47.437Z zigbee-herdsman:adapter:zStack:startZnp Restoring coordinator from backup
2020-10-14T13:44:47.441Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {}
2020-10-14T13:44:47.441Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35]
2020-10-14T13:44:47.446Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,14,97,2,2,1,2,7,1,221,61,52,1,0,255,255,255,255,191]
2020-10-14T13:44:47.446Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,14,97,2,2,1,2,7,1,221,61,52,1,0,255,255,255,255,191]
2020-10-14T13:44:47.446Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 14 - 3 - 1 - 2 - [2,1,2,7,1,221,61,52,1,0,255,255,255,255] - 191
2020-10-14T13:44:47.447Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200925}
2020-10-14T13:44:47.447Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-14T13:44:47.449Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvWrite - {"id":1,"offset":0,"osal":true,"product":-1,"value":[228,9,204,33,0,75,18,0],"len":8}
2020-10-14T13:44:47.450Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,12,33,9,1,0,0,8,228,9,204,33,0,75,18,0,116]
2020-10-14T13:44:47.455Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,97,9,2,107]
2020-10-14T13:44:47.455Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,97,9,2,107]
2020-10-14T13:44:47.455Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 1 - 9 - [2] - 107
2020-10-14T13:44:47.456Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvWrite - {"status":2}
2020-10-14T13:44:47.456Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... 5 lines cut ...]
[... cut ...]
2020-10-14T15:44:47: PM2 log: App [npm:0] online
```
## What did you expect to happen
To not broke. USB devices can anytime be disconnected by specs, so it needs to be resistant such a behaviour (even if in our case is pretty rare).
## How to reproduce it (minimal and precise)
Start z2m then randomly remove and connected back (I've added extension cable under that time) and after reconnection, I've almost immediately sent a Zigbee signal.
## Debug info
Zigbee2MQTT version: newest
Adapter hardware: CC26X2R1
Adapter firmware version: newest

Koenkk, 2020-10-14:
Looks like some corruption occurred in the coordinator firmware, did you have to reflash to make it work again?

I've tried to reproduce but failed to do so, could you provide steps to reproduce this issue? Since we cannot solve this issue from Zigbee2MQTT I need to provide clear steps to Texas Instruments to fix this in the firmware.

golddragon007, 2020-10-14:
Yes, a firmware flash solved it.
I did the following:
1. Coordinator run with z2m
2. Unplug coordinator without stopping z2m
3. Plug it back with an extension cable
4. Send commands within 1-3 seconds after plugin and repeated it around 5 seconds because it didn't notice.

Then I've viewed the log and I saw it's broken. It can be that when I unplugged it, it's communicated at that moment and that's caused the issue.

Koenkk, 2020-10-15:
After unplugging the coordinator Zigbee2MQTT should immediately stop (which should prevent 3 and 4 occurring) Does this happen in your case? Should look like this:

[image]

closed 2020-11-23 by github-actions[bot]