Koenkk/zigbee2mqtt #4663: Multiple messages from a single device

golddragon007, 2020-10-14:
## What happened
I've pressed a button on my zigbee device and I've noticed in the logs that 5 message was sent.
```
2020-10-14T14:04:30.999Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,23,68,129,0,0,6,0,90,159,1,47,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,41]
2020-10-14T14:04:30.999Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,68,129,0,0,6,0,90,159,1,47,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,41]
2020-10-14T14:04:30.999Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,90,159,1,47,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29] - 41
2020-10-14T14:04:30.999Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":40794,"srcendpoint":1,"dstendpoint":47,"wasbroadcast":1,"linkquality":90,"securityuse":0,"timestamp":6243738,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[17,6,1]}}
2020-10-14T14:04:31.000Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":1},"Payload":{}},"address":40794,"endpoint":1,"linkquality":90,"groupID":0}'
2020-10-14T14:04:31.005Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2020-10-14 16:04:31: Received Zigbee message from '0x04cf8cdf3c77c750', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-14 16:04:31: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77c750', payload '{"action":"button_2_single","battery":100,"linkquality":90,"voltage":3100}'
Zigbee2MQTT:info  2020-10-14 16:04:31: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77c750', payload '{"action":"","battery":100,"linkquality":90,"voltage":3100}'
Zigbee2MQTT:info  2020-10-14 16:04:31: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77c750/action', payload 'button_2_single'
2020-10-14T14:04:31.016Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,23,68,129,0,0,6,0,90,159,1,13,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,11,254,23,68,129,0,0,6,0,90,159,1,110,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,104,254,23,68,129,0,0,6,0,90,159,1,11,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,13,254,23,68,129,0,0,6,0,90,159,1,8,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,14,254,23,68,129,0,0,6,0,90,159,1,1,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,7]
2020-10-14T14:04:31.016Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,68,129,0,0,6,0,90,159,1,13,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,11,254,23,68,129,0,0,6,0,90,159,1,110,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,104,254,23,68,129,0,0,6,0,90,159,1,11,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,13,254,23,68,129,0,0,6,0,90,159,1,8,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,14,254,23,68,129,0,0,6,0,90,159,1,1,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,7]
2020-10-14T14:04:31.017Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,90,159,1,13,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29] - 11
2020-10-14T14:04:31.017Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":40794,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":1,"linkquality":90,"securityuse":0,"timestamp":6243738,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[17,6,1]}}
2020-10-14T14:04:31.019Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":1},"Payload":{}},"address":40794,"endpoint":1,"linkquality":90,"groupID":0}'
Zigbee2MQTT:debug 2020-10-14 16:04:31: Received Zigbee message from '0x04cf8cdf3c77c750', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-14 16:04:31: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77c750', payload '{"action":"button_2_single","battery":100,"linkquality":90,"voltage":3100}'
Zigbee2MQTT:debug 2020-10-14 16:04:31: Received Zigbee message from '0x04cf8cdf3c77c750', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-14 16:04:31: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77c750', payload '{"action":"button_2_single","battery":100,"linkquality":90,"voltage":3100}'
2020-10-14T14:04:31.025Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,68,129,0,0,6,0,90,159,1,110,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,104,254,23,68,129,0,0,6,0,90,159,1,11,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,13,254,23,68,129,0,0,6,0,90,159,1,8,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,14,254,23,68,129,0,0,6,0,90,159,1,1,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,7]
2020-10-14T14:04:31.025Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,90,159,1,110,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29] - 104
2020-10-14T14:04:31.025Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":40794,"srcendpoint":1,"dstendpoint":110,"wasbroadcast":1,"linkquality":90,"securityuse":0,"timestamp":6243738,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[17,6,1]}}
2020-10-14T14:04:31.027Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":1},"Payload":{}},"address":40794,"endpoint":1,"linkquality":90,"groupID":0}'
2020-10-14T14:04:31.033Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,68,129,0,0,6,0,90,159,1,11,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,13,254,23,68,129,0,0,6,0,90,159,1,8,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,14,254,23,68,129,0,0,6,0,90,159,1,1,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,7]
2020-10-14T14:04:31.034Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,90,159,1,11,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29] - 13
2020-10-14T14:04:31.034Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":40794,"srcendpoint":1,"dstendpoint":11,"wasbroadcast":1,"linkquality":90,"securityuse":0,"timestamp":6243738,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[17,6,1]}}
Zigbee2MQTT:debug 2020-10-14 16:04:31: Received Zigbee message from '0x04cf8cdf3c77c750', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
2020-10-14T14:04:31.035Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":1},"Payload":{}},"address":40794,"endpoint":1,"linkquality":90,"groupID":0}'
Zigbee2MQTT:info  2020-10-14 16:04:31: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77c750', payload '{"action":"button_2_single","battery":100,"linkquality":90,"voltage":3100}'
Zigbee2MQTT:debug 2020-10-14 16:04:31: Received Zigbee message from '0x04cf8cdf3c77c750', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-14 16:04:31: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77c750', payload '{"action":"button_2_single","battery":100,"linkquality":90,"voltage":3100}'
2020-10-14T14:04:31.040Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,68,129,0,0,6,0,90,159,1,8,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,14,254,23,68,129,0,0,6,0,90,159,1,1,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29,7]
2020-10-14T14:04:31.041Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 4 - 129 - [0,0,6,0,90,159,1,8,1,90,0,154,69,95,0,0,3,17,6,1,90,159,29] - 14
2020-10-14T14:04:31.042Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":40794,"srcendpoint":1,"dstendpoint":8,"wasbroadcast":1,"linkquality":90,"securityuse":0,"timestamp":6243738,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[17,6,1]}}
2020-10-14T14:04:31.043Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandI [... cut ...]
Zigbee2MQTT:info  2020-10-14 16:04:31: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77c750/action', payload 'button_2_single'
Zigbee2MQTT:info  2020-10-14 16:04:31: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c77c750/action', payload 'button_2_single'
```
## What did you expect to happen
I suppose only one message needs to be sent.
## How to reproduce it (minimal and precise)
Maybe by pressing a button on an aqara opple switch assuming this is a global issue.
## Debug info
Zigbee2MQTT version: newest
Adapter hardware: CC26X2R1
Adapter firmware version: newest

Koenkk, 2020-10-14:
Can you make a sniff of this? Looking at the sniff it seems that the device indeed produces multiple messages.

Shaquu, 2020-10-14:
It looks like I have the same with mine TS0011 (I believe also 5 times).
I added debounce but it is not a proper solution.

@Koenkk if you tell me what kind of logs you need then I can try to produce them.

Part of log:
```log
Zigbee2MQTT:debug 2020-10-15 00:21:04: Received Zigbee message from 'Bedroom Main Light Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0                                                                                                      
2020-10-14T22:21:04.198Z zigbee-herdsman:controller:endpoint DefaultResponse 0x60a423fffe9d9771/1 6(10, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})                                                                                                                                                                                                                                                                                     
2020-10-14T22:21:04.199Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []                                                                                                                                                                                                                
2020-10-14T22:21:04.498Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,73,251,1,60,212,223]                                                                                                                                                                                          
2020-10-14T22:21:04.498Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,73,251,1,60,212,223]                                                                                                                                                                                
2020-10-14T22:21:04.498Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [73,251,1,60,212] - 223                                                                                                                                                                            
2020-10-14T22:21:04.498Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":64329,"relaycount":1,"relaylist":[54332]}                                                                                                                                                         
2020-10-14T22:21:04.498Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []                                                                                                                                                                                                                
2020-10-14T22:21:04.546Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,73,251,1,1,0,92,0,182,164,183,0,0,7,8,27,10,0,0,16,0,60,212,28,105]                                                                                                                                  
2020-10-14T22:21:04.547Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,73,251,1,1,0,92,0,182,164,183,0,0,7,8,27,10,0,0,16,0,60,212,28,105]                                                                                                                        
2020-10-14T22:21:04.547Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,73,251,1,1,0,92,0,182,164,183,0,0,7,8,27,10,0,0,16,0,60,212,28] - 105                                                                                                                    
2020-10-14T22:21:04.547Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":64329,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":92,"securityuse":0,"timestamp":12035254,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,27,10,0,0,16,0]}}                                                                                                                                                                                                                                                                                
2020-10-14T22:21:04.549Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":27,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}]},"address":64329,"endpoint":1,"linkquality":92,"groupID":0}'                                                                                                                                                                              
Zigbee2MQTT:debug 2020-10-15 00:21:04: Received Zigbee message from 'Bedroom Main Light Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0                                                                                                      
2020-10-14T22:21:04.552Z zigbee-herdsman:controller:endpoint DefaultResponse 0x60a423fffe9d9771/1 6(10, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNu [... cut ...]                                                                                     
2020-10-14T22:21:09.080Z zigbee-herdsman:controller:endpoint DefaultResponse 0x60a423fffe9d9771/1 6(10, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'Timeout' (9999))                                              
```

Shaquu, 2020-10-14:
Actually I got another problem too. I was unable to control mentioned switch with zigbee2mqtt client.
Resolved by removing device and pairing again. Now message is no longer multiplied and I can control it (via homekit in this case).

golddragon007, 2020-10-14:
@Shaquu This looks two different issues, your message timestamp is different per message. Therefore normally it's not one single action was made (there was multiple button press). Also, you have one single MQTT publish, I have 5 for the same.

@Koenkk I looking now a bit deeper my logs, WTF is this dstendpoint? Why is different for all my messages? Does my device remember for the previous networks and tries to send for all?

Shaquu, 2020-10-14:
@golddragon007 are you sure it is different? I was clicking once my switch and it was generating multiple mqtt messages for real.
Maybe my cut log above is not enough to tell that but it was a little longer then that.

Koenkk, 2020-10-15:
@golddragon007 found a possible fix.

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

golddragon007, 2020-11-15:
Actually, if I remember well, a repair solved this issue, at least mine.

closed 2020-12-24 by github-actions[bot]