Koenkk/zigbee2mqtt #4664: Turning on a group of lights will turn on at low brightness

LukeChannings, 2020-10-14:
I use Zigbee2MQTT with Home Assistant, and it's a regular occurrence that I issue a command: "Alexa, turn on the kitchen lights".

The lights turn on, but the brightness of all the bulbs will be so low that they're almost off.

I then correct it with "Alexa, set the kitchen lights to 100".

Is this a Home Assistant problem?

Shaquu, 2020-10-14:
@LukeChannings to decide who's fault it is you should debug logs and see if HA is sending request containing brightness or not.
In my case I set manually 6% brightness then switch lights off manaully (iPhone Home.app). If I turn it on manually (iPhone Home.app) then it is 6% again. But when I said "Hey Siri, turn wardrobe lights on" then it turned on with 100% brightness.

LukeChannings, 2020-10-15:
It appears to be zigbee2mqtt that's at fault. Tailing the logs for z2mqtt when I say to Alexa "turn on the lounge lights" (a group), the MQTT events are:

```
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:info  2020-10-15 12:24:16: MQTT publish: topic 'zigbee2mqtt/lounge_lights', payload '{"brightness":254,"state":"ON"}'
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:info  2020-10-15 12:24:16: MQTT publish: topic 'zigbee2mqtt/lounge_2', payload '{"brightness":254,"last_seen":1602713730145,"state":"ON","update_available":false}'
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:info  2020-10-15 12:24:25: MQTT publish: topic 'zigbee2mqtt/lounge_1', payload '{"brightness":254,"last_seen":1602713727796,"state":"OFF","update_available":false}'
```

The brightness is actually set at something like 30 (I don't know the exact value because lights aren't updated).

My lights config in Home Assistant is:

```yaml
- platform: mqtt
  schema: json
  name: lounge_lights
  command_topic: "zigbee2mqtt/lounge_lights/set"
  state_topic: "zigbee2mqtt/lounge_lights"
  color_temp: true
  brightness: true
```

Koenkk, 2020-10-15:
Can you provide the herdsman debug logging when you say Alexa "turn on the lounge lights".

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

LukeChannings, 2020-10-15:
```
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:info  2020-10-15 15:37:59: Switching log level to 'debug'
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:info  2020-10-15 15:37:59: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"4afcff5","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200805,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":20,"extendedPanID":"0xdddddddddddddddd","panID":6759},"permit_join":false,"version":"1.14.4-dev"}'
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:debug 2020-10-15 15:38:25: Received MQTT message on 'zigbee2mqtt/lounge_lights/set' with data '{"state": "ON"}'
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:debug 2020-10-15 15:38:25: Publishing 'set' 'state' to 'lounge_lights'
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:info  2020-10-15 15:38:25: MQTT publish: topic 'zigbee2mqtt/lounge_1', payload '{"brightness":254,"last_seen":1602713727796,"state":"ON","update_available":false}'
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:info  2020-10-15 15:38:25: MQTT publish: topic 'zigbee2mqtt/lounge_lights', payload '{"brightness":254,"state":"ON"}'
[zigbee2mqtt-65dc8467c6-6hhxv] Zigbee2MQTT:info  2020-10-15 15:38:25: MQTT publish: topic 'zigbee2mqtt/lounge_2', payload '{"brightness":254,"last_seen":1602713730145,"state":"ON","update_available":false}'
```

Koenkk, 2020-10-15:
@LukeChannings this is not the herdsman debug logging, note that it's only logged to stdout (not the log files). Improved the docs.

LukeChannings, 2020-10-16:
Is this what you were expecting?

```
[zigbee2mqtt-5578489556-v2z9v] Zigbee2MQTT:info  2020-10-16 17:17:52: MQTT publish: topic 'zigbee2mqtt/lounge_1', payload '{"brightness":254,"last_seen":1602713727796,"state":"ON","update":{"state":"idle"},"update_available":false}'
[zigbee2mqtt-5578489556-v2z9v] Zigbee2MQTT:info  2020-10-16 17:17:52: MQTT publish: topic 'zigbee2mqtt/lounge_lights', payload '{"brightness":254,"state":"ON"}'
[zigbee2mqtt-5578489556-v2z9v] Zigbee2MQTT:info  2020-10-16 17:17:52: MQTT publish: topic 'zigbee2mqtt/lounge_2', payload '{"brightness":254,"last_seen":1602713730145,"state":"ON","update":{"state":"idle"},"update_available":false}'
[zigbee2mqtt-5578489556-v2z9v] 2020-10-16T16:17:57.550Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,39,68,129,0,0,25,0,83,66,2,1,0,84,0,129,254,167,0,0,19,1,207,3,2,11,16,9,1,157,111,0]
[zigbee2mqtt-5578489556-v2z9v] 2020-10-16T16:17:57.551Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,39,68,129,0,0,25,0,83,66,2,1,0,84,0,129,254,167,0,0,19,1,207,3,2,11,16,9,1,157,111,0]
[zigbee2mqtt-5578489556-v2z9v] 2020-10-16T16:17:57.554Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [66,40,136,0,0,64,0,0,133,213,28,182]
[zigbee2mqtt-5578489556-v2z9v] 2020-10-16T16:17:57.554Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,39,68,129,0,0,25,0,83,66,2,1,0,84,0,129,254,167,0,0,19,1,207,3,2,11,16,9,1,157,111,0,66,40,136,0,0,64,0,0,133,213,28,182]
[zigbee2mqtt-5578489556-v2z9v] 2020-10-16T16:17:57.554Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 39 - 2 - 4 - 129 - [0,0,25,0,83,66,2,1,0,84,0,129,254,167,0,0,19,1,207,3,2,11,16,9,1,157,111,0,66,40,136,0,0,64,0,0,133,213,28] - 182
[zigbee2mqtt-5578489556-v2z9v] 2020-10-16T16:17:57.555Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":16979,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"linkquality":84,"securityuse":0,"timestamp":11009665,"transseqnumber":0,"len":19,"data":{"type":"Buffer","data":[1,207,3,2,11,16,9,1,157,111,0,66,40,136,0,0,64,0,0]}}
[zigbee2mqtt-5578489556-v2z9v] 2020-10-16T16:17:57.557Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":207,"manufacturerCode":null,"commandIdentifier":3},"Payload":{"fieldControl":2,"manufacturerCode":4107,"imageType":265,"fileVersion":1107324829,"fileOffset":34856,"maximumDataSize":64}},"address":16979,"endpoint":2,"linkquality":84,"groupID":0}'
[zigbee2mqtt-5578489556-v2z9v] 2020-10-16T16:17:57.557Z zigbee-herdsman:controller:log Skipping command 'imageBlockRequest' because it is missing from the lookup
[zigbee2mqtt-5578489556-v2z9v] 2020-10-16T16:17:57.558Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

LukeChannings, 2020-10-16:
Note: I'm using a ZZH stick, which I've just upgraded to the latest firmware, and I'm now running zigbee2mqtt 1.15.0 in a Kubernetes v1.18.9 cluster, on a Raspberry Pi 4.

Koenkk, 2020-10-16:
@LukeChannings yes that is the logging, but I need the part starting from `Received MQTT message on` till `MQTT publish..` (now the log starts when the the command has already been executed).

LukeChannings, 2020-10-16:
Ok, this time I'm using the kitchen_lights group, but the same issue is present:

```
Zigbee2MQTT:debug 2020-10-16 17:48:23: Received MQTT message on 'zigbee2mqtt/kitchen_lights/set' with data '{"state": "ON"}'
Zigbee2MQTT:debug 2020-10-16 17:48:23: Publishing 'set' 'state' to 'kitchen_lights'
  zigbee-herdsman:controller:group Command 4 genOnOff.on({}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000004","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":2,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,3,1]}} +16s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,2,1,4,0,0,0,0,0,0,0,255,0,0,1,6,0,2,0,30,3,0,17,3,1,192] +16s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +16s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +16s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,2,196] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,2,196] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2} +16s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:info  2020-10-16 17:48:23: MQTT publish: topic 'zigbee2mqtt/kitchen', payload '{"brightness":254,"last_seen":1602713726402,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 17:48:23: MQTT publish: topic 'zigbee2mqtt/kitchen_lights', payload '{"brightness":254,"state":"ON"}'
Zigbee2MQTT:info  2020-10-16 17:48:23: MQTT publish: topic 'zigbee2mqtt/kitchen_spot1', payload '{"brightness":254,"last_seen":1602713728418,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 17:48:23: MQTT publish: topic 'zigbee2mqtt/kitchen_spot2', payload '{"brightness":254,"last_seen":1602713726806,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 17:48:23: MQTT publish: topic 'zigbee2mqtt/kitchen_spot3', payload '{"brightness":254,"last_seen":1602713726084,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 17:48:23: MQTT publish: topic 'zigbee2mqtt/kitchen_spot4', payload '{"brightness":254,"last_seen":1602713730289,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

Koenkk, 2020-10-16:
Can you provide the same logging when you turned off the bulbs? (what was the brigthness at the time you turned them off)

LukeChannings, 2020-10-16:
I have a Philips Hue switch that controls the group (I use NodeRED for the automation). The remote has a brightness associated with it, but it's only used when the brightness up / down is explicitly pressed.

The big problem I have at the moment is that when I say "Alexa, set the kitchen lights to 100", the lights are fixed, and I can't reproduce the issue. It will come back eventually, though. I'll post all of the related events below, but I'd really like to be able to reproduce it consistently...

# Alexa Off

```
Zigbee2MQTT:debug 2020-10-16 19:03:40: Received MQTT message on 'zigbee2mqtt/kitchen_lights/set' with data '{"state": "OFF"}'
Zigbee2MQTT:debug 2020-10-16 19:03:40: Publishing 'set' 'state' to 'kitchen_lights'
  zigbee-herdsman:controller:group Command 4 genOnOff.off({}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000004","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":2,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,3,0]}} +50s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,2,1,4,0,0,0,0,0,0,0,255,0,0,1,6,0,2,0,30,3,0,17,3,0,193] +50s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +50s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +50s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,2,196] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,2,196] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2} +51s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen', payload '{"brightness":254,"last_seen":1602713726402,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen_lights', payload '{"brightness":254,"state":"OFF"}'
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen_spot1', payload '{"brightness":254,"last_seen":1602713728418,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen_spot2', payload '{"brightness":254,"last_seen":1602713726806,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen_spot3', payload '{"brightness":254,"last_seen":1602713726084,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen_spot4', payload '{"brightness":254,"last_seen":1602713730289,"state":"OFF","update":{"state":"idle"},"update_available":false}'
```

# Alexa On

```
Zigbee2MQTT:debug 2020-10-16 19:03:40: Received MQTT message on 'zigbee2mqtt/kitchen_lights/set' with data '{"state": "OFF"}'
Zigbee2MQTT:debug 2020-10-16 19:03:40: Publishing 'set' 'state' to 'kitchen_lights'
  zigbee-herdsman:controller:group Command 4 genOnOff.off({}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000004","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":2,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,3,0]}} +50s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,2,1,4,0,0,0,0,0,0,0,255,0,0,1,6,0,2,0,30,3,0,17,3,0,193] +50s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +50s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +50s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,2,196] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,2,196] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2} +51s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen', payload '{"brightness":254,"last_seen":1602713726402,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen_lights', payload '{"brightness":254,"state":"OFF"}'
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen_spot1', payload '{"brightness":254,"last_seen":1602713728418,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 19:03:40: MQTT publish: topic 'zigbee2mqtt/kitchen_spot2', payload '{"brightness":254,"last_seen":1602713726806,"state":"OFF","update":{"state":"idle"},"update_available":false}'
[... cut ...]
Zigbee2MQTT:info  2020-10-16 19:09:15: MQTT publish: topic 'zigbee2mqtt/kitchen_spot2', payload '{"brightness":2,"last_seen":1602713726806,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 19:09:15: MQTT publish: topic 'zigbee2mqtt/kitchen_spot3', payload '{"brightness":2,"last_seen":1602713726084,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-10-16 19:09:15: MQTT publish: topic 'zigbee2mqtt/kitchen_spot4', payload '{"brightness":2,"last_seen":1602713730289,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

Koenkk, 2020-10-17:
With the Alexa on command, the bulbs will turn on at the brightness they were turned off with. So if they were turned off with Alexa Off at brightness 20, they will turn on at brightness 20 again when you say Alexa on.

closed 2020-11-24 by github-actions[bot]