Koenkk/zigbee2mqtt #4671: Philips Hue motion outdoor sensor leave the network

arteck, 2020-10-15:
https://github.com/Koenkk/zigbee2mqtt/issues/2693

the issue is closed..but the problem is still present
```

2020-10-14 20:15:50.875  - [34mdebug[39m: zigbee.0 (1894) handleDeviceLeave. {"ieeeAddr":"0x001788010647b3a5"}
2020-10-14 20:15:50.875  - [34mdebug[39m: zigbee.0 (1894) Device '0x001788010647b3a5' left the network
2020-10-14 20:15:50.875  - [34mdebug[39m: zigbee.0 (1894) Leave device event: 0x001788010647b3a5
2020-10-14 20:15:50.875  - [34mdebug[39m: zigbee.0 (1894) Delete device 001788010647b3a5 from iobroker.
2020-10-14 20:15:50.938  - [32minfo[39m: history.0 (1141) disabled logging of zigbee.0.001788010647b3a5.link_quality                                                                                                               
2020-10-14 20:15:50.953  - [32minfo[39m: history.0 (1141) disabled logging of zigbee.0.001788010647b3a5.occupancy

```

Koenkk, 2020-10-15:
This requires a firmware fix for CC2538: https://github.com/Koenkk/Z-Stack-firmware/commit/f07c6fbf4b33370f105334ec2f36842c11bf02ca#diff-7708f06bce3ff0e0005331ca95cf7874891b1bf02a6de7540d33cf04748a2313R1663

arteck, 2020-10-15:
revision":20200211  dosnt work also

ccorderod, 2020-10-15:
Morning. This also happen with firmware 20200925 on CC1352P_2. I have 2 ZB networks, each supported by a CC1352P_2 coordinator running Z2M 1.15.0-dev, and Hue motion sensors (indoor) leave the network; sometimes they rejoin, sometimes they don't and in this case I have to reset them. I have a BUNCH of hue motion indoor sensors on each network. Even some VERY CLOSE to the coordinator show this behaviour.
This did not happen with previous releases (I am running Z2M since 2018). With previous releases hue motion sensor had sometimes the "red led", but did not leave the network.

Koenkk, 2020-10-15:
@arteck does this CC2538 firmware have this fix?

@ccorderod  I will support you in the issue you've created yourself

netweaver1970, 2020-10-29:
Same issue here but using an LAUNCHXL-CC26X2R1 board. Does it also needs a FW update? I'm still on an old version (april or so).

Koenkk, 2020-10-29:
@netweaver1970 I recommend the latest firmware indeed (20200925 currently). Also switch to the latest dev branch as it contains a possible fix not in a release version yet. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

timota, 2020-11-13:
Issue still persist with the latest dev branch and latest FW
SW: `1.16.1-dev commit: ca33e22`
FW: `20201026`
Coordinator: `cc1352p-2`

timota, 2020-11-15:
Some info regarding sensor.
When sensor disappeared from network - 5 min later i just allowed joined (Permit Join) and it back again.
Hope this cab be helpful. Unfortunately no debug.

felix63, 2020-11-18:
I am a newbee when it comes to zigbee2mqtt. Have the same problem with a disappearing Hue Outdoor Motion Sensor. I am using latest stable version combined with a new Electrolama zig-a-zig-ah! flashed with CC26X2R1_20201026.

Is there anything I could do to fix this or is there any info I could provide to help find a solution for this issue?

netweaver1970, 2020-11-24:
mmm, fully updated and still the same behaviour, after a while (day or so) it left the network.

zigbee2mqtt version : 1.16.1-dev commit: 5f7553f
coordinator type: zStack3x0
coordinator revision: 20201113

Koenkk, 2020-11-24:
@netweaver1970 can you provide the herdsman debug log when this happens?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

timota, 2020-11-27:
Hi @Koenkk, 
finally i have managed to catch debug for sensor. (link below) It contains 2000 lines before leave message and 500 after, hope this is enough for debug. Sensor is `0x001788010647face`

[Debug](https://pastebin.com/vpUU67k5)

Please let me know if you need more info. 

Thanks in advance.

closed 2021-01-04 by github-actions[bot]