Koenkk/zigbee2mqtt #4673: Device to be added

Valliess, 2020-10-15:
Hi all,

Just received 2 Iris iL02-1 (V3) security keypads.
A hubitat forum states that this model is identical to the CentraLite 3400 (which is listed in the supported devices list) and some Xfinity model.
So I know this sounds silly but I don't know how to add this (I'm using Loxberry with Zigbee2mqtt and a Mosquitto bridge)
Could someone please add this device to the list in order for me to be able to implement this keypad into my loxone system?

This is a small extraction from the log:
warn 2020-10-15 13:59:51: Received message from unsupported device with Zigbee model '1112-S' 

Hopefully this will do for a gentle guy to help me out.

Thanks,

Valentijn

Unsupported device?
- https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Yes I know, but linux containers and other special effects are not my cup of tea..

Koenkk, 2020-10-23:
Try adding the `zigbeeModel` when pairing the device to https://github.com/Koenkk/zigbee-herdsman-converters/blob/9f058601246c48032b8a9c40eb9363b68c64fecf/devices.js#L7982

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

Valliess, 2020-10-23:
I've added the device 1112-S again to the zigbeeModel '3400-D' and this time (maybe the way I did the pairing part??) it was paired, configured and supported (seen in the logs) strange. But only the Temperature and linkquality are regularly communicated.
When pressing "ON" (arm), "Partial" (partial arm), "OFF" (Disarm) or a cipherkey, nothing happens in the logs. 
info  2020-10-23 20:26:20: MQTT publish: topic '/Iris', payload '{"linkquality":39,"temperature":22.7}' 

Any ideas to go further ? maybe this could help also:
info  2020-10-23 20:24:40: Iris (0xd0cf5efffeb4d023): 3400-D or iMagic - CentraLite or Greatstar Security keypad (EndDevice)
info  2020-10-23 20:24:40: Configuring 'Iris'
info  2020-10-23 20:25:49: Successfully interviewed 'Iris', device has successfully been paired
info  2020-10-23 20:25:49: Device 'Iris' is supported, identified as: CentraLite or Greatstar Security keypad (3400-D or iMagic)
info  2020-10-23 20:25:49: Configuring 'Iris'
info  2020-10-23 20:25:49: MQTT publish: topic '/bridge/event', payload '{"data":{"definition":{"description":"Security keypad","model":"3400-D or iMagic","supports":"action, arm","vendor":"CentraLite or Greatstar"},"friendly_name":"Iris","ieee_address":"0xd0cf5efffeb4d023","status":"successful","supported":true},"type":"device_interview"}'
info  2020-10-23 20:25:49: MQTT publish: topic '/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]}},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]}},"110":{"bindings":[],"clusters":{"input":[],"output":[]}},"12":{"bindings":[],"clusters":{"input":[],"output":[]}},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]}},"2":{"bindings":[],"clusters":{"input":[],"output":[]}},"242":{"bindings":[],"clusters":{"input":[],"output":[]}},"3":{"bindings":[],"clusters":{"input":[],"output":[]}},"4":{"bindings":[],"clusters":{"input":[],"output":[]}},"47":{"bindings":[],"clusters":{"input":[],"output":[]}},"5":{"bindings":[],"clusters":{"input":[],"output":[]}},"6":{"bindings":[],"clusters":{"input":[],"output":[]}},"8":{"bindings":[],"clusters":{"input":[],"output":[]}}},"friendly_name":"Coordinator","ieee_address":"0x00124b001cd47497","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"},{"date_code":"","definition":{"description":"10A UK or 16A EU smart plug","model":"TS0121_plug","supports":"on/off, power measurements","vendor":"TuYa"},"endpoints":{"1":{"bindings":[{"cluster":"genOnOff","target":{"endpoint":1,"ieee_address":"0x00124b001cd47497","type":"endpoint"}},{"cluster":"haElectricalMeasurement","target":{"endpoint":1,"ieee_address":"0x00124b001cd47497","type":"endpoint"}}],"clusters":{"input":["genBasic","genGroups","genScenes","genOnOff","seMetering","haElectricalMeasurement"],"output":["genOta","genTime"]}}},"friendly_name":"Blitzwolf","ieee_address":"0x842e14fffe186ba3","interview_completed":true,"interviewing":false,"network_address":33618,"power_source":"Mains (single phase)","supported":true,"type":"Router"},{"date_code":"20191026","definition":{"description":"TRADFRI control outlet","model":"E1603/E1702","supports":"on/off","vendor":"IKEA"},"endpoints":{"1":{"bindings":[{"cluster":"genOnOff","target":{"endpoint":1,"ieee_address":"0x00124b001cd47497","type":"endpoint"}}],"clusters":{"input":["genBasic","genIdentify","genGroups","genScenes","genOnOff","genLevelCtrl","touchlink"],"output":["genScenes","genOta","genPollCtrl","touchlink"]}},"242":{"bindings":[],"clusters":{"input":["greenPower"],"output":["greenPower"]}}},"friendly_name":"Tradfri1","ieee_address":"0x588e81fffe99315c","interview_completed":true,"interviewing":false,"network_address":2518,"power_source":"Mains (single phase)","software_build_id":"2.0.024","supported":true,"type":"Router"},{"date_code":"20191026","definition":{"description":"TRADFRI control outlet","model":"E1603/E1702","supports":"on/off","vendor":"IKEA"},"endpoints":{"1":{"bindings":[{"cluster":"genOnOff","target":{"endpoint":1,"ieee_address":"0x00124b001cd47497","type":"endpoint"}}],"clusters":{"input":["genBasic","genIdentify","genGroups","genScenes","genOnOff","genLevelCtrl","touchlink"],"output":["genScenes","genOta","genPollCtrl","touchlink"]}},"242":{"bindings":[],"clusters":{"input":["greenPower"],"output":["greenPower"]}}},"friendly_name":"Tradfri2","ieee_address":"0x588e81fffe64d3c7","interview_completed":true,"interviewing":false,"network_address":28507,"power_source":"Mains (single phase)","software_build_id":"2.0.024","supported":true,"type":"Router"},{"definition":{"description":"MiJia Honeywell smoke detector","model":"JTYJ-GD-01LM/BW","supports":"smoke","vendor":"Xiaomi"},"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]}}},"friendly_name":"Brandmelder_bovengang","ieee_address":"0x00158d00052d4072","interview_completed":true,"interviewing":false,"network_address":41050,"power_source":"Battery","supported":true,"type":"EndDevice"},{"date_code":"20191118","definition":{"description":"MiJia light intensity sensor","model":"GZCGQ01LM","supports":"illuminance","vendor":"Xiaomi"},"endpoints":{"1":{"bindings":[{"cluster":"genPowerCfg","target":{"endpoint":1,"ieee_address":"0x00124b001cd47497","type":"endpoint"}},{"cluster":"msIlluminanceMeasurement","target":{"endpoint":1,"ieee_address":"0x00124b001cd47497","type":"endpoint"}}],"clusters":{"input":["genBasic","msIlluminanceMeasurement","gen [... cut ...]
info  2020-10-23 20:25:49: MQTT publish: topic '/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Security keypad","friendly_name":"Iris","model":"3400-D or iMagic","supported":true,"vendor":"CentraLite or Greatstar"},"type":"pairing"}'
info  2020-10-23 20:25:55: MQTT publish: topic '/Blitzwolf', payload '{"current":0.04,"energy":0.11,"linkquality":39,"power":6,"state":"ON","voltage":228}'
info  2020-10-23 20:25:55: Successfully configured 'Iris'

Thanks very much for your help, 

Valentijn

Koenkk, 2020-10-24:
@Valliess what shows up in the herdsman debug log when arming the keypad?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Valliess, 2020-10-24:
OK. This is new terrain for me.. Trying to figure out how to do this on the loxberry zigbee2mqtt plugin..
Hopefully romanlum gives me a helpful hand.

Thanks for your help. Really looking forward to integrate this keypad in my installation. 

Valentijn

Valliess, 2020-10-24:
@Koenkk Hi, 
I suppose that I've need to add "zigbee_herdsman_debug: true" to the configuration.yaml, which is what I did.

When monitoring the log afterwards while pushing the 'ON'-button - which should mean ARM - this is what's showing up:
**debug** 2020-10-24 21:43:31: Received Zigbee message from 'Iris', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
**debug** 2020-10-24 21:43:31: No converter available for '3400-D or iMagic' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":0}'
**debug** 2020-10-24 21:43:59: Received Zigbee message from 'Iris', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":268657200,"imageType":18,"manufacturerCode":4667}' from endpoint 1 with groupID 0

But not sure if the DEBUG mode for the herdsman works as you intended. Being new in this matter, doubting about the how and what to do..

When pressing the "partial" or "OFF" button, nothing appears in the log. 

Can you already use the provided info (above) from the debug log?

Valentijn

Koenkk, 2020-10-26:
For the home assistant addon, switch to edge and set `zigbee_herdsman_debug` to `true` in the web config of Zigbee2MQTT. If you can provide this logging for each button pressed I can check if I can do something.

Valliess, 2020-10-26:
No it's not home assistant. HA is 'deactivated' in the configuration.yaml. This is for the loxberry version, some kind of extension for loxone users. But I re-checked and the log provided earlier is the only thing I'm getting when pressing buttons..

Valliess, 2020-10-27:
@Koenkk Well I've was able to realise a herdsman-debug as you asked with another raspberry. just basic setup, installed zigbee2mqtt and joined the Iris keypad 'as if it was an 3400-D => Using this in devices.js.
Then started the herdsman in debug mode and pressed one by one all the buttons, starting with ON (ARM) if I'm not mistaken.
I've copied the whole log in the joined txt file.
If you need something specific, please don't hesitate to ask !

Many thanks in advance Koen,

Valentijn
[Iris iL02_1_1112-S_herdsman_debug_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5445452/Iris.iL02_1_1112-S_herdsman_debug_log.txt)

Koenkk, 2020-10-28:
@Valliess you've provided the whole pairing log, but I'm only interested in what is logged directly after you press a button. Please provide the logging of this in the following format:
```
----- Pressed ARM button
........ logging....
```

Valliess, 2020-10-29:
@Koenkk Yes, sorry for that. The herdsman debug STDOUT spits a lot of (unnecessary? information) plus the fact that is is difficult to determine what **you** want to see from my point of view (I'm not really into these zigbee debug things... This is all new to me) 
So today, as a test, I've pressed the ON button, suggesting a demand to arm all zones I suppose, and I've tried to filter what came out the console debug log:
 __zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,0,0,232,165,1,1,0,86,0,141,144,117,0,0,8,8,9,1,1,0,0,32,0,232,165,29,211] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,0,232,165,1,1,0,86,0,141,144,117,0,0,8,8,9,1,1,0,0,32,0,232,165,29,211] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,0,0,232,165,1,1,0,86,0,141,144,117,0,0,8,8,9,1,1,0,0,32,0,232,165,29] - 211 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":42472,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":86,"securityuse":0,"timestamp":7704717,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[8,9,1,1,0,0,32,0]}} +1s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":9,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":1,"status":0,"dataType":32,"attrData":0}]},"address":42472,"endpoint":1,"linkquality":86,"groupID":0}' +2s
Zigbee2MQTT:debug 2020-10-29 19:58:17: Received Zigbee message from 'Iris', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-10-29 19:58:17: No converter available for '3400-D' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":0}'
  zigbee-herdsman:controller:endpoint DefaultResponse 0xd0cf5efffeb4d023/1 0(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +2s__

So, can you use this, and if yes, what type of information are you searching for exactly ? In this way I can try the other buttons.

Thanks Koen,

Valentijn

Koenkk, 2020-10-30:
First, I think we can do it with only debug logging so you can drop the herdsman debug logging.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Next, first make sure the device is interviewed successfully, the above log seems to be from the interview process. To provide what is logged after each button press. In HA press refresh, immediately trigger the action and press refresh again.

Valliess, 2020-11-24:
Hi Koen, sorry for the late reply.

I just want to mention, not being a specialist at all in this Iriss matter, that the Iriss device doen NOT send a message at each button press. The device works in a certain logic. In fact the logic is simple (as seen in the logs there is simply not always a message recognised in Z2M at each button press) it is based on the simple process as follows:
You can Press ON when leaving the house for example, the device generates a command and awaites a feedback stating that the alarm is armed. This illuminates the ON button. This implies that the installation is armed ALL ZONES.
Another possibility is that you arm via the PARTIAL button. This arms, for example, the installation only with doorsensors instead of movement sensors. After pressing the button, the Iriss awaits a feedback stating the partial arm of zones, and than illuminated the PARTIAL button. As a last part of the sequence (regarding the armed ALL or PARTIAL zones) you can press a cipher code followed by OFF. When you've entered a correct code, the device can be put in OFF mode depending on the feedback from the disarmed central.

This is really the logiv from the Iriss device and can not ba interpreted differently because it is hardware coded this way.

So since this is an issue at the moment and since I'm not into Z2M  coding, I would dare to ask you the following:

Is it an option for you that I (living in Belgium) could contact you by mail or telephone and, for example offer you an PayPal donation and send you the Iriss by post, so that you could do your thing with it and successfully do the integration in Z2M?

Valentijn

Koenkk, 2020-11-25:
Thanks for the offer but due to time constraints I have to reject it. In the past such tasks turned out to take a lot of time. I hope somebody else can integrate this device.

Dinth, 2020-11-25:
@Valliess you mentioned that this keypad is quite similar to Centralite 3400 keypads which also has issue opened: https://github.com/Koenkk/zigbee2mqtt/issues/4613

Valliess, 2020-11-25:
@Dinth yes indeed. After adding it to devices.js as an centralite, it works partially but not all the commands are recognized.

closed 2021-01-02 by github-actions[bot]

goodwin, 2021-02-07:
@Koenkk Finally, received my Iris V3 keypad. So as I have a more technical background and more possibilities - let's try to bring new life into this device support.
Currently, playing with device config, converters etc - I'm getting big part of things working more or less properly, but accidentally found one very annoying (and strange for me) issue - I'm not getting ANY requests in the IAS Ace cluster from the keypad. So, for example, I can send a request from z2m to keypad to set some specific arm mode, and I can receive sensors data from the keypad into z2m. But, when pressing ARM button on the keypad - nothing is registered on the Z2M side (and yes - tried debug log for z2m intself, and tried debug mode for the herdsman - no observable activity on any level)
However, watching with the sniffer - I can definitely see IAS Ace packages sent from device to 0x0000 with proper command code and some payload
Here is Wireshark short capture during "arm" button was pressed on the keypad itself - you can see two IAS Ace messages with commands 0x07 (I suppose it is panelChange, for motion sensor activity) and 0x00 (arm command itself)
[iris1.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5940113/iris1.pcapng.zip)
Maybe, you have any idea what can be a problem (I'm testing all of this on the latest trunk z2m), or where I can look for an issue?

goodwin, 2021-02-08:
Also, found one strange thing in the herdsman log - after re-binding device, after some time, this part always appear - maybe, this is somehow related to missing incoming IAS packages?

021-02-08T10:32:38.651Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0xd0cf5efffeb38410:30127,205,4)
2021-02-08T10:32:38.651Z zigbee-herdsman:controller:endpoint Read 0xd0cf5efffeb38410/1 ssIasZone(["iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disa
bleRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":fa
lse}) failed (Data request failed with error: 'No network route' (205))
2021-02-08T10:32:38.651Z zigbee-herdsman:controller:device:log Interview - failed for device '0xd0cf5efffeb38410' with error 'Error: Read 0xd0cf5efffeb38410/1 ssIasZone(["
iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,
"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.<anonymous> (/Users/goodwin/wrk/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:339:27)
    at Generator.next (<anonymous>)
    at fulfilled (/Users/goodwin/wrk/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)'
2021-02-08T10:32:38.652Z zigbee-herdsman:controller:error Interview failed for '0xd0cf5efffeb38410 with error 'Error: Read 0xd0cf5efffeb38410/1 ssIasZone(["iasCieAddr","zo
neState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCo
de":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

goodwin, 2021-02-08:
However, after this error I'm able to send packets to the keypad, and receive battery/temperature reports

goodwin, 2021-02-08:
From digging more - i can't catch IAS Enroll Request from the keypad, going to the Coordinator. However, z2m writes IAS CIE address correctly. Still trying to play with a different sequence of actions for triggering IAS enroll, but no luck as of now...

goodwin, 2021-02-08:
@Koenkk Now i really need some expert help from you - I've found another strange fact - all communication between keypad and coordinator is made between endpoints 1-to-1, however all events that are sent by keypad with IAS Ace  packages - are sent to coordinator endoint 3. And I have no idea why and what to do with this... IAS Zone looks properly enrolled, and sniffer shows packets in the air, but I have strong feeling that they are addressed to the wrong doors...

blashman, 2021-08-18:
Hi there,
Is this implementation still active?
Interested in adding the device into HA

Gahrlaag, 2021-11-17:
Is this still on the wall..? Just got one of these and would love to get it working

nphil, 2023-09-13:
Hate to bump this after so long - but has anyone managed to get this completely functional in Z2M? I have 2 of these lying around and would love to use them.

KameDomotics, 2024-07-27:
Hi, is this keypad not integrated yet?