Koenkk/zigbee2mqtt #4674: My ZB networks are really unstable

ccorderod, 2020-10-15:
## What happened
I am running 2 zigbee networks:
- 64 bulbs (Philips, Ikea), 1 xiaomi gas detector (powered), 1 cc2530 router, 7 meazon meters (routers), 2 innr sp120 plugs, 10 hue motion, 3 hue remote, 2 aqara cube, 1 nyce motion, 1 nyce door contact, 5 mccgq01LM xiaomi contact, 4 mccgq11lm contact, 5 djt11lm vibration, 5 sjcgq11lm water leak - those are spread among 2 floors and the garden.
- 42 bulbs (Philips, Ikea), 2 cc2530 router, 5 meazon meters (routers), 1 innr sp120 plug, 9 hue motion, 1 nyce motion, 1 hue remote, 1 ikea dimmer, 3 ikea remotes, 4 aqara cube, 1 mccgq01LM xiaomi contact, 12 mccgq11lm contact, 4 djt11lm vibration, 3 sjcgq11lm water leak - those are spread in 1 floor.

I have 3 issues:
- Philips Hue motion (indoor) disconnect a lot, and sometimes do not reconnect (I am forced to reset them, open the network and make them rejoin). This is happening since 20200925. Previously they show the "red led" but were mostly working. This happen even if the sensor is VERY close to the coordinator.
- Some of the end devices disconnect and do not rejoin the network.
- My logs are full of those:
`error 2020-10-15 16:42:00: Failed to ping 'LuzLavaboBanoPrincipal3'
error 2020-10-15 16:46:31: Failed to ping 'LightStripSE'
error 2020-10-15 16:46:43: Failed to ping 'LuzTorre1'
error 2020-10-15 16:46:55: Failed to ping 'HuePlay1'
error 2020-10-15 16:47:08: Failed to ping 'HuePlay2'
error 2020-10-15 17:01:58: Failed to ping 'LuzTorre2'
error 2020-10-15 17:08:44: Failed to ping 'LuzTechoDP2'
error 2020-10-15 17:16:48: Failed to ping 'LuzTorre1'
`
Side effect of this as you know is a bunch of entities not available on home assistant.

## What did you expect to happen
I built my home automation with hue bridges and ikea bridge, then I moved to z2m in 2018, supported by a cc2531. As the network grew, I split the network and move to cc1352P_2. 
Those network are big but not massive, and the huge number of routers in each one should be enough for the network to run smoothly. This is really not the case and my home automation setup is not fun at all to manage as I have to keep a lot of attention on the zb network every day.
I am running out of ideas. Looking for advise. I would like to know if other users with a similar setup are running smoothly or are suffering similar issues. Thanks.

## How to reproduce it (minimal and precise)
I wish I knew what is happening :(.

## Debug info
Zigbee2MQTT version: 1.15.0-dev
Adapter hardware: CC1352P-2
Adapter firmware version: 20200925

Koenkk, 2020-10-15:
- do you have a different `network_key` and `channel` for both networks?
- Regarding the instability, I would recommend disabling availability. It's a really heavy feature, especially for larger networks.

ccorderod, 2020-10-15:
Yes, I have different pan_id, network keys and they are on channel 20 and 25 (as far as wifi as possible). I am also using a usb extender for both coordinators. Availability is 30 in one, 60 on the other. 
I have a bunch of devices on the availability blocklist:
`    - LuzTechoPasilloPlanta1
    - LuzTechoVestidor1
    - LuzTechoVestidor2
    - LuzTechoVestidor3
    - LuzRoperoDespacho
    - LuzTechoBanoDespacho
    - LuzCentroLavaboBanoDespacho
    - LuzDerLavaboBanoDespacho
    - LuzIzqLavaboBanoDespacho
    - LuzTechoBanoPrincipal1
    - LuzTechoBanoPrincipal2
    - LuzRoperoDJ1
    - LuzRoperoDJ2
    - LuzTechoBanoJuan
    - LuzLavaboBanoJuan1
    - LuzLavaboBanoJuan2
    - LuzLavaboBanoJuan3
    - LuzRoperoDP1
    - LuzRoperoDP2
    - LuzRoperoDP3
    - LuzRoperoSaladeEstar
`
All Ikea bulbs are on those lists. I will try right now to disable availability on both to see if it help.
Regarding Hue motion sensor, any idea? those are mostly all a critical part of the home automation, and UAF (user Acceptance Factor is really bad as of now :().
Thank you Koen.

Koenkk, 2020-10-15:
@ccorderod about the hue sensor, first make sure availability is disabled. These lags in the network can cause acknowledgments to the sensor fail, making the device remove itself from the network.

ccorderod, 2020-10-15:
Understood. Txs.

ooii, 2020-10-16:
Hi,

I have almost similar issues but I have a unique network with 116 devices, based on CC2652R and 20200925.
My `availability_timeout` was not set so I assume was disabled right?
How can I fix my issues (setup and issues are described below)?
This gateway has been used in replacement of a CC2531 which was replacing a ConBee I. Should I split my network and use 2 different gateways?
Thanks.

---

Here is the setup:
- Host: 
    - Nuc with 4 cores Intel(R) Core(TM) i5-5300U CPU @ 2.30GHz
    - 16 Gb of memory
    - Ubuntu 18.04.4 LTS

- HA 0.116.2
- installation_type: Home Assistant Supervised
- zigbee2mqtt add-on version: 1.15.0
- zigbee gateway:
    - hardware: CC2652R bought from Electrolama
    - firmware: Z-Stack_3.x.0 of Aug 05, 2020
    - plugged with a USB 1m extension cable

- Devices and their placement:
    - Zigbee devices:
        - 49 bulbs (38 Hue + 4 Osram + 1 Inner + 2 Hue Go)
        - 2 Hue lightstrips
        - 3 Osram wall outlets
        - 1 Tuya valve switch (wire powered)
        - 61 battery powered devices (18 Hue dimmers, 1 Xiaomi switch, 9 Hue motion sensors, 11 Aqara weather station, 15 Aqara door/window sensor)
        - Total of 116 devices
    - House: old house with probably some thick walls
    - Placement of devices:
        - Basement: 
            - Host running HA and the gateway
            - 9 bulbs
            - 1 lightstrip
            - 1 Tuya valve
            - Several battery powered devices
        - Ground floor:
            - 11 bulbs
            - 1 Hue Go
            - 1 lightstrip
            - 1 Osram outlet
            - Several battery powered devices
        - 1st floor:
            - 16 bulbs
            - 1 Hue Go
            - 1 Osram outlet
            - Several battery powered devices
        - 2nd floor:
            - 10 bulbs
            - Several battery powered devices

- Issues:
    - Issues faced with Hue hub:
        - Using the dimmers results in a reliability of less of 50%. I had to press several times a button to have it react.

    - Issues with z2m:
        - Dimmers are lost after few days of use. They blink red and their commands never reach the gateway.
            - Force removing then repairing them usually fixes the issues for a couple of days
            - Restarting z2m may fix the issue 
        - Some hue sensors are lost from time to time.
        - Some bulbs of the 1st floor are slow to react. All the bulbs of the 1st floor are close to each other and it's always the same that misbehave. 
        - Commanding the bulbs from HA interface usually works without issues. Most of the issues occur when using the dimmers and hue motion sensors.

Koenkk, 2020-10-16:
@ooii 

> Some bulbs of the 1st floor are slow to react. All the bulbs of the 1st floor are close to each other and it's always the same that misbehave. 

Are Hue the misbehaving ones?

ooii, 2020-10-16:
Used to have issues with the Osram bulbs. Now, I'd say that bulbs do not cause any issue. The real issue comes from Hue dimmers that blink red and their command do not reach the gateway.

Koenkk, 2020-10-17:
@ooii I've created a new issue for this: https://github.com/Koenkk/zigbee2mqtt/issues/4694

ooii, 2020-10-19:
@Koenkk Thanks for creating this issue. I'll provide all what I can do to help fix this issue.
However, my issue does not only concern a device that blinks red but a device which commands never reach the coordinator anymore. That happens on several hue dimmers at my place.
First, they work perfectly but after a couple of days, pressing one time the button has no effect. If I press a button again, let's say in the next 2 or 3 seconds, then the dimmer blinks red.

ccorderod, 2020-10-19:
@Koenkk update: 
- networks are MUCH more stable. Availability is disabled on both. I also increase power to 19 on coordinators.
- updated firmware to 20201017
- updated z2m to yesterday’s dev (#4629 )

Since then, no hue motion drop from the network, and no red lights.
Seems really promising :-) thank you.

ooii, 2020-10-26:
Hi,

I described my setup above (https://github.com/Koenkk/zigbee2mqtt/issues/4674#issuecomment-709972653).
Since I was having issues, I moved my HA host and tried to put it in a central location, in my first floor. And things are really not good, even lamps not that far require delay to react. The host is a Nuc with 4 cores Intel(R) Core(TM) i5-5300U CPU @ 2.30GHz 16 Gb of memory and a lot of available resources. I cannot get why I'm having all these issues. Maybe my house is not compatible with radio IoT? 
@koenkk: when will next version including https://github.com/Koenkk/zigbee2mqtt/issues/4629 released? I'm running z2m in HA add-on and don't want to try a new install and take the risk to have to repair everything.
Thanks.

ooii, 2020-10-30:
@Koenkk, I used `zigpy_znp.tools.energy_scan` to scan my wireless environment and I found that the channel I'm using is the most noisy one (cf. attached figure).
Do you think changing the channel (and repairing everything) can solve my issue?

[image]

Koenkk, 2020-10-31:
@ooii next version will be released tomorrow, you can try changing channel indeeds (but no guarantee that it will solve your issue).

ooii, 2020-10-31:
Thanks @Koenkk 
One question: is there any suggestion you can make for me to solve my issues?

Koenkk, 2020-10-31:
@ooii try changing the channel first

ooii, 2020-11-03:
@Koenkk since I'm not willing to pair my 100+ devices, I only updated z2m, my zzh firmware, and moved the host to a location where there is much less interferences. I'm still having issues with some dimmers but others are working.
Thanks for providing advices and releasing these latest versions.

closed 2020-12-12 by github-actions[bot]