Koenkk/zigbee2mqtt #4676: Ubisys C4 binding broken after re-pairing

theFork, 2020-10-15:
I had to reset and re-pair one of my Ubisys C4 quad push-button-devices (most likely due to its firmware hanging).
After that, it was not possible to restore the bindings.

## What happened
In the new frontend I can see, that only endpoint 1 has clusters, whereas the other endpoints have none:

[image]

As described in https://github.com/Koenkk/zigbee-herdsman-converters/issues/973, the four buttons (genOnOff) are represented by the endpoints 1..4. Hence, if I want to bind button no. 3, I would have to publish something like that:

```
topic: zigbee2mqtt/bridge/bind/c4_eg_flur/3
payload: cube_tradfri_eg_flur/1
```

Doing so results in this log output:

```
ERROR: Nothing to bind from 'c4_eg_flur' to 'cube_tradfri_eg_flur'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"from":"c4_eg_flur","to":"cube_tradfri_eg_flur"},"type":"device_bind_failed"}'
ERROR: Nothing to bind
```

## What did you expect to happen
As @felixstorm documented in https://github.com/Koenkk/zigbee-herdsman-converters/issues/973, binding endpoint 3 to a single-endpoint-device (lamp) should result in binding the genOnOff cluster to the device.

## How to reproduce it (minimal and precise)
Well, TBH I don't know. I observed, that the other C4s in my household can be bound from EPs 1..4. Getting into this state appears to be related to the state ot the C4 in which I couldn't pair it anymore. I had to open the in-wall installation to push the reset button. Reset via the magic power-on-off-pattern was not possible anymore. Maybe something went wrong during the pairing process?

## Debug info
* Zigbee2MQTT version: 1.15.0 (Homeassistant Add-On)
* Adapter hardware: CC2652R (zig-a-zig-ah!)
* Adapter firmware version: 20200417
* Host: NUC with Homeassisant

I assume, that there is useful information in the database:
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/5386699/database.db.txt)

felixstorm, 2020-10-16:
Looking at your screenshot and your database it's very likely that the interview process (to detect all available endpoints and clusters) has not completed successfully after re-pairing. I would suggest that you again remove the device from your network and then pair it again - hopefully the interview process will complete successfully this time. You should also keep an eye on your log during the pairing process.

If that still does not help, we would probably need a debug log of the pairing process to investigate further...

theFork, 2020-10-17:
Hi Felix, just as you assumed another pairing process fixed it. Thanks for the quick help!

Just in case something comparable happens again, here the debug log of the pairing process when working as expected:
[new-pairing-fixes-it_debug_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5395202/new-pairing-fixes-it_debug_log.txt)

closed 2020-10-17 by theFork