Koenkk/zigbee2mqtt #4677: Problem with zigbee2mqtt and device's network

skumka, 2020-10-15:
I have 55 zigbee devices. 
After 3 days of working my network instead of being reinforced by nature of zigbee netwrk started to lose conectivity and devices became unavailable.

initially the problem existed with my first CC2531 controller. Per advice I got from Kean in similar issue I changed it and now I use cc2538+cc2592. In addition I re-flashed cc2532 into router and put it into network. 
I have also 7 ther routers (5 outlets/sockets and 2 bulbs)
I rediscovered all devices and in day 1 all was nicely connected. Problems started after 48 hours. More and more devices become not connected and hung in the air. On day three system is not possible to use. No reaction from devices, many errors in the log file:

Here is result after 3 days:
[image]

## Error messages:
rror 2020-10-15 13:15:22: Failed to ping 'Socket UPS'
error 2020-10-15 13:18:43: Failed to ping 'Router CC2531'
error 2020-10-15 13:18:58: Failed to ping 'Socket Router'
error 2020-10-15 13:20:35: Failed to ping 'Lamp Bed Door'
error 2020-10-15 13:21:58: Failed to ping 'Socket Router'
error 2020-10-15 13:23:28: Failed to ping 'Socket UPS'
error 2020-10-15 13:23:48: Failed to ping 'Lamp Bed Door'
error 2020-10-15 13:24:03: Failed to ping 'Socket Router'
error 2020-10-15 13:24:56: Failed to ping 'Lamp Bed Door'
error 2020-10-15 13:27:22: Failed to ping 'Lamp Bed Door'
error 2020-10-15 13:27:36: Failed to ping 'Router CC2531'
error 2020-10-15 13:28:06: Failed to ping 'Socket UPS'
error 2020-10-15 13:28:29: Failed to ping 'Router CC2531'
error 2020-10-15 13:31:24: Failed to ping 'Router CC2531'
error 2020-10-15 13:32:32: Failed to ping 'Socket Router'
error 2020-10-15 13:33:24: Failed to ping 'Socket UPS'
error 2020-10-15 13:34:52: Failed to ping 'Socket Router'
error 2020-10-15 13:35:54: Failed to ping 'Lamp Bed Door'
error 2020-10-15 13:36:24: Failed to ping 'Socket Router'
error 2020-10-15 13:37:18: Failed to ping 'Router CC2531'
error 2020-10-15 13:37:54: Failed to ping 'Socket UPS'
error 2020-10-15 13:39:39: Failed to ping 'Router CC2531'
error 2020-10-15 13:40:24: Failed to ping 'Lamp Bed Door'
error 2020-10-15 13:40:29: Failed to ping 'Router CC2531'
error 2020-10-15 13:41:29: Failed to ping 'Router CC2531'
error 2020-10-15 13:42:29: Failed to ping 'Router CC2531'
error 2020-10-15 13:43:47: Failed to ping 'Lamp Bed Door'
error 2020-10-15 13:43:58: Failed to ping 'Socket Router'
error 2020-10-15 13:44:48: Failed to ping 'Socket UPS'
error 2020-10-15 13:45:29: Failed to ping 'Router CC2531'
error 2020-10-15 13:50:57: Failed to ping 'Socket UPS'
error 2020-10-15 13:50:57: Failed to ping 'Socket Router'
error 2020-10-15 13:51:09: Failed to ping 'Lamp Bed Door'
error 2020-10-15 13:53:57: Failed to ping 'Socket UPS'
error 2020-10-15 13:56:29: Failed to ping 'Router CC2531'
error 2020-10-15 13:57:11: Failed to execute LQI for 'Socket ACAR'
error 2020-10-15 13:57:31: Failed to execute LQI for 'Socket UPS'
error 2020-10-15 13:57:33: Failed to ping 'Router CC2531'
error 2020-10-15 13:57:37: Failed to ping 'Lamp Bed Door'
error 2020-10-15 13:57:43: Failed to execute routing table for 'Socket UPS'
error 2020-10-15 13:57:53: Failed to execute LQI for 'Socket Lamp'
error 2020-10-15 13:58:03: Failed to execute routing table for 'Socket Lamp'
error 2020-10-15 13:58:14: Failed to execute LQI for 'Socket Router'
error 2020-10-15 13:58:24: Failed to execute LQI for 'Lamp Bed Window'
error 2020-10-15 13:58:34: Failed to execute routing table for 'Lamp Bed Window'
error 2020-10-15 13:58:45: Failed to execute LQI for 'Router CC2531'
error 2020-10-15 14:00:02: Failed to execute routing table for 'Socket Router'
error 2020-10-15 14:00:07: Failed to ping 'Socket Router'
error 2020-10-15 14:00:12: Failed to execute LQI for 'Lamp Bed Door'
error 2020-10-15 14:00:25: Failed to execute routing table for 'Lamp Bed Door'
error 2020-10-15 14:00:35: Failed to execute LQI for 'Lamp Bed Window'
error 2020-10-15 14:00:45: Failed to execute routing table for 'Lamp Bed Window'
error 2020-10-15 14:01:15: Failed to execute routing table for 'Socket UPS'
error 2020-10-15 14:01:25: Failed to execute LQI for 'Socket Lamp'
error 2020-10-15 14:01:35: Failed to execute LQI for 'Socket Consoles'
error 2020-10-15 14:01:49: Failed to execute LQI for 'Socket Router'
error 2020-10-15 14:02:00: Failed to execute routing table for 'Socket Router'
error 2020-10-15 14:02:10: Failed to execute LQI for 'Lamp Bed Window'
error 2020-10-15 14:02:20: Failed to execute routing table for 'Lamp Bed Window'
error 2020-10-15 14:03:01: Failed to ping 'Socket UPS'
error 2020-10-15 14:04:26: Failed to ping 'Lamp Bed Door'
error 2020-10-15 14:04:56: Failed to ping 'Socket Router'
error 2020-10-15 14:05:12: Failed to ping 'Lamp Bed Door'
error 2020-10-15 14:06:02: Failed to ping 'Socket UPS'
error 2020-10-15 14:07:15: Failed to ping 'Socket Router'
error 2020-10-15 14:08:45: Failed to ping 'Socket Router'
error 2020-10-15 14:09:07: Failed to ping 'Lamp Bed Door'
error 2020-10-15 14:09:34: Failed to ping 'Socket Router'
error 2020-10-15 14:12:37: Failed to ping 'Lamp Bed Door'
error 2020-10-15 14:12:54: Failed to ping 'Socket UPS'
error 2020-10-15 14:14:11: Failed to ping 'Socket Router'
error 2020-10-15 14:14:39: Failed to ping 'Router CC2531'
error 2020-10-15 14:16:31: Failed to ping 'Socket Router'
error 2020-10-15 14:18:34: Failed to ping 'Lamp Bed Door'
## Debug info
Zigbee2MQTT version: 0.115
Adapter hardware: cc2538+cc2592  and cc2531 as a router
Adapter firmware version: the newest recommended

Koenkk, 2020-10-16:
- In case of stability issues, my first recommendation is to disable the availability feature
- Xiaomi devices appearing disconnected in the networkmap is most of the times not something to worry about. Xiaomi devices sleep for a very long time, this will cause the parent (router) to remove it from the child table (but this highly depends on the software of the router, some do it some don't). This results in the Xiaomi devices having no parent in the networkmap.

skumka, 2020-10-16:
Thanks @Koenkk ... but this means that those devices are not detecting any events. Let's consider a motion sensor. It stops reporting any occupancy. If this comes to buttons - all the automation programmed is not working... Is this working as designed?

Koenkk, 2020-10-16:
@skumka a floating Xiaomi device in the network map doesn't mean it's not able to communicate over the network anymore. It just means the parent is unknown.

skumka, 2020-10-16:
@Koenkk Unfortunately in my case on HA Dashboard I any action to switch off-on Lights, detection of movement and automations behind this (like switching on Lams, sending notifications) do ne work..... This is the problem I need to resolve

skumka, 2020-10-16:
Is there any way to force devices to be connected to the router in the same room first or keep them primairly linked to this router? and do not allow to disconnect, be removed from the network?

Koenkk, 2020-10-16:
I first recommend disabling the availability feature as this may cause the adapter to completely crash (the failed pings are not a good sign, either the adapter or network collapsed). When the adapter/network crashes the motion sensor will also not work anymore.

skumka, 2020-10-16:
@Koenkk I have deactivated availability. Let's see what is the result after several hours.

ooii, 2020-10-16:
Hi @skumka, how did you disable availability, by setting `availability_timeout` to `0`?

skumka, 2020-10-16:
Personally I have removed the option from configuration file, but as far as I understand documentation setting it up to 0 also switches availability off.

ooii, 2020-10-16:
Actually, I did not have it in my conf but was having similar issues to yours.

zhapwey, 2020-10-16:
I had a similar problem that little by little the repeaters stopped responding to the ping. Although the channels did not bother the wifi on 1 and the zigbee on 25, I took the raspi to another room of the rooter and the system has not blocked again. I think that anyway the ping system brutality the network in large networks by pinging all the rooters at the same time, you could look at the possibility of pinging only the devices that have not sent more time than in the availability that will be launched at the time of exceeding that time in that way the pins would be more spaced in time

closed 2020-11-23 by github-actions[bot]