Koenkk/zigbee2mqtt #4678: problem with migration to slaesh CC2652RB

arnold-maderthaner, 2020-10-15:
Hi !

I'm trying to migrate to a new slaesh CC2652RB stick with newest firmware (20200925) but I cannot pair my devices.
When I try to reset a Xiaomi WSDCGQ01LM I get the following error:
`(node:2877) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:2877) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
`
 
same happens when I try to readd a CC2531 router :(

arnold-maderthaner, 2020-10-15:
I also tried a different firmware mentioned on slaesh website but still no luck. 
Also used his tool to check the coordniator:
`root@nuc:~/cc2652-stick/coordinator-test# python3 ./test.py -p /dev/ttyUSB0
Opening port /dev/ttyUSB0, baud 115200

--> Bootloader is working!

--> ZigBee coordinator is working!
---> transportrev 2, product 1, majorrel 2, minorrel 7, maintrel 1, revision: 20200715

Check done!
`

Koenkk, 2020-10-16:
Can you provide the herdsman debug log of `Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms at Timeout._onTimeout`?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

arnold-maderthaner, 2020-10-16:
I think I fixed it already - the issue is that I also had openhab running with a zwave stick and openhab was blocking zigbee2mqtt. it helped to upgrade to a patched zwave binding  - see:
https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245/21

closed 2020-10-16 by arnold-maderthaner