Koenkk/zigbee2mqtt #4680: MCCGQ11LM Door sensor

xfrankie62, 2020-10-16:
## Is your feature request related to a problem? Please describe
The Xiaomi Aqara sensor MCCGQ11LM has a temperature sensor, but is not retrieved by Z2M

## Describe the solution you'd like
Tmhe Aqara sensor has an environment temperature sensor embedded and Deconz platfomr is able to retrieve it, could you consider to add this value into the retrieving values from MCCGQ11LM Xiaomi sensor?

## Describe alternatives you've considered
Not retrieve it and use another sensor T/H

IVI053, 2020-10-16:
What's the source of this information? I looked at the PCB FCC photos and did not see a component that looks fitting to be a temperature or humidity sensor.

xfrankie62, 2020-10-16:
I based my request only to an experience with the Deconz plugin that show a (real) temperature of the environment where the sensor is working. I supposed that the sensor has a component that read the temperature (not the humidity)

IVI053, 2020-10-18:
Could you pleased provide a link and screenshot?

xfrankie62, 2020-10-20:
This is a MCCGQ11LM Xiaomi Door Sensor, connected to Deconz. The first screenshot is the temperature in kitchen and the second screen is the temperature after 20 minutes in the refrigerator. The temperature is provided by the sensor, but I cannot give you more infos about, except this.....
[image]
[image]

IVI053, 2020-10-20:
I looked into it a little bit more and it could be that this is the temp value of the Sensor integrated Info the JN5169 microcontroller this device uses. Should not be the Most accurate or quick responding reading but it would be better than nothing in some cases.

IVI053, 2020-10-20:
I captured some herdsman debug. Maybe someone with better zigbee knowledge comes around and takes a look.

```
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,255,212,2,147] +5s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,255,212,2,147] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [106,196,187,41] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,255,212,2,147,106,196,187,41] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [255,212,2,147,106,196,187] - 41 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":54527,"relaycount":2,"relaylist":[27283,48068]} +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0] +45ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [255,212,1,1,0,81,0,161,23,232,0,0,7,24,180,10] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,255,212,1,1,0,81,0,161,23,232,0,0,7,24,180,10] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,16,1,196,187,27,40] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,255,212,1,1,0,81,0,161,23,232,0,0,7,24,180,10,0,0,16,1,196,187,27,40] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,255,212,1,1,0,81,0,161,23,232,0,0,7,24,180,10,0,0,16,1,196,187,27] - 40 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":54527,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":81,"securityuse":0,"timestamp":15210401,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,180,10,0,0,16,1]}} +48ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":180,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":1}]},"address":54527,"endpoint":1,"linkquality":81,"groupID":0}' +6s
Zigbee2MQTT:info  2020-10-21 00:13:51: MQTT publish: topic 'z2m/OGBZ-S-WinCont-048', payload '{"battery":100,"contact":false,"last_seen":"2020-10-20T22:13:51.376Z","linkquality":81,"voltage":3025}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,255,212,2,147] +892ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,255,212,2,147] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [106,196,187,41] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,255,212,2,147,106,196,187,41] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [255,212,2,147,106,196,187] - 41 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":54527,"relaycount":2,"relaylist":[27283,48068]} +902ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0] +45ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [255,212,1,1,0,78,0,149,255,232,0,0,7,24,181,10] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,255,212,1,1,0,78,0,149,255,232,0,0,7,24,181,10] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,16,0,196,187,27,235] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,255,212,1,1,0,78,0,149,255,232,0,0,7,24,181,10,0,0,16,0,196,187,27,235] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,255,212,1,1,0,78,0,149,255,232,0,0,7,24,181,10,0,0,16,0,196,187,27] - 235 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":54527,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":78,"securityuse":0,"timestamp":15269781,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,181,10,0,0,16,0]}} +49ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":181,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}]},"address":54527,"endpoint":1,"linkquality":78,"groupID":0}' +951ms
Zigbee2MQTT:info  2020-10-21 00:13:52: MQTT publish: topic 'z2m/OGBZ-S-WinCont-048', payload '{"battery":100,"contact":true,"last_seen":"2020-10-20T22:13:52.327Z","linkquality":78,"voltage":3025}'

```

xfrankie62, 2020-10-21:
> I looked into it a little bit more and it could be that this is the temp value of the sensor integrated in the JN5169 microcontroller this device uses. Should not be the most accurate (+-7°C) or quick responding reading but it would be better than nothing in some cases.

Agree with you related the accuracy of the sensor ( in a Shelly 1 EM the internal temp sensor expose the controller temp and not the environment one), but I've tested it and compared the measure of temperature with a Xiaomi sensor WSDCGQ11LM designed for TH&P and the difference of temp between them is no more 2°C, so I suppose that Xiaomi equipped this device with a mean sensor for temperature in the environment, unfortunately I am not so skilled in the technical specifications of the sensors and cannot give you more details about this

xfrankie62, 2020-10-21:
Check the follow thread if could help you in the analysis https://github.com/dresden-elektronik/deconz-rest-plugin/issues/530

IVI053, 2020-10-27:
I can't help at this point as I do not have enought understanding of the zigbee protocol or herdsman output to make sense of it.

closed 2020-12-04 by github-actions[bot]

Koenkk, 2021-01-10:
Support now! thanks to @BerndAmend  (https://github.com/Koenkk/zigbee-herdsman-converters/pull/2054)

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

groenmarsmannetje, 2021-01-13:
Is there anyway to calibrate the temperature? I indeed notice that the temperature is off by a couple degrees. If not then I guess I need to make my own template sensor in Home Assistant to adjust it.

Koenkk, 2021-01-13:
@groenmarsmannetje check `temperature_calibration`: https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html#device-type-specific-configuration

Morphy99, 2021-03-05:
Who knew I had temperature sensors all over my windows! Well done @xfrankie62 !