Koenkk/zigbee2mqtt #4681: IKEA Lightbulb: Does not light up and error is produced

kremik, 2020-10-16:
## What happened
NodeRed sends MQTT message to set state and brightness to IKEA TRADFRI lightbulb, but it doesn't light up. It happen only sometimes, there are also successful tries, can't recognize the reason or cause.

MQTT message:
```
zigbee2mqtt/ikeabulb1/set
{"state":"ON","brightness":254}
```

There is this error in a log:

```
Oct 16 11:04:36 raspberrypi4 npm[32757]: Zigbee2MQTT:error 2020-10-16 11:04:36: Publish 'set' 'brightness' to 'ikeabulb1' failed: 'Error: Command 0xd0cf5efffe261c49/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":32599,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":53,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,15,4,254,0,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))'
Oct 16 11:04:36 raspberrypi4 npm[32757]: Zigbee2MQTT:info  2020-10-16 11:04:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'ikeabulb1' failed: 'Error: Command 0xd0cf5efffe261c49/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":254,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (SREQ '--> AF - dataRequest - {\"dstaddr\":32599,\"destendpoint\":1,\"srcendpoint\":1,\"clusterid\":8,\"transid\":53,\"options\":0,\"radius\":30,\"len\":6,\"data\":{\"type\":\"Buffer\",\"data\":[1,15,4,254,0,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"ikeabulb1"},"type":"zigbee_publish_error"}'
```

Or this error in a log:

```
Oct 16 10:56:12 raspberrypi4 npm[32757]: Zigbee2MQTT:error 2020-10-16 10:56:12: Publish 'set' 'brightness' to 'ikeabulb1' failed: 'Error: Command 0xd0cf5efffe261c49/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'Timeout' (9999))'
Oct 16 10:56:12 raspberrypi4 npm[32757]: Zigbee2MQTT:info  2020-10-16 10:56:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'ikeabulb1' failed: 'Error: Command 0xd0cf5efffe261c49/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":254,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'Timeout' (9999))'","meta":{"friendly_name":"ikeabulb1"},"type":"zigbee_publish_error"}'
```

## What did you expect to happen
To have light ON after this action.

## How to reproduce it (minimal and precise)
Send the MQTT message again.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2020-10-16:
`(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))` indicates the adapter is overloaded, are you already on the source routing firmware? (it provides more stability) https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

kremik, 2020-10-16:
Hello, I bought the adapter preinstalled so I'm not sure about the version installed. In zigbee2mqtt log is `"softwareBuildID":"zStack12"`. Overloaded is strange, I have only 10 devices (2 of them are routers - lightbulbs). Network map looks pretty strange :( 
[image]

Koenkk, 2020-10-17:
@kremik 10 devices should be fine. Could you provide me the herdsman debug logging of this, starting at the z2m start until the error appears?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

kremik, 2020-10-17:
Hello, here is a log of the start:
https://hastebin.com/elezuwequw.yaml

And here is the error when state of IKEA Lightbulb changes and it produces error as well:
https://hastebin.com/ogukolucoq.rust

Koenkk, 2020-10-18:
Sorry, I meant the log from starting Zigbee2MQTT until the error (I would like to see what happens between).

ahmadnbl, 2020-11-23:
Facing the same issue, even after restart it still got the issue.
https://hastebin.com/zuvexuwaye.rust

closed 2020-12-31 by github-actions[bot]