Koenkk/zigbee2mqtt #4683: Add to Xiaomi Aqara D1 switch QBKG24LM decoupled mode

Amarrant, 2020-10-16:
Now is no way to set decoupled mode for this switch.

ERROR: No converter available for 'operation_mode' ([object Object]) in logs.

Tried on zigbee2mqtt edge release 1.15.0-dev (commit #9e4e8ac)

Koenkk, 2020-10-16:
Can you try applying the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/1672/files.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify these files.

Amarrant, 2020-10-16:
Seems it doesn't work.

**If i publish topic zigbee2mqtt/KIDSROOM_SWITCH/system/set**
{
  "operation_mode": {
    "button": "right",
    "state": "decoupled"
  }
}

it prints 

ERROR: Publish 'set' 'operation_mode' to 'KIDSROOM_SWITCH' failed: 'Error: Write 0x00158d00052bfccc/1 aqaraOpple({"512":{"value":0,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'operation_mode' to 'KIDSROOM_SWITCH' failed: 'Error: Write 0x00158d00052bfccc/1 aqaraOpple({\"512\":{\"value\":0,\"type\":32}}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"KIDSROOM_SWITCH"},"type":"zigbee_publish_error"}'

**If i try zigbee2mqtt/KIDSROOM_SWITCH/right/set**

{
  "operation_mode": {
    "state": "decoupled"
  }
}

ERROR: Publish 'set' 'operation_mode' to 'KIDSROOM_SWITCH' failed: 'Error: Write 0x00158d00052bfccc/2 aqaraOpple({"512":{"value":0,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null}) failed (Timeout - 13794 - 2 - 18 - 64704 - 4 after 10000ms)'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'operation_mode' to 'KIDSROOM_SWITCH' failed: 'Error: Write 0x00158d00052bfccc/2 aqaraOpple({\"512\":{\"value\":0,\"type\":32}}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null}) failed (Timeout - 13794 - 2 - 18 - 64704 - 4 after 10000ms)'","meta":{"friendly_name":"KIDSROOM_SWITCH"},"type":"zigbee_publish_error"}'

Amarrant, 2020-10-16:
I tried to add to toZigbee.js:

` xiaomi_switch_operation_mode: {
        key: ['operation_mode'],
        convertSet: async (entity, key, value, meta) => {
            if (['QBKG11LM', 'QBKG04LM', 'QBKG03LM', 'QBKG12LM', 'QBKG21LM', 'QBKG22LM', 'QBKG24LM'].includes(meta.mapped.model)) {
...
`

This helped. Switch turns to decoupled mode, but buttons don't work. No messages in logs about pressed buttons.

Koenkk, 2020-10-16:
@Amarrant in the debug log, do you see someting like `Recieved zigbee...` message when pressing the buttons?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Amarrant, 2020-10-16:
@Koenkk yep. While pressing decoupled button

`Zigbee2MQTT:debug 2020-10-16 20:50:02: Received Zigbee message from 'KIDSROOM_SWITCH', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
Zigbee2MQTT:debug 2020-10-16 20:50:02: No converter available for 'QBKG24LM' with cluster 'genMultistateInput' and type 'attributeReport' and data '{"presentValue":1}'
Zigbee2MQTT:debug 2020-10-16 20:50:03: Received Zigbee message from 'KIDSROOM_SWITCH', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
Zigbee2MQTT:debug 2020-10-16 20:50:03: No converter available for 'QBKG24LM' with cluster 'genMultistateInput' and type 'attributeReport' and data '{"presentValue":1}'
Zigbee2MQTT:debug 2020-10-16 20:50:05: Received Zigbee message from 'KIDSROOM_SWITCH', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
Zigbee2MQTT:debug 2020-10-16 20:50:05: No converter available for 'QBKG24LM' with cluster 'genMultistateInput' and type 'attributeReport' and data '{"presentValue":1}'
Zigbee2MQTT:debug 2020-10-16 20:50:06: Received Zigbee message from 'KIDSROOM_SWITCH', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
Zigbee2MQTT:debug 2020-10-16 20:50:06: No converter available for 'QBKG24LM' with cluster 'genMultistateInput' and type 'attributeReport' and data '{"presentValue":1}'
Zigbee2MQTT:debug 2020-10-16 20:50:07: Received Zigbee message from 'KIDSROOM_SWITCH', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
Zigbee2MQTT:debug 2020-10-16 20:50:07: No converter available for 'QBKG24LM' with cluster 'genMultistateInput' and type 'attributeReport' and data '{"presentValue":1}'
`

Koenkk, 2020-10-16:
Check with latest changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/1672/files again.

Amarrant, 2020-10-16:
YES! It works.
Supports single and double click. I think, it can send long press, but z2m shows in logs:

`Zigbee2MQTT:debug 2020-10-16 21:07:42: Received Zigbee message from 'KIDSROOM_SWITCH', type 'attributeReport', cluster 'genBasic', data '{"65520":{"data":[170,16,5,65,135,1,1,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-10-16 21:07:42: No converter available for 'QBKG24LM' with cluster 'genBasic' and type 'attributeReport' and data '{"65520":{"data":[170,16,5,65,135,1,1,16,0],"type":"Buffer"}}'
`

While long pressing button

Koenkk, 2020-10-17:
@Amarrant I'm not sure about the long press, usually the `genBasic` messages indicate something else (but it hard to know what exactly). I've merged the other changes.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-17 by Koenkk