Koenkk/zigbee2mqtt #4686: switching lights a group of IKEA devices using IKEA remote does not update status of IKEA lights

heggink, 2020-10-16:
When controlling a group of IKEA devices through an IKEA remote bound to that group, the status of an IKEA plug is updated but no status update for the light

## What happened
Only the plug status changes, not the light status

## What did you expect to happen
All devices that are in the group report their status change so also the light, not just the plug

## How to reproduce it (minimal and precise)
Pair an ikea light LED1537R6/LED1739R5
Pair an ikea power plug IKEA E1603/E1702
Pair a remote IKEA E1743
Ungroup the remote from its default group
Create a group with the plug and the light
Bind the remote to that group
Switch the status of the group and observe the members being changed BUT only see the plug being updated through zigbee2mqtt, not the light

## Debug info
Zigbee2MQTT version: latest stable (1.15.0)
Adapter hardware: zig-a-zig-ah
Adapter firmware version: latest stable

  '5':
    friendly_name: hestudygroup_ct
    devices:
      - 0x680ae2fffe2b8638/1
      - 0x588e81fffe658a54/1

  '0x588e81fffe658a54':
    friendly_name: spare ikea plug

  '0x680ae2fffe2b8638':
    friendly_name: hestudy

okt 16 20:52:56 domoticz npm[521]: Zigbee2MQTT:debug 2020-10-16 20:52:56: Received Zigbee message from 'onoff1', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 5
okt 16 20:52:56 domoticz npm[521]: Zigbee2MQTT:info  2020-10-16 20:52:56: MQTT publish: topic 'zigbee2mqtt/onoff1', payload '{"action":"off","click":"off","linkquality":72,"update_available":true}'
okt 16 20:52:56 domoticz npm[521]: Zigbee2MQTT:debug 2020-10-16 20:52:56: Received Zigbee message from 'spare ikea plug', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
okt 16 20:52:56 domoticz npm[521]: Zigbee2MQTT:info  2020-10-16 20:52:56: MQTT publish: topic 'zigbee2mqtt/hestudygroup_ct', payload '{"brightness":247,"color_temp":376,"state":"OFF"}'
okt 16 20:52:56 domoticz npm[521]: Zigbee2MQTT:info  2020-10-16 20:52:56: MQTT publish: topic 'zigbee2mqtt/spare ikea plug', payload '{"brightness":247,"color_temp":376,"linkquality":63,"state":"OFF","update_available":true}'

tabnul, 2020-10-16:
Do you have the report setting on true? It works for me.

I did have this issue with the cheap zigbee sticks cc2531 with source routing firmware.
Doesnt occur anymore with Cc1352p-2
report: true

heggink, 2020-10-16:
Ah, brilliant. My bad! Works great now! Many thanks!

closed 2020-10-16 by heggink