Koenkk/zigbee2mqtt #4687: database.db overwritten after data migration

tabnul, 2020-10-16:
I am trying to migrate from hass.io to docker.
version 1.15 to 1.15
I copied over devices.yaml, groups.yaml, database.db, state.json
Identical configuration.yaml.

Every time, after starting zigbee2mqtt, database.db gets overwritten with an empty one.
Logs show nothing special. Only that 0 devices are joined.

I allready tried fixing permissions and such, nothing helps. Whatever i do, database.db gets overwritten on startup..

Did i miss something?

tabnul, 2020-10-17:
i joined all devices manually and now the file persists.. strange issue!

closed 2020-10-17 by tabnul

masantiago, 2021-04-03:
I have exactly the same issue, and it is frustrating because I cannot recover my database :-S I'm using the container as well. How can I restore my previous data? I'm even using the latest 1.17 and 1.18 versions.