Koenkk/zigbee2mqtt #4688: Aqara door sensors not updating

dennisdek, 2020-10-17:
## What happened
When I open a door that has an Aqara door sensor attached, this event doesn't register in Home Assistant, nor in the Z2M log file.
When I close the door, that event registers, but that means that the status remains closed, so the fact that the door opened in between isn't registered. Similar, when a door has been open for a long time, closing it doesn't register.
The second time I try it, it does register.
My suspicion is that the sensor is asleep, and therefore doesn't respond, but the second time I try it the sensor is awake, and therefore it registers.
This behaviour only started a couple of weeks ago, and applies to all my door sensors (15 of them). Until then they worked without issues for over a year.
I have replaced the battery, and I have repaired devices, but that didn't fix the issue.

## What did you expect to happen
the opening of the door was expected to register in Z2M and HA

## How to reproduce it (minimal and precise)
Trigger the door sensor after it hasn't been used in a while.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC26X2R1
Adapter firmware version:
20191106, transportrev: 2, type zStack3x0
Total devices: 52, including 3 Ikea repeaters and 1 aqara plug (acting as repeater)

Koenkk, 2020-10-17:
Would you be able to sniff the traffic when you open the door and the sensor does not send anything? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

dennisdek, 2020-10-17:
Hi Koen, if I read the link correctly this requires a CC2531 stick or a HUSBZB-1 stick, I have neither of those available.
I'm using a CC26X2R1 development board, can I use that for it too?
and if I flash the sniffer firmware on that, I assume that the normal firmware gets overridden? in other words, it won't work as a normal z2m controller anymore?

dennisdek, 2020-10-17:
Hi Koen,
I have pasted my log here: https://pastebin.com/5SsRL9bd
I restarted Z2M to make the log as small as possible, and then opened & closed the door twice.
entity name is: z2m_door_garage_inside

First time I opened & closed the door:
- there is no entry for the initial opening
- at 2020-10-18 08:57:13 you'll see an entry for the door closing (contact= true)

right after that I tried again:
- 2020-10-18 08:57:16 - contact=false, so door is open
- 2020-10-18 08:57:23 - contact = true, door is closed again

I hope this information is helpful

Koenkk, 2020-10-18:
@dennisdek you indeed need a separate adapter to sniff. I'm not sure what causes it but I doubt it is solvable from Zigbee2MQTT. I think the issue is caused by either a bad range of the door sensor or there is some problem with the router where the device is connected to.

dennisdek, 2020-10-19:
Cool. 
I'll restart all my repeaters and keep an eye on it to see if that has any effects. 
Thanks for the suggestion!

dennisdek, 2020-10-19:
Hi Koen, thanks for the suggestion; looks like it was indeed the router / repeater - likely the plug.
I restarted all of them and monitored for a day, everything seems to be working now.
Apologies for wasting your time for what turned out to be a local problem!

closed 2020-10-19 by dennisdek