Koenkk/zigbee2mqtt #4690: Ubisys S1/S2 could also report haElectricalMeasurement

sjorge, 2020-10-17:
## Is your feature request related to a problem? Please describe
While browsing the S1 technical spec I found out it can also report data from haElectricalMeasurement cluster and other values from the saMetering cluster.

It would be nice to report on a few and have the option to use a get call to get the some of the others.

[image]
[image]

## Describe the solution you'd like
Having reporting for rmsVoltage, rmsCurrent and activePower would be nice.

Being able to manually read currentMaxReceived and currentMaxDelviered.

## Describe alternatives you've considered
Had a go at trying to configure it myself but was met with:

```
Zigbee2MQTT:error 2020-10-17 10:41:51: Failed to configure 'office/lights/switch', attempt 3 (Error: Read 0x001fee0000005e6c/3 haElectricalMeasurement(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,
"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')                                                                                                                       
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:228:26)
    at Generator.next (<anonymous>)                                                                                                                                                              
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))    
```

I had added `haElectricalMeasurement` to S1's bind for ep3 and tried a readEletricalMeasurementPowerConverterAttributes and bind of rmsCurrent + rmsVoltage.

It looks to fail on the readEletricalMeasurementPowerConverterAttributes.

## Additional context
Add any other context or screenshots about the feature request here.

Koenkk, 2020-10-17:
Leave out `readEletricalMeasurementPowerConverterAttributes` and check if the reported values make sense. If not you can manually tweak the divisors through e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/9638565f046b3a65329db8b67328c2e932fe4433/devices.js#L1732

sjorge, 2020-10-17:
```
Zigbee2MQTT:error 2020-10-17 15:26:18: Failed to configure 'office/lights/switch', attempt 3 (Error: ConfigureReporting 0x001fee0000005e6c/3 haElectricalMeasurement
([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery
":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)                                                    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:386:26)                                                    at Generator.next (<anonymous>)                                                                                                                                     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
```

Which contradicts the tech docs :/

Currently this is my entry:
```
    {
        zigbeeModel: ['S1 (5501)', 'S1-R (5601)'],
        model: 'S1',
        vendor: 'Ubisys',
        description: 'Power switch S1',
        supports: 'on/off, power measurement',
        exposes: [exposes.switch(), exposes.numeric('power').withUnit('W')],
        fromZigbee: [fz.on_off, fz.metering_power],
        toZigbee: [tz.on_off, tz.ubisys_device_setup],
        meta: {configureKey: 4},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(3);
            await bind(endpoint, coordinatorEndpoint, ['seMetering', 'haElectricalMeasurement']);
            await readMeteringPowerConverterAttributes(endpoint);
            await configureReporting.instantaneousDemand(endpoint);
            await configureReporting.rmsVoltage(endpoint);
            await configureReporting.rmsCurrent(endpoint);
            await configureReporting.activePower(endpoint);
        },
        ota: ota.ubisys,
    },

```

sjorge, 2020-10-17:
I'm actually blind, only instantaneousDemand is reportable, the others are read-only... just leaving the bind and adding the correct fromZigbee should be able to manually get them I think. Going to try that.

sjorge, 2020-10-17:
I also can't seem to read them at all, probably a reason why it was never added in the past. :(

closed 2020-10-17 by sjorge