Koenkk/zigbee2mqtt #4692: E1743 Tradfri ON/OFF switch off sync with Group

Anjerlaan, 2020-10-17:
## What happened
I have created a group "Stopcontacten" and added a device "Stopcontact_1" (E1603/E1702 Tradfri control outlet).
Then I did bind succesfully a small remote E1743 Tradfri ON/OFF switch to this group.
But when I see the log it looks like the Remote is off sync with the group "Stopcontacten":

```
debug 2020-10-17 14:46:40: Received Zigbee message from 'Small_Remote_1', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 300
info  2020-10-17 14:46:40: MQTT publish: topic 'zigbee2mqtt/Small_Remote_1', payload '{"action":"off","click":"off","linkquality":51,"update_available":false}'
debug 2020-10-17 14:46:40: Received MQTT message on 'zigbee2mqtt/Stopcontact_1/set' with data '{"state": "ON"}'
debug 2020-10-17 14:46:40: Publishing 'set' 'state' to 'Stopcontact_1'
info  2020-10-17 14:46:40: MQTT publish: topic 'zigbee2mqtt/Stopcontacten', payload '{"brightness":254,"state":"ON"}'
info  2020-10-17 14:46:40: MQTT publish: topic 'zigbee2mqtt/Stopcontact_1', payload '{"brightness":254,"linkquality":81,"state":"ON","update_available":false}'
debug 2020-10-17 14:46:40: Received Zigbee message from 'Stopcontact_1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-10-17 14:46:40: MQTT publish: topic 'zigbee2mqtt/Stopcontacten', payload '{"brightness":254,"state":"ON"}'
info  2020-10-17 14:46:40: MQTT publish: topic 'zigbee2mqtt/Stopcontact_1', payload '{"brightness":254,"linkquality":81,"state":"ON","update_available":false}'
```

## What did you expect to happen
That the remote is in sync with the Group

## Debug info
Zigbee2MQTT version: v.1.15.0 
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200925

Koenkk, 2020-10-17:
Check https://www.zigbee2mqtt.io/information/report.html

Anjerlaan, 2020-10-17:
> 
> 
> Check https://www.zigbee2mqtt.io/information/report.html

I figured it out, in Domoticz I have created a sync script between Group/Stopcontact_1 and the Remote.
If the switch "Stopcontact_1" is turned on/off in Domoticz (not with the phisical Remote) then the Remote ON/OFF switch must be synced with the Stopcontact_1 switch to avoid sync issues

I copied this from another sync script I have, but this is for the IKEA E1524-E1810 remote, which is a Toggle Device and not an ON/OFF switch.

Now I have it working properly

Anjerlaan, 2020-10-19:
Hi,

@Koenkk 
I still have problems to understand the Remote.
This device is called an "IKEA TRADFRI ON/OFF Switch".
Why does it not respond (or update the state) to ON/OFF commands then?

closed 2020-11-26 by github-actions[bot]