Koenkk/zigbee2mqtt #4693: New added device battery-powered becomes unavailable

Hacker-CB, 2020-10-17:
## What happened

New added device with battery power becomes unavailable when availablity checking is enabled.

## What did you expect to happen

z2m should push `true` to `zigbee2mqtt/<DEVICE>/availability`

## How to reproduce it (minimal and precise)

Enable availablity checking.
```
advanced:
  availability_timeout: 60
```
Join any battery-powered device. It stays unavailable until z2m restart.
## Debug info
Zigbee2MQTT version: dev
Adapter hardware: CC2652
Adapter firmware version: https://github.com/jethome-ru/zigbee-firmware/blob/master/ti/coordinator/cc2538_cc2592/JH_2538_2592_ZNP_USB_20201010.hex

Koenkk, 2020-10-17:
Can you provide the debug logging of this? Zigbee2MQTT should immediately publish availability online when a battery powered device joins.

Hacker-CB, 2020-10-19:
I checked on b9662ed with fresh install and all works fine.
Before I used 9e4e8acdb0ca937f3b361533f7bfd6ab4492eafa with 120+ devices and this problem was no it. 
I will check with latest dev in large network and reopen ticket if problem will be reproduced again.

closed 2020-10-19 by Hacker-CB