Koenkk/zigbee2mqtt #4694: Hue connectivity issues

Koenkk, 2020-10-17:
It seems that still some users experience connectivity issues with Hue devices (especially the end devices like Hue dimmer remote and the motion sensors). While the issue of devices completely leaving seems to occur much less (#2693) the famous red LED still occurs (also for my Hue outdoor motion sensor). 

The red LED will blink if the device did not receive an acknowledgment from the coordinator after it sends a command (default response in Zigbee terminology). I believe this happens when the Hue end device switches parent while the coordinator still has an old route to the end device. Eventually the coordinator recovers, but this will take a few seconds which is too late for these Hue devices causing the red LED to blink.

The latest dev branch contains a possible fix, it tries to detect when a Hue device switches parent and immediately rediscovers the new route after that. 

If with the latest dev branch people still experience we need to herdsman debug logging combined with a sniff at the moment the red LED blinks:
- https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
- To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

timota, 2020-11-13:
Issue  still persist with the latest dev branch and latest FW 
SW: `1.16.1-dev commit: ca33e22`
FW: `20201026 `
Coordinator: ` cc1352p-2`

timota, 2020-11-15:
Some info regarding sensor.
When sensor disappeared from network -  5 min later i just allowed joined (Permit Join) and it back again. 
Hope this cab be helpful. Unfortunately no debug.

timota, 2020-11-27:
Hi @Koenkk,
finally i have managed to catch debug for sensor. (link below) It contains 2000 lines before leave message and 500 after, hope this is enough for debug. Sensor is 0x001788010647face

[Debug](https://pastebin.com/vpUU67k5)

Please let me know if you need more info.

Thanks in advance.

Koenkk, 2020-11-27:
@timota thanks for the good logs. I think the problem here is that the motion sensor switches parent, sends a message to the coordinator, the coordinator fails to acknowledge because it still think it has a different parent, the recovery takes too long making the motion sensor start rejoining/leaving. Looking at the logs we cannot fix this from zigbee2mqtt since we know too late if the acknowledgement fails and we have no indication that the motion sensor switched parent. That means it has to be fixed in the coordinator firmware. 

I know this is hard but would you be able to create a sniff from the working situation till it stops working? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

timota, 2020-11-29:
@Koenkk 
Sorry for late response. Debug and sniffer have been started. As soon as sensor fails i will post my findings. 

Thanks

timota, 2020-12-03:
@Koenkk, finally i managed to get dump (hope i sniffed traffic correctly).
So, some background info:
```
issue happened: 2020-12-02T17:51:36.233Z zigbee-herdsman:controller:log Device leave '0x001788010647face'
```
Debug is huge (201m) so can't put it on pastebin - sharing google drive link
[sniff](https://drive.google.com/drive/folders/1C2dhiYYjuS2U2IyMqmQ_I5orb7SCZ7PP?usp=sharing)

If you will need zigbee2mqtt logs for this sniff - let me know.

Hope this helps. Thank.

Koenkk, 2020-12-03:
The problem seems to be that the **coordinator** sends a leave request to the motion sensor with rejoin set to **false**:

[image]

Not sure why it happens, but can you check if it is fixed with the following firmware: 
[znp_CC1352P_2_LAUNCHXL_20201203_skip_leave.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/5638897/znp_CC1352P_2_LAUNCHXL_20201203_skip_leave.hex.zip)

timota, 2020-12-04:
FW installed. Container started with Full debug. 

Will let you know results. Hopefully it will work as expected, fingers crossed.

netweaver1970, 2020-12-04:
Fingers crossed, hoping for good results at your side so the FW can be fully released. Thanks both for the investigation/fixing time !

netweaver1970, 2020-12-14:
@timota No news is good news? No funny things happening/happened and no drop-out of the Hue motion sensor anymore?

0rsa, 2020-12-21:
Unreliable here

warn  2020-12-20 19:37:47: Device '_ID_HIDDEN_' left the network
info  2020-12-20 19:37:48: Device 'Sensor living room' joined
info  2020-12-20 19:37:48: Starting interview of 'Sensor living room'
info  2020-12-20 19:37:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Sensor living room"},"type":"device_connected"}'
info  2020-12-20 19:37:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Sensor living room"},"type":"pairing"}'
error 2020-12-20 19:40:40: Failed to interview 'Sensor living room', device has not successfully been paired

I use Electrolama zig-a-zig-ah! with official CC26X2R1_20201026.hex firmware
Zigbee2mqtt 1.16.2

timota, 2020-12-21:
@netweaver1970  @Koenkk 
Morning guys, 

sorry for a long response, was busy...
During my testing with a new firmware i didn't see any disruptions. All sensors, including Hue are working as expected. Before consider fw as stable i would like to do another final test - move sensor outdoor as it should be and connect it via router. I'm going to do that today. I will let you know, hope to the end of this week, usually, before it (hue) fails at 2-3 days. 

Regards.

ellnic, 2020-12-22:
I have the LAUNCHXL-CC26X2R1 and am experiencing this issue. Dinners I've never had a problem with, but motion sensors are the bane of my life right now. 

It wasn't really a problem before the last 2 months or so, but I have added a lot more routers and this seems to tie in with what is being described here. Before they were most likely directly connected to the coordinator, now they are not. 

It'll be good to see if this fixes the issue 😃

ChessSpider, 2020-12-23:
I have CC2530_CC2591_20190523 zigbee 3.0 firmware now, and i sometimes experience the same problem.  Unsure if it also has the same cause. Anyway, new firmware is always nice

0rsa, 2021-01-02:
I can see this fix in latest release: #1925 #2274 Xiaomi Motion Sensor RTCGQ11LM not reporting occupancy true
I have exactly the same behaviour for Hue sensors.
After a couple of day, one sensor (random one) is sending occupancy: false even if the even is triggered by a motion.
I give up, I purchased and I'm waiting for Xiaomi sensors working with gateway to remove zigbee2mqtt from my system.

podi62, 2021-01-04:
Any news on this? @Koenkk, if patch znp_CC1352P_2_LAUNCHXL_20201203_skip_leave.hex.zip is working are you going to port this on the main CC fw and on the Source Routing version? Thx

Koenkk, 2021-01-04:
@timota is it working correctly now?

@podi62 yes but it needs more testing (first CC2652, after that CC2531)

foxylion, 2021-01-04:
I have a [CC2652RB](https://slae.sh/projects/cc2652/) and the same symptoms with my hue dimmers.

So I could est the CC2652RB firmware. Is there a prebuilt one? And a tutorial how to flash the new firmware?

Koenkk, 2021-01-06:
Issue doesn't seem to be fixed, got a red light today on my hue motion outdoor sensor. 99% sure the issue has to be fixed in the firmware. Asked TI for support: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/968896

TheJulianJES, 2021-01-06:
Same issue also happens on ZHA (HomeAssistant) sometimes. So it doesn't look like it's a bug in either ZHA / Z2M.

Koenkk, 2021-01-08:
TI has provided a potential fix, link to new firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0

ellnic, 2021-01-08:
I will give this a go and report back ASAP. My drop offs were every week or so. We'll see how this goes! :-) thanks

TheJulianJES, 2021-01-09:
Upgraded yesterday and today I experienced a red blink on my Hue outdoor motion sensor (message still seems to have arrived though). Motion sensor is inside (for testing) and does one hop through a "ZB3" Hue light. (Using ZHA)
Does the potential fix work for others? (Thanks, Koenkk btw!)

sjorge, 2021-01-09:
> Issue doesn't seem to be fixed, got a red light today on my hue motion outdoor sensor. 99% sure the issue has to be fixed in the firmware. Asked TI for support: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/968896

Ah good I'm not the only one, I started having it again since switching to the zzh-lite.

TheJulianJES, 2021-01-09:
> Ah good I'm not the only one, I started having it again since switching to the zzh-lite.

You're also running the 20210107 version from the develop branch, I guess?

sjorge, 2021-01-09:
> You're also running the 20210107 version from the develop branch, I guess?

Correct

ellnic, 2021-01-12:
Same here. Still experiencing the same.

peter-dolkens, 2021-01-16:
> The problem seems to be that the **coordinator** sends a leave request to the motion sensor with rejoin set to **false**:
>
> [image]
>
> Not sure why it happens, but can you check if it is fixed with the [...]

Any chance you've got a version of this for CC2652RB?

TheJulianJES, 2021-01-23:
> Upgraded yesterday and today I experienced a red (long) blink on my Hue outdoor motion sensor (message still seems to have arrived though). Motion sensor is inside (for testing) and does one hop through a "ZB3" Hue light. (Using ZHA)
> Does the potential fix work for others? (Thanks, Koenkk btw!)

Although it may not be related to the firmware update, it seems to be a bit better. I had less red blinking on my (testing) outdoor motion sensor. Can anyone report if it (mostly) "fixed" itself after some time with the latest firmware perhaps?

ellnic, 2021-01-24:
@TheJulianJES from my end, I thought it was about the same. But last night I lost connection entirely to an outdoor sensor I've never had problems with before. Could be coincidence but it might be slightly worse.

Koenkk, 2021-01-24:
In case this happens, please try to capture a sniff of this + herdsman debug logging (otherwise I cannot debug this further).

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

ellnic, 2021-01-26:
I've ordered a CC2531. I will try to get a sniff when it's here.

foxylion, 2021-01-26:
> I've ordered a CC2531. I will try to get a sniff when it's here.

I did the same, but will take a while until it is delivered... ;-)

ellnic, 2021-02-04:
CC2531 is here! Sniff started, is there some way to filter out helpful information? Or just submit entire sniff once issue occurs?

Koenkk, 2021-02-04:
@ellnic just share the whole sniff.

I'm starting to think this issue also depends on the network setup. Since I removed all my CC2530 routers from my network (replaced ptvo CC2530 router with a CC2652R and a Gledopto led controller with a Tuya one) I never saw this issue again.

sjorge, 2021-02-04:
Interesting, I do think mine is connected over an Ubisys S1 which use CC253x IIRC. (I already found a few bugs in there firmware that I reported to them, so having more lurk in there wouldn't surprise me)

ellnic, 2021-02-04:
Ok will do. Btw, I noticed this:

`3352	161.626585	0x7ffd	0x0000	ZigBee HA	58	Unknown Command: 0x00, Seq: 80`

when pressing a button on a Hue dimmer. I've never had a problem with my Hue dimmers, but that doesn't look like it should happen...

Koenkk, 2021-02-04:
@ellnic that's expected, the hue dimmer sends manufacturer specific commands (not defined in the ZCL spec so wireshark will show them as unknown)

ellnic, 2021-02-04:
I'm running into an issue where Wireshark stops capturing after about 5 minutes. I have repeated this several times, tried a different USB port etc. Is there an obvious setting I have missed that is causing this or should I just jump to a different machine?

Koenkk, 2021-02-04:
Is `modemmanager` installed? https://www.zigbee2mqtt.io/information/FAQ.html#modemmanager-is-installed

ellnic, 2021-02-04:
It was, but I have removed it and the problem persists. I will fire up a minimal Debian box tomorrow and try again.

ellnic, 2021-02-06:
Phew.. well it turns out that (after a lot of head scratching, service disabling and at a last resort different machines/distros) the problem is this:

`whsniff -c ZIGBEE_CHANNEL_NUMBER | wireshark -k -i -`

For some reason, Debian 10 and Ubuntu 18.04 do not like the way this is formatted and ultimately fail after 300-40K packets, for what reason I am not sure but the pipe breaks. 

The error `end of file on pipe magic during open` when attempting to resume wireshark by clicking the fin icon, lead me to [here](https://github.com/GNS3/gns3-gui/issues/3021) and in turn to the whsniff project page [here](https://github.com/homewsn/whsniff) where it suggests also using:

`wireshark -k -i <( path/to/whsniff -c channel_number )`

So far so good, we're at 45K packets. If this hasn't failed by this evening, I'll PR some changes to the docs.

ellnic, 2021-02-07:
@Koenkk still stable and at 550K packets, so have opened PR https://github.com/Koenkk/zigbee2mqtt.io/pull/611

ellnic, 2021-03-10:
I'm still capturing. Typical, one month and no drop offs yet! 🤨

Koenkk, 2021-03-10:
@ellnic let's see if we can catch a drop

ellnic, 2021-03-11:
@Koenkk Ok, no problem. I've saved the [massive] cap file up until now and started a fresh one. As soon as I have something I'll post back.

audiobah, 2021-03-11:
@Koenkk Some information about my experience with Hue outdoor sensor. My coordinator is CC2538. I can connect the sensor only if I am near the coordinator. Then I go outside of the house to install it on another building, there is a lot of routers there nearby but sensor (I guess) do not see signal from coordinator and not trying to connect to any of the routers. And then I see that sensor has left the network.

cmorlok, 2021-03-11:
I have the very same issue with the Hue Outdoor Sensor. After pairing it, it either remains in the network but stops reporting motion and temperature. Or it leaves the network very soon. I can enforce it to leave for example by starting an OTA update.

SW: `1.18.1`
FW: `JH_2538_2592_ZNP_USB_20201010`
Coordinator: `cc2538`

frank3523, 2021-03-18:
same issue here i switched from cc2531 to cc2652rb from slae but no luck.
Zigbee2mqtt detects the sensor but no motion detection.
motion, temp, illuminance all is unknown.

cmorlok, 2021-03-18:
@frank3523 Which FW are you using on cc2652?

frank3523, 2021-03-18:
The latest from Koenk

frank3523, 2021-03-18:
Again removed the hue motion sensor restarted zigbee2mqtt and suddenly its working again.
Strange behaviour all my other devices ikea gledopto worked right away.

cmorlok, 2021-03-18:
@frank3523 latest means CC2652RB_coordinator_20210120? I was hoping that the fix would work (https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/CHANGELOG.md#20210120)

frank3523, 2021-03-19:
Yes that is the firmware is am using. But the device started working again suddenly.

ASNNetworks, 2021-03-28:
SO I'm going batshit crazy for the last 2 months. I'm not sure if I'm having the same issues, or just similar. All of my sensors (8 Hue motion sensors) randomly refuse to send out either the `occupied` or `clear` signal. The sensor itself does register motion internally, since I see that the led turns red briefly. Sometimes I have no issues for a few days. Then sometimes I have a few sensors acting up throughout the day multiple times.mThis results in either:

**z2m not seeing `occupied`**. If I keep moving, it remains `clear`. Since the Hue sensor still sees me, it doesn't send out a new command. Resulting in a stuck `clear` state. If I stand still for 10 seconds and then start moving, it does send out the signal `occupied` signal, since it sets `clear` internally.

**z2m not seeing `clear`**. If I stay out of sight, the state will be occupied indefinitely. A reboot of z2m or HA doesn't matter. It remains at the `occupied` state till the ends of time. If I however enter the room, and **then** leave, it will send out the `clear` signal. So exactly the same, but in reverse. Since it sees me, it will internally send out the `occupied` command, after which it will clear after 10s of no motion.

I have used **Conbee II** through an usb 2,0 extension cord, connected to the usb 2.0 port. Using a Raspberry Pi 4, with a USB 3.0 SSD drive for boot. Hence using a 2.0 extension cord on the 2.0 port, to prevent the interference. Using the latest firmware.

I just got a **zig-a-zig-ah!** stick and unfortunately, experience the exact same issues. Having tried directly on the usb 3.0 port and also using an extension cord to 2.0 port. Using the latest firmware.

I really hoped it was due to the experimental support of Conbee II. Sad to see that after re-pairing my 79 devices, the issue still remains. Having this issue ever since I got Hue motion sensors (for around 2 months), tried all latest Z2M updates + dev builds. Currently using latest dev branch (installed through add-on store on HA).

If there is anything I can do/try/report I'm more than happy to help.

One thing I notice, is that the z2m logs get flooded with logs from all my devices. This is in 10 seconds, without doing anything. I'm not changing any states of my lights, just sitting behind my desk, without anyone at home.

```
MQTT publish: topic 'zigbee2mqtt/Kantoor Lamp Plafond 3', payload '{"brightness":124,"color":{"x":0.615,"y":0.372},"color_temp":298,"last_seen":"2021-03-28T21:49:19+02:00","linkquality":39,"state":"ON","update":{"state":"idle"},"update_available":false}'
MQTT publish: topic 'zigbee2mqtt/Woonkamer Z Lampen', payload '{"brightness":124,"color":{"hue":26,"saturation":94,"x":0.615,"y":0.372},"color_mode":"color_temp","color_temp":800,"state":"ON"}'
MQTT publish: topic 'zigbee2mqtt/Woonkamer Salon Ledstrip TV', payload '{"brightness":124,"color":{"hue":26,"saturation":94,"x":0.615,"y":0.372},"color_mode":"xy","color_temp":800,"last_seen":"2021-03-28T21:49:24+02:00","linkquality":162,"state":"ON","update":{"state":"idle"},"update_available":false}'
MQTT publish: topic 'zigbee2mqtt/Kantoor LEDStrip Bureau', payload '{"brightness":124,"color":{"x":0.615,"y":0.372},"color_mode":"xy","color_temp":800,"last_seen":"2021-03-28T21:49:25+02:00","linkquality":36,"state":"ON","update":{"state":"idle"},"update_available":false}'
MQTT publish: topic 'zigbee2mqtt/Woonkamer Salon Schakelaar TV', payload '{"current":0,"energy":96.97,"last_seen":"2021-03-28T21:49:27+02:00","linkquality":162,"power":0,"power_outage_memory":"restore","state":"ON","voltage":238}'
MQTT publish: topic 'zigbee2mqtt/Badkamer Schakelaar Wasmachine', payload '{"current":0.04,"energy":11.13,"last_seen":"2021-03-28T21:49:27+02:00","linkquality":69,"power":0,"power_outage_memory":"on","state":"ON","voltage":235}'
MQTT publish: topic 'zigbee2mqtt/Kantoor Schakelaar PC', payload '{"current":0.61,"energy":256.95,"last_seen":"2021-03-28T21:49:27+02:00","linkquality":69,"power":121,"power_outage_memory":"restore","state":"ON","voltage":235}'
MQTT publish: topic 'zigbee2mqtt/Slaapkamer Z Lampen', payload '{"brightness":1,"color":{"hue":354,"saturation":95,"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":500,"state":"OFF"}'
MQTT publish: topic 'zigbee2mqtt/Slaapkamer Lamp Nachtkast 2', payload '{"brightness":1,"color":{"hue":354,"saturation":95,"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":500,"last_seen":"2021-03-28T21:49:32+02:00","linkquality":72,"state":"OFF","update":{"state":"idle"},"update_available":false}'
MQTT publish: topic 'zigbee2mqtt/Gang Lamp Plafond 2', payload '{"brightness":0,"color":{"x":0.5071,"y":0.4178},"color_mode":"color_temp","color_temp":454,"last_seen":"2021-03-28T21:49:33+02:00","linkquality":84,"state":"OFF"}'
MQTT publish: topic 'zigbee2mqtt/Kantoor Lamp Plafond', payload '{"brightness":123,"color":{"x":0.615,"y":0.372},"color_temp":298,"state":"ON"}'
MQTT publish: topic 'zigbee2mqtt/Kantoor Lamp Plafond 1', payload '{"brightness":123,"color":{"x":0.615,"y":0.372},"color_temp":298,"last_seen":"2021-03-28T21:49:34+02:00","linkquality":126,"state":"ON","update":{"state":"idle"},"update_available":false}'
MQTT publish: topic 'zigbee2mqtt/Kantoor Z Lampen', payload '{"brightness":123,"color":{"x":0.615,"y":0.372},"color_mode":"xy","color_temp":800,"state":"ON"}'
[... cut ...]
MQTT publish: topic 'zigbee2mqtt/Gang Lamp Plafond 1', payload '{"brightness":0,"color":{"x":0.5071,"y":0.4178},"color_mode":"color_temp","color_temp":454,"last_seen":"2021-03-28T21:49:37+02:00","linkquality":72,"state":"OFF"}'
MQTT publish: topic 'zigbee2mqtt/Badkamer Lamp Wand 2', payload '{"brightness":0,"color":{"x":0.5071,"y":0.4178},"color_mode":"color_temp","color_temp":454,"last_seen":"2021-03-28T21:49:39+02:00","linkquality":72,"state":"OFF"}'
```

Not sure if this is just normal, or if my zigbee network gets flooded.

ellnic, 2021-04-28:
I'm still going. Really odd. 

This is the longest I've not had an issue, not that I'm complaining. 

I'm happy to keep capturing, but not sure what the cut off point should be? I bet as soon as I stop capturing that something will happen! ;-)

foxylion, 2021-04-28:
I'm in the same situation. Since I flashed the latest firmware (`20210120`) to the coordinator (`CC2652`) I'm no longer seeing any red LEDs or drops of motion sensors from the network.

But I also still capturing traffic in case it drops in the future. :-)

ASNNetworks, 2021-04-28:
I'm running latest firmware on zzh! and still have these issues. However, sometimes the sensor flashes red but does trigger motion.

Also still have the issue it doesn't send the 'off' command after motion is gone. Happens rarely, but is present.

ellnic, 2021-04-28:
I still get red LEDs, but they're not causing any issues. I had an outdoor sensor drop off, but it was when I first started on the newer firmware and I am wondering if this was something else entirely and that the issue we're looking at here is solved. 

I wondered a while back if it could be related to routers, but I have kept my network in an identical state in terms of devices and positioning in order to maintain as fair as possible test environment. 

I'll probably keep capturing for another month or so and then stop if nothing tangible happens. We can always revisit if the situation regresses.

Koenkk, 2021-05-11:
@ellnic you can send them to me on telegram (@koenkk)

[44 more replies]

closed 2022-03-04 by github-actions[bot]