Koenkk/zigbee2mqtt #4701: 'No network route' on manuSpecificTuyaDimmer.setData command

arturdobo, 2020-10-17:
## What happened

I try to figure out why I can't send any message to change any property on Siterwell GS361A-H04 using zzh stick.  The same issue on all three instances. There is no problem regarding reading changes caused by manual activity on mentioned TRV. They're available immediately on mqtt broker.

## What did you expect to happen

## How to reproduce it (minimal and precise)

https://pastebin.com/xJ4gdrsM

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC26X2R1
Adapter firmware version: 20200925

Koenkk, 2020-10-18:
How does your networkmap look?

arturdobo, 2020-10-18:
As simple as that currently, because I wanted to reduce possible errors to minimum.  I'm not sure, why they're shown as not connected to coordinator now, while I''m able to receive any updates from them 🤔  I'm pretty new in this area and maybe some of my questions are rookie-like. 

[image]
[image]

Koenkk, 2020-10-18:
- Did you every upgrade the firmware? (did you pair everything under the `20200925` firmware). 
- Can you try reparing the radiator, does it show a direct connection to the coordinator now? (you should be able to send commands if so).

arturdobo, 2020-10-18:
> Did you every upgrade the firmware? (did you pair everything under the 20200925 firmware).

I received zzh stick couple of days ago, so everything I did was on the newest firmware version

> Can you try reparing the radiator, does it show a direct connection to the coordinator now? (you should be able to send commands if so).

I flashed the stick once again and try to repair all radiators and couple of test sensors. And something wired happened.  I was able to change temperature on `0x14b457fffe3cb140` (the nearest to coordinator radiator). But after couple of minutes the connection between them has been lost. However another one to `0x14b457fffe3cbadc` has been established (the farthermost to coordinator one) and currently I'm still able to manipulate its settings. What do you think, where should I look for issue?

[image]

arturdobo, 2020-10-18:
The last connection has been lost as well. When I switched them all back to pairing mode, they all disappeared from the list of available devices and I had to restart zigbee2mqtt to allow them to reconnect. 

Is it something normal it takes a while the radiator to be shown on Home page even, it's available on network map?

Koenkk, 2020-10-19:
Would you be able to sniff the traffic on a fresh setup with only one thermostat connected? I think the thermostat has a very low poll timeout which makes the coordinator drop the thermostats. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

arturdobo, 2020-10-19:
Please give me couple of days to arrange something

arturdobo, 2020-10-21:
See the captured traffic. It look a while the device to be listed on zigbee2mqtt fronted, but finally I was able to send some command to change temperature. And they were captured as well. But as above, with no luck.

[zb.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/5418497/zb.pcapng.zip)

arturdobo, 2020-10-21:
I've just flashed CC2531 I was using as a sniffer above with coordinator firmware and it works like a charm. All the radiators were discovered almost limitedly  and I'm able to steer them all. So it looks like something is wrong regarding zzh or the firmware 🤔 

[image]

Koenkk, 2020-10-22:
Can you modify `/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js` and change 

```
                    // NWK_NO_ROUTE: no network route => rediscover route
                    // MAC_NO_ACK: route may be corrupted
                    // MAC_TRANSACTION_EXPIRED: Mac layer is sleeping
                    if (!assocRemove && dataConfirmResult === ZnpCommandStatus.MAC_TRANSACTION_EXPIRED &&
                        dataRequestAttempt >= 1 && this.version.product === tstype_1.ZnpVersion.zStack3x0) {
```

to 

```
                    // NWK_NO_ROUTE: no network route => rediscover route
                    // MAC_NO_ACK: route may be corrupted
                    // MAC_TRANSACTION_EXPIRED: Mac layer is sleeping
                    assocRemove = true;
                    if (!assocRemove && dataConfirmResult === ZnpCommandStatus.MAC_TRANSACTION_EXPIRED &&
                        dataRequestAttempt >= 1 && this.version.product === tstype_1.ZnpVersion.zStack3x0) {
```

Does it keep working after this?

arturdobo, 2020-10-22:
I did the change, replace the stick with zzh and when zigbee2mqtt had started I was able to control all radiators. Then I  tried to repair one of the radiators and after 2 or 3 attempts, it's been paired again. I changed the target temperature successfully as well.

Then I reverted the change above, restarted zigbee2mqtt, and tried to repair exactly the same radiator. After the first attempt, it's been paired as an unsupported device. After a second it's been shown as a correct device but with no link established, so I wasn't able to steer it.

Koenkk, 2020-10-23:
@arturdobo so with this change the issue is fixed?

arturdobo, 2020-10-24:
I think so. What the flag I needed to change above is responsible for?

Koenkk, 2020-10-26:
Should be fixed in the latest dev branch in combination with using this firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

Can you check?

arturdobo, 2020-10-28:
It looks like it works after changing firmware and branch do dev. But let me test during a couple of next days. I see there an issue regarding network map drawing, but I'm not sure whether it's just a nature of the dev branch

arturdobo, 2020-11-01:
No issues so far

Koenkk, 2020-11-02:
Great, I assume this solved the issue, if not feel free to reopen!

closed 2020-11-02 by Koenkk

arturdobo, 2020-11-02:
Sure @Koenkk. When will it be merged into master?

Koenkk, 2020-11-02:
@arturdobo its in zigbee2mqtt 1.16.0