Koenkk/zigbee2mqtt #4703: Make non-pingable devices availability timeout configurable

tabnul, 2020-10-18:
## Is your feature request related to a problem? Please describe
Some of my battery powered devices have a rather important role and i know they send data every few hours. I would like to be alerted when they are down for a few hours. However the out-of-the-box availability setting only puts them on 'unavailable' after 25 hours .
From the website;
"
Note that not all devices can be pinged for availability (e.g. battery powered devices). This category of devices will be marked as unavailable when no message has been received from them for 25 hours (persistent across restarts) This is only enabled when the availability feature is enabled.
"

## Describe the solution you'd like
Make the 25 hours configurable.
https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/availability.js

## Describe alternatives you've considered
currently i have binary sensors for every device that check for last_seen and via that route i get alerted.
IT works, but its not maintenance friendly.

tabnul, 2020-10-18:
I do my alerting via Prometheus and i saw that 2 days ago a change for Home Assistant went live that adds last_updated to the prometheus exporter! Meaning that i can fix this issue in a simple prometheus query. 

Still i think this feature request makes sense (hard coded value) .

closed 2020-11-26 by github-actions[bot]

Morphy99, 2021-01-11:
I'd like to see this too. Critical sensors that update once an hour could do with being checked for availability more often.

Koenkk, 2021-01-11:
@Morphy99 I'm currently gathering feedback/suggestions for the availability feature for an upcoming refactor. Please request it here: https://github.com/Koenkk/zigbee2mqtt/issues/5330