Koenkk/zigbee2mqtt #4704: PGC410 (US model) SmartThings Presence sensor will not pair

FoxConsult, 2020-10-18:
## What happened
I have a PGC410, US model (Not the EU model PGC410EU) SmartThings Presence sensor that will not pair.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
I have posted debug log and zigbee-herdsman log as a comment to issue [4055](https://github.com/Koenkk/zigbee2mqtt/issues/4055).

Koenkk, 2020-10-18:
This device requires a special CC2531 firmware, you can use this one: https://github.com/Koenkk/zigbee2mqtt/issues/4632#issuecomment-706691967

FoxConsult, 2020-10-27:
Flashed special CC2531 firmware as per #4632 and the PGC410 now pairs. Thank you.
Will the code in the special CC2531 firmware branch make into the main branch?

Koenkk, 2020-10-28:
@FoxConsult these changes will be integrated in https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2 later

closed 2020-10-28 by Koenkk