Koenkk/zigbee2mqtt #4705: Let esp8266 connect to zigbee2mqtt (instead of connecting zigbee2mqtt to esp8266)

jeroenst, 2020-10-18:
## Is your feature request related to a problem? Please describe
I have remote sites that now connecting with an esp8266 to mosquitto on my server on the internet. It is not possible to let the server connect to the esp8266/zigbee gateway because the clients are behind a firewall and have dynamic IP's.

## Describe the solution you'd like
An ssl server socket on zigbee2mqtt where esp8266 can connect to, authenticate and send the data of zigbee to.

## Describe alternatives you've considered
Build an esp8266 zigbee 2 m2qtt gateway, but I think it is to complex.

## Additional context
[image]

IVI053, 2020-10-18:
I dont get it why this would ever be a considerable solution for any problem. Latency would be horrible or at least unpredictable.

jeroenst, 2020-10-18:
With Tuya everything for domotica goes through China servers (which I don't want) and latency is pretty low.
Also current day's internet in the netherlands is pretty stable (when you have a good provider).
I have already running several esp8266 (reflashed tuya) modules in the field without problem, but more and more tuya devices are removing the esp8266 for a proprietary chipset, to the choice becomes use China servers or no domotica.
That's why my interest has grown for remote zigbee devices which can be controlled by my own server.

danieledwardgeorgehitchcock, 2020-10-19:
This can potentially already be done using any serial zigbee device (not USB) such as the [CC2530](https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc2530-optionally-with-a-cc2591-or-cc2592-rf-frontend) and [Ser2Net](https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html#how-to-connect-to-a-remote-adapter) - This could possibly be piped through to your Zigbee2MQTT instance (although I am not aware of anybody who has tried this).

As mentioned above, I think reliability would be a huge factor in piping something like this over the internet. The Tuya devices mentioned above, convert the raw data into a message format - that is what is sent to Tuya.

A possible alternative would be to create a driver for the CC2530 serial device and use it with something like Tasmota, which has a built in MQTT client which could then send the data to a central MQTT server..?

renne, 2020-10-21:
I also see latency problems with a remote serial port. MQTT via internet should not be a problem.
An interesting question is if zigbee2mqtt runs on an ESP32 with [Espruino](https://www.espruino.com/ESP32).

closed 2020-11-29 by github-actions[bot]