Koenkk/zigbee2mqtt #4706: TRADFRI bulb E27 opal 1000lm does not pair with Zstack3 type?

BoukeHaarsma23, 2020-10-18:
## What happened
When moving from the CC2531 to slaesh’s CC2652RB stick I noticed that 1 bulb type in my network is not able to pair with it. It is the Ikea 1000lm bulb. It is not able to interview the device and does not show in logs. However, interviewing works fine with the CC2531 with zstack 1.2 (source-routing) firmware. The CC2652RB and also the CC2531 with zStack 3.0 firmware fail to interview the device. Other ikea bulbs pair fine. It is just this type that seems to fail. I checked that there are no firmware updates available for the bulb. I seem to have pinpointed it to the zStack version running on the coordinator.
## What did you expect to happen
The bulb to interview.
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531 and CC2652RB
Adapter firmware version:
CC2531 (20190619 - source routing zStack 1.2) works
CC2531 (20190425 - zStack 3.0) not working
CC2652RB (20200830 - zStack) not working

Koenkk, 2020-10-18:
Try to reset the bulb via touchlink (broken in 1.15.0, so use the dev branch(https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

BoukeHaarsma23, 2020-10-19:
I was able to pair it just now after a touchlink reset as suggested. Thanks :+1: 

Closing this

closed 2020-10-19 by BoukeHaarsma23