Koenkk/zigbee2mqtt #4709: Publish 'set' 'state' to '0xbc33acfffe5ec849' failed: 'Error: Command 0xbc33acfffe5ec849/1 genOnOff.on({},

domnich, 2020-10-18:
## What happened
Hello, I have a really small network of zigbee devices (8). 

I have problem with my wall socket:  zigbeeModel: ['TS0012'].

I started to observe strange error after added Router to my network (https://www.zigbee2mqtt.io/devices/TS0121_plug.html) 

The error from my zigbee2mqtt log is located here https://hastebin.com/lulaxomuru.sql.

I found a similar issue https://github.com/Koenkk/zigbee2mqtt/issues/4294. So first what I did was applying  same fix you provided, I  added disableDefaultResponse: true param  to my existing device,  but with out luck...

It also could be that my issue is similar to this one https://github.com/Koenkk/zigbee2mqtt/issues/4392 as here also discussed a Blitzwolf shp-13 power plug router.

And I also know that if I will totally remove my wall socket from mqtt and from zigbee db files and after will pair my wall switch again --> it will work as expected with out this error.
 

How I see it: If socket is broken and to solve the problem I just need to pair it again, could it be that zigbee2mqtt save and store some device related information somewhere in the storage ? 

Maybe the easiest fix for this problem will be just change Blitzwolf shp-13 power plug router to some other, BUT there is no certainty  that this will not happen with another router ..... especially when I have so few devices

Thanks, any information will be useful for me!

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: сс2538 + сс 2592
Adapter firmware version:

Koenkk, 2020-10-18:
Can you post your networkmap?

domnich, 2020-10-19:
[image]

domnich, 2020-10-19:
I power off my Blitzwolf shp-13 for now as I was able to reproduce this issue 3 times to day...

One remark from my side. At the ticket description I wrote next: **And I also know that if I will totally remove my wall socket from mqtt and from zigbee db files and after will pair my wall switch again --> it will work as expected with out this error.**

I was wrong here, to fix this problem I just need to power off and then power on my Blitzwolf shp-13  router

Koenkk, 2020-10-20:
If you could sniff the traffic we can dive deeper into this (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

domnich, 2020-10-21:
Thanks Koenkk. After my previous comment I added property "disableDefaultResponse: ture" into Blitzwolf shp-13 object to the "meta" (in devices.js) and for now everything works Ok...I can even say that my wall sockets now connected to this router. At the moment I can't say for sure that my fix  fixed the problem, as little time has passed. Now I want to watch and if the error will appears again I will necessarily sniff my traffic and provide it to you.

Koenkk could you please share some good resources with detailed info about zigbee device object model (e.g. disableDefaultResponse etc) ? I will be very grateful.

domnich, 2020-10-21:
[image]

Shaquu, 2020-10-21:
@Koenkk I am not sure if I should open a new Issue. I have simillar error:
herdsman logs:
```logs
2020-10-21T13:17:04.098368695Z  2020-10-21T13:17:04.097Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []                                                                                                                                                                                
2020-10-21T13:17:24.577373455Z  2020-10-21T13:17:24.576Z zigbee-herdsman:controller:endpoint Command 0xbc33acfffef4d2e6/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})                                                                                                                                                                                                                                                  
2020-10-21T13:17:24.578576392Z  2020-10-21T13:17:24.577Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xbc33acfffef4d2e6:46526/1 (0,0,1)                                                                                                                                   
2020-10-21T13:17:24.578674594Z  2020-10-21T13:17:24.578Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":46526,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":83,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,58,1]}}                      
2020-10-21T13:17:24.579784810Z  2020-10-21T13:17:24.578Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,190,181,1,1,6,0,83,0,30,3,1,58,1,81]                                                                                                                                     
2020-10-21T13:17:24.582077409Z  2020-10-21T13:17:24.581Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]                                                                                                                                                                         
2020-10-21T13:17:24.582193981Z  2020-10-21T13:17:24.581Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]                                                                                                                                                               
2020-10-21T13:17:24.582431420Z  2020-10-21T13:17:24.581Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100                                                                                                                                                            
2020-10-21T13:17:24.582505400Z  2020-10-21T13:17:24.582Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}                                                                                                                                                                
2020-10-21T13:17:24.582615416Z  2020-10-21T13:17:24.582Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []                                                                                                                                                                                
2020-10-21T13:17:34.583898825Z  2020-10-21T13:17:34.583Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0xbc33acfffef4d2e6:46526,9999,0)                                                                                                                                                
2020-10-21T13:17:34.598129853Z  2020-10-21T13:17:34.583Z zigbee-herdsman:controller:endpoint Command 0xbc33acfffef4d2e6/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'Timeout' (9999))                                                                                                                                                                                        
2020-10-21T13:18:01.293898723Z  2020-10-21T13:18:01.293Z zigbee-herdsman:controller:endpoint Command 0x60a423fffe9d9771/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})                                                                                                                                                                                                                                                 
2020-10-21T13:18:01.294864461Z  2020-10-21T13:18:01.294Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x60a423fffe9d9771:64329/1 (0,0,1)                                                                                                                                   
2020-10-21T13:18:01.296356985Z  2020-10-21T13:18:01.295Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":64329,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":84,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,59,0]}}                      
[... cut ...]
INFO: MQTT publish: topic 'zigbee2mqtt/Aqara Wardrobe Door Sensor', payload '{"battery":100,"contact":false,"device":{"applicationVersion":3,"dateCode":"20161128","friendlyName":"Aqara Wardrobe Door Sensor","hardwareVersion":2,"ieeeAddr":"0x00158d00042d1906","manufacturerID":4151,"manufacturerName":"LUMI","model":"MCCGQ11LM","networkAddress":8937,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"last_seen":1603286586253,"linkquality":94,"voltage":3005}'
```

[image]

Koenkk, 2020-10-22:
@Shaquu when the switches stopped working, do they show no connection on the networkmap? (is it kitchen/bedroom main light switch?)

Shaquu, 2020-10-22:
@Koenkk yes, they are on the map. It is main switch for bedroom and bathroom I believe

Konsts, 2020-10-22:
> @Shaquu, когда переключатели перестали работать, они не показывают соединение на карте сети? (это главный выключатель света в кухне / спальне?)

Confirm? i have same problem with TuYa TS0012 2gang swithes.
The state changes when the switch is pressed
but not controlled remotely
Always occurs after reboot z2m

Shaquu, 2020-10-22:
@Konsts indeed. In my case they are 1 gang.
They work for day or couple days and then stops responding to remote control (via Home.app => NRCHKB => node-red => zigbee2mqtt)
But yes, they are visible everywhere.

Koenkk, 2020-10-23:
> @Koenkk yes, they are on the map. It is main switch for bedroom and bathroom I believe

But do they still show as connected on the map after they **stopped** working?

Shaquu, 2020-10-23:
> > @Koenkk yes, they are on the map. It is main switch for bedroom and bathroom I believe
> 
> 
> 
> But do they still show as connected on the map after they **stopped** working?

Yes, it is on the map. Bedroom switch LQI 84

Koenkk, 2020-10-24:
@Shaquu would you able to sniff the traffic for the following two scenarios (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html):
- Controlling works (send e.g. on command): 
- Stopped working

Shaquu, 2020-10-25:
@Koenkk question to you. Do I have to sniff with the same stick my zigbee devices are connected with?
Or can I sniff other stick? If second than there is no logs. If first then I am unable to start zigbee2mqtt on my Win10 to pair devices.

Shaquu, 2020-10-26:
Possibly related with #4150

Koenkk, 2020-10-26:
You need to sniff with an other stick.

Shaquu, 2020-10-28:
@Koenkk no results while sniffing:
[image]

Koenkk, 2020-10-29:
Make sure you sniff on the correct channel, you should se traffic even when you don't execute a command.

domnich, 2020-10-29:
> Thanks Koenkk. After my previous comment I added property "disableDefaultResponse: ture" into Blitzwolf shp-13 object to the "meta" (in devices.js) and for now everything works Ok...I can even say [...]

Just a minor update from my side... Since I added property "disableDefaultResponse: ture" into Blitzwolf shp-13 everything works OK in my case, was not able to reproduce thia bug anymore..

Shaquu, 2020-10-29:
> Make sure you sniff on the correct channel, you should se traffic even when you don't execute a command.

According to z2m UI it is channel 11.
[image]

Koenkk, 2020-10-30:
@Shaquu then something is wrong with the sniffer setup, I see you use ZBOSS, maybe you can try with a CC2531.

Shaquu, 2020-10-30:
@Koenkk I am using ZBOSS gui app according to https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html
My stick is CC2531 as you said.

Koenkk, 2020-10-31:
I see, not sure what is wrong with your setup. You might want to try with linux/wirehsark.

closed 2020-12-08 by github-actions[bot]

biejay, 2021-12-30:
Hey @domnich 
it's a while ago but how did you edit your devices.js to `disableDefaultResponse: true` ?

I run Zigbee2mqtt Addon on Hassio and my multiple TS0121_plug and TS011F_plug_3 sometimes don't toggle or only after a really long time. So I guess I got the same problem as you.

Thanks in advance!

rdl987, 2023-09-28:
Hello,
I have the same problem but in my case it's regarding all devices, it became unresponsive.

I'm using zigbee2mqtt on docker running in a qnap nas and a SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P.

Could anyone help me?
Thanks

KostyaSha, 2023-09-29:
Seems starts failing with the latest update

marcusletric, 2023-09-29:
I have this exact problem but my setup is way different. 
I got a SonOff ZB BRIDGE-P that I put tasmota on, also got a USB zigbee repeater

My HA as well as Z2M and Mosquitto on docker. I get this issue with a Moes light switch TS0011 and Moes TRV too TS0601, when I operate the devices the status updates on HA but when I remotely try to change state from HA, I get this error.

Additionally when I rearranged my network and gave some space for zigbee channels and moved coordinator and repeater around, this issue appeared less often, but still does.

Unpairing and re-pairing solves the issue for a while.

I haven't tried disable default response but I don't know how that can be done, should that solve my issue for good? Why is this even happening in the first place?

rdl987, 2023-10-07:
Good morning, 
I tried to downgrade to the 20221226 firmware but this solve only for a while.

Could adding also `disableDefaultResponse: true` solve?

Macht59, 2024-02-21:
I have the same issue with SONOFF S31ZB

```
Error 2024-02-20 20:37:01Publish 'set' 'state' to 'Knit Lamp' failed: 'Error: Command 0x00124b002a634909/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
```

raadms, 2024-05-10:
> Thanks Koenkk. After my previous comment I added property "disableDefaultResponse: ture" into Blitzwolf shp-13 object to the "meta" (in devices.js) and for now everything works Ok...I can even say [...]

Thank @domnich, 

I have the same issue can you please explain where can i find that file and what shall I add. much appreciate it. 

thnx