Koenkk/zigbee2mqtt #4711: Reporting stopped working after upgrading to 1.15.0/20200925

esbenr, 2020-10-18:
## What happened
I have ZZH running stable with 63 devices on 1.14.4/20200805, primarily with Trådfri and Aqara. Suddenly over night Z2M stopped responding and after restarting Z2M the ZZH adapter wasn't found anymore. Tried re-plugging the ZZH stick many, many times and restarting my Pi 3B+ (stick is already on extension cord).

After updating to latest Z2M and latest 3.x.0 firmware ZZH was found again and Z2M started as usual.
But, now reporting doesn't work anymore, for any device on the network.

## What did you expect to happen
After upgrading the ZZH firmware and Z2M to latest, I expected operation as usual and devices to report state to the coordinator.

## How to reproduce it (minimal and precise)
Have an existing earlier version network with working reporting, upgrade to newest and confirm reporting when changing state using switches that are bound to the bulbs.

## Debug info
Zigbee2MQTT version: 1.14.4 upgrade to 1.15.0 
Adapter hardware: CC26X2R1
Adapter firmware version: 20200805 upgrade to 20200925

Koenkk, 2020-10-19:
Can you try to set `report: false`in the configuration, start z2m, wait a minute, stop z2m and set report to true again? Do devices start reporting again?

fergus, 2020-10-20:
Hi @Koenkk I have tried setting and resetting `report: false` To `report: true` As you suggested. But there is no change in behaviour for myself in build 1.15. Devices are not reporting.  
I am using the ConBee 2 however.

Koenkk, 2020-10-20:
Can you share your database.db  and log when starting zigbee2mqtt? (first 60 seconds)

esbenr, 2020-10-26:
Re-setting report:false and then report: true didn't help either at the first time. 
Restarting a couple of times made it succesfully setting up reporting for most bulbs.
So now most of my Trådfri bulbs report back, though with a noticeable delay.

It would be cool to have some kind of indication of reporting is set up on each bulb in the overview and an option to try tp force it.

Dielee, 2020-11-01:
Same issue here! Setting `report: false` restart, and then `report: true` fixes the problem. But the problem occurs again after some Time. Tryed the latest dev brach, without improvement.

closed 2020-12-09 by github-actions[bot]

esbenr, 2020-12-10:
This is still a problem.

I occasionally have bulbs that "loose" reporting on Z2M after a while.

Here is the scenario:
I have a Trådfri 1000lm bulb bound to a Trådfri switch and both in Z2M. When turning off the light using the switch, the light goes out as expected, but the status is still ON in Homeassistant. This indicates that the bulb doesent report correctly.

In the Z2M UI when clicking "configure" it fails with error "could not configure...".

Permitting join and resetting the bulb announces the bulb but reporting fails:
´´´
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living room Black floor lamp","ieee_address":"0x000b57fffe39fb68"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Living room Black floor lamp"},"type":"device_announced"}'
ERROR: Failed to setup reporting for '0x000b57fffe39fb68' - Error: Bind 0x000b57fffe39fb68/1 genLevelCtrl from '0x00124b00215ff95a/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Timeout._onTimeout (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:549:17) at processTimers (internal/timers.js:492:7)
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living room Black floor lamp","ieee_address":"0x000b57fffe39fb68"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Living room Black floor lamp"},"type":"device_announced"}'
´´´
For some reason setting up reporting is failing.