Koenkk/zigbee2mqtt #4717: TERNCY-WS01-D1

jccg315, 2020-10-19:
## Information about the device + link
..TERNCY-WS01-D1
https://www.xiaoyan.io/zh-cn/wallswitch-withneutral/

## data/database.db entry of the device
..

jccg315, 2020-10-19:
data/database.db entry of the device
info  2020-07-20 23:20:03: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.1","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
info  2020-07-20 23:20:20: Device '0x000d6f0014463374' joined
info  2020-07-20 23:20:20: Starting interview of '0x000d6f0014463374'
info  2020-07-20 23:20:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f0014463374"}}'
info  2020-07-20 23:20:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f0014463374"}}'
debug 2020-07-20 23:20:20: Device '0x000d6f0014463374' announced itself
info  2020-07-20 23:20:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x000d6f0014463374"}}'
debug 2020-07-20 23:20:20: Device '0x000d6f0014463374' announced itself
info  2020-07-20 23:20:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x000d6f0014463374"}}'
debug 2020-07-20 23:20:20: Received Zigbee message from '0x000d6f0014463374', type 'attributeReport', cluster 'genBasic', data '{"modelId":"TERNCY-WS01-S2"}' from endpoint 1 with groupID 0
debug 2020-07-20 23:20:20: No converter available for 'TERNCY-WS01-S2' with cluster 'genBasic' and type 'attributeReport' and data '{"modelId":"TERNCY-WS01-S2"}'
debug 2020-07-20 23:20:20: Received Zigbee message from '0x000d6f0014463374', type 'attributeReport', cluster 'genOta', data '{"currentFileVersion":95}' from endpoint 1 with groupID 0
debug 2020-07-20 23:20:20: Received Zigbee message from '0x000d6f0014463374', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-07-20 23:20:20: MQTT publish: topic 'zigbee2mqtt/0x000d6f0014463374', payload '{"state":"OFF","linkquality":31,"state_left":"OFF","state_right":"OFF"}'
debug 2020-07-20 23:20:20: Received Zigbee message from '0x000d6f0014463374', type 'attributeReport', cluster 'manuSpecificClusterAduroSmart', data '{"25":1}' from endpoint 1 with groupID 0
debug 2020-07-20 23:20:20: No converter available for 'TERNCY-WS01-S2' with cluster 'manuSpecificClusterAduroSmart' and type 'attributeReport' and data '{"25":1}'
debug 2020-07-20 23:20:27: Received Zigbee message from '0x000d6f0014463374', type 'attributeReport', cluster 'manuSpecificClusterAduroSmart', data '{"32":0}' from endpoint 2 with groupID 0
debug 2020-07-20 23:20:27: No converter available for 'TERNCY-WS01-S2' with cluster 'manuSpecificClusterAduroSmart' and type 'attributeReport' and data '{"32":0}'

closed 2020-11-26 by github-actions[bot]

Maozhenlei, 2022-10-06:
Did you use it successfully?