Koenkk/zigbee2mqtt #4719: Cannot start z2m after dev path changed

ooii, 2020-10-19:
## What happened
Hi, I restarted the host where HA is running. After restart, the serial port of my zzh moved from /dev/ttyUSB0 to /dev/ttyUSB1. z2m could not start so I changed the pan_id. z2m starts now but all the devices disappeared from my MQTT integration!! (which is expected). Restarted with only the zzh, got it in /dev/ttyUSB0, reverted back to the original pan_id but z2m does not start.
Here are the logs
```
Zigbee2MQTT:error 2020-10-17 22:01:38: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-17 22:01:38: Failed to start zigbee
Zigbee2MQTT:error 2020-10-17 22:01:38: Exiting...
Zigbee2MQTT:error 2020-10-17 22:01:38: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
```
I flashed the stick and now I have a new error message: 
```
Error: Cannot restore backup, panID of backup is different
    at Object.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/backup.js:181:19)
```
I have then deleted the backup, then restarted. z2m does not complain anymore about the backup but cannot start. Left it running without being able to make it start. Next morning, it seems it has managed to start, it recognizes only my Hue sensors and dimmers. However, the friendly name in HA is reset back to the one of z2m. I have to pair again all other devices and reconfigure all that.

## What did you expect to happen
z2m should start

## How to reproduce it (minimal and precise)
In my case, changing the dev path requires a new pan_id. A new pan_id make all paired devices with HA disappear.

## Debug info
Zigbee2MQTT version: 1.15 as add-on of Home assistant 0.116.2
Adapter hardware: CC26X2R1
Adapter firmware version: Z-Stack_3.x.0 of 20200925

Koenkk, 2020-10-19:
Can you reproduce this again? (don't change pan_id). And provide the herdsman debug log of this.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

ooii, 2020-10-19:
Unfortunately not now that it has managed to start properly.

closed 2020-11-26 by github-actions[bot]