Koenkk/zigbee2mqtt #4720: Blaupunkt PSM-S1 recognized as NOT SUPPORTED (PSMP5_00.00.03.16TC)

tomekpap, 2020-10-19:
# Bug Report
## What happened
Blaupunkt Power Metering Socket PSM-S1 joined the network with zigbee device model: PSMP5_00.00.03.16TC as NOT SUPPORTED, I found that regular zigbee device model for this device is: PSMP5_00.00.05.10TC

## What did you expect to happen
Successful pairing a s supported device

## How to reproduce it (minimal and precise)
Enter Pairing mode in PSM-S1, wait a few seconds

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: latest dev

Koenkk, 2020-10-19:
Added support for it.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

tomekpap, 2020-10-19:
You're the best!

Thanks

tomekpap, 2020-10-20:
I can confirm everything looks like working well, device paired and gives power readings etc.

Thanks

closed 2020-10-20 by tomekpap

Metzlmane, 2021-01-09:
Can the "PSMP5_00.00.03.19TC" also be added please?
From what I can see, there is no difference to the .16TC version zigbee wise

Koenkk, 2021-01-09:
@Metzlmane added, 

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)