Koenkk/zigbee2mqtt #4725: edge: query aps data spam in log

andriej, 2020-10-20:
## What happened
After update to latest z2m-edge, even when log_level set to warn, there's spam in log:
```
Zigbee2MQTT:info  2020-10-20 02:04:53: Switching log level to 'warn'
query aps data confirm
query aps data indication
query aps data confirm
query aps data indication
query aps data confirm
query aps data indication
query aps data confirm
query aps data confirm
query aps data indication
query aps data confirm
query aps data indication
query aps data confirm
query aps data indication
query aps data confirm
query aps data confirm
query aps data indication
query aps data confirm
query aps data indication
[...]
```

warn is set both to MQTT log and log_level config

## What did you expect to happen
Cleaner log

## How to reproduce it (minimal and precise)
Update zigbee2mqtt-edge to latest (2020/10/20) and run

## Debug info
Zigbee2MQTT version: latest dev
Adapter hardware: Conbee II
Adapter firmware version: latest

andriej, 2020-10-20:
Also what's noticable - there's slight delay in whole MQTT network regarding how it performs with same setup on HA (CPU load didn't change)

Koenkk, 2020-10-20:
@chrishae can you check this?

andriej, 2020-10-21:
Is it related to https://github.com/Koenkk/zigbee-herdsman/commit/1e7005f2cf880f4bfd6e72608f626d3e896da9b5 ?
I've tried re-installing the addon in HA (so latest -edge) but still there's noticeable delay, especially seen in buttons and reed switches.

Although I'm not sure if it made to addon yet, as in logs only entries I see are still 
```
query aps data indication
query aps data confirm
query aps data confirm
query aps data confirm
query aps data indication
query aps data confirm
query aps data indication
query aps data confirm
[...]
```
only

ChrisHae, 2020-10-22:
It is fixed now

andriej, 2020-10-22:
Hmm, somehow the update doesn't want to hop into my addon ;-)

ChrisHae, 2020-10-22:
Ok, it is in the latest zigbee-herdsman commit. Perhaps soon in mqtt-dev or hassio addon

andriej, 2020-10-22:
@Koenkk can you push it to _dev so -edge will download fix? 
My WAF is dropping... ;-)))

ChrisHae, 2020-10-22:
I think only @koenkk can do it.

andriej, 2020-10-22:
@ChrisHae right, I'll wait. And about those messages - will they respect log_level with the fix?

ChrisHae, 2020-10-22:
yes, they are now debug-log messages

Koenkk, 2020-10-22:
Updated zigbee-herdsman, edge is now rebuilding and should be done in an hour from now.

andriej, 2020-10-22:
Fixed.
Buttons work normally back again, thanks.

closed 2020-10-22 by andriej